Koenkk/zigbee2mqtt #12650: TuYa TS0001_switch_module pairing issue

basdelfos, 2022-05-30:
### What happened?

Device is discovered as TS0001_switch_module and seems to discover the endpoints. After pairing seems to be successful, the blue led on the device stays flashing. After ~10 seconds Z2M reports that the device as left the network. After a few seconds the devices is found again and starts pairing again and leaving the network again. This all starts over again.

I bought this device on Aliexpress: https://www.aliexpress.com/item/1005003417122659.html?spm=a2g0o.order_list.0.0.663079d2rqeR0I&gatewayAdapt=glo2nld

### What did you expect to happen?

Device successfully pairs with the network and the led stops blinking.

### How to reproduce it (minimal and precise)

Put Z2M in pairing mode and start the device or reset into paring mode

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20210708

### Adapter

SONOFF ZB Dongle-P USB Dongle Plus Zigbee 3.0

### Debug log

[image]

basdelfos, 2022-05-30:
I have the same issue with the following device `TuYa TS1101_dimmer_module_1ch`, also seems to pair successfully while led stays blicking and then device leaves network. Pairing process starts over again.

Aliexpress: https://www.aliexpress.com/item/1005003693644741.html?spm=a2g0o.order_list.0.0.10dc1802GhbE6i

basdelfos, 2022-05-30:
Seems like Z2M thinks the device paired successfully but the device is not informed that pairing (if there is such a thing in the pairing process)

basdelfos, 2022-05-31:
I've enabled debug on `zigbee-herdsman-converters` with environment variable `DEBUG=zigbee-herdsman*` and found some messages that are undefined. Maybe this is related to the issue.

[image]

Koenkk, 2022-06-01:
Just to be sure, can you update to the latest firmware first? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

If that doesn't help provide the herdsman debug logging starting at the point where you pair the device until it starts to leave.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

basdelfos, 2022-06-08:
Issue has been solved by updating the firmware on the Sonoff Zigbee 3.0 USB Dongle Plus to the latest version 20220219. (https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20220219.zip).

Thanks for your advice @Koenkk.

closed 2022-06-08 by basdelfos