Koenkk/zigbee2mqtt #12651: Müller Licht Remote identified wrong

RubenKelevra, 2022-05-30:
### What happened?

I got a remote from Müller Licht which identifies itself as:

```

Device type EndDevice
Zigbee Model tint-Remote-white
Zigbee Manufacturer MLI
Description Tint dim remote control
Support status Supported
Network address 0xB0A5
Manufacturer Müller Licht
Model 404022
```

But the picture is completely wrong, it's looks similar to this one with some minor differences (maybe the picture in the wiki is just simplified).

https://www.zigbee2mqtt.io/devices/MLI-404011.html

On the device itself is a sticker which states:

"Art-Nr 404049C" which means "Type number 404049C"

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB stick

Koenkk, 2022-05-31:
Similar as https://github.com/Koenkk/zigbee2mqtt/issues/12648

> As a result of the wrong identification, it seems that only ON/OFF and the Brightness and color temperature buttons work. Everything else isn't shown in Zigbee2MQTT.

What more would you expect?

dk5ra, 2022-06-03:
still gets identified as 404022
Zigbee2MQTT Version    1.25.2-dev commit: [e525919]

maybe the "C" at the end of the modelnumber is missing? im new into zigbee/z2m. is there a way to debug this or to help you with some logs to solve it?

MartB, 2022-06-04:
> still gets identified as 404022 Zigbee2MQTT Version 1.25.2-dev commit: [[e525919](https://github.com/Koenkk/zigbee2mqtt/commit/e525919160a4ac2b824932ffbec49a0ef9513c96)]
>
> maybe the "C" at the end of [...]

You could try to "hack" your `node_modules/zigbee-herdsman-converters/devices/müller_licht.js` file locally and change `MLI-404011/MLI-404049` to `MLI-404011/MLI-404049C`.

However I'm not sure how this gets matched internally but its worth a shot if you wanna try :)

RubenKelevra, 2022-06-04:
> Similar as https://github.com/Koenkk/zigbee2mqtt/issues/12648
> 
> > As a result of the wrong identification, it seems that only ON/OFF and the Brightness and color temperature buttons work. Everything else isn't shown in Zigbee2MQTT.
> 
> What more would you expect?

Well, there are 6 scene buttons, four color hue buttons, a light selection button with has 1/2/3 and "all three" selections (via LED feedback).

Would be cool if those would work too. :)

RubenKelevra, 2022-06-05:
Here's how the Remote looks like:

[image]

Which is similar or the same as the old remote (feature wise):

[image]

Koenkk, 2022-06-05:
Fixed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Koenkk, 2022-06-05:
Do you get any `no converter available` messages in the z2m log when triggering the currently unsupported actions?

RubenKelevra, 2022-06-13:
@Koenkk wrote:
> Do you get any `no converter available` messages in the z2m log when triggering the currently unsupported actions?

Well here are all messages per button press:

### _brightness down_

```log
debug 2022-06-13 18:51:54Received Zigbee message from 'Light Remote Lina's Room', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":43,"transtime":10}' from endpoint 1 with groupID 16388
info 2022-06-13 18:51:54MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"brightness_step_down","action_group":16388,"action_step_size":43,"action_transition_time":0.1,"linkquality":83}'
info 2022-06-13 18:51:54MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"","linkquality":83}'
info 2022-06-13 18:51:54MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room/action', payload 'brightness_step_down'
```

### _brightness up_

```log
debug 2022-06-13 18:52:46Received Zigbee message from 'Light Remote Lina's Room', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":43,"transtime":10}' from endpoint 1 with groupID 16388
info 2022-06-13 18:52:46MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"brightness_step_up","action_group":16388,"action_step_size":43,"action_transition_time":0.1,"linkquality":72}'
info 2022-06-13 18:52:46MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"","linkquality":72}'
info 2022-06-13 18:52:46MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room/action', payload 'brightness_step_up'
```

### _color temperature warm white_

```log
debug 2022-06-13 18:54:41Received Zigbee message from 'Light Remote Lina's Room', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":417,"transtime":10}' from endpoint 1 with groupID 16388
info 2022-06-13 18:54:41MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"color_temperature_move","action_color_temperature":417,"action_group":16388,"action_transition_time":10,"linkquality":83}'
info 2022-06-13 18:54:41MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"","linkquality":83}'
info 2022-06-13 18:54:41MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room/action', payload 'color_temperature_move'
```

### _color temperature neutral white (center of the bar)_

```log
debug 2022-06-13 18:56:16Received Zigbee message from 'Light Remote Lina's Room', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":250,"transtime":10}' from endpoint 1 with groupID 16388
info 2022-06-13 18:56:16MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"color_temperature_move","action_color_temperature":250,"action_group":16388,"action_transition_time":10,"linkquality":72}'
info 2022-06-13 18:56:16MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"","linkquality":72}'
info 2022-06-13 18:56:16MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room/action', payload 'color_temperature_move'
```

### _color temperature cool white_

```log
debug 2022-06-13 18:56:33Received Zigbee message from 'Light Remote Lina's Room', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":200,"transtime":10}' from endpoint 1 with groupID 16388
info 2022-06-13 18:56:33MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"color_temperature_move","action_color_temperature":200,"action_group":16388,"action_transition_time":10,"linkquality":80}'
info 2022-06-13 18:56:33MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"","linkquality":80}'
info 2022-06-13 18:56:33MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room/action', payload 'color_temperature_move'
```

### _On/Off button (twice)_

```log
debug 2022-06-13 18:57:26Received Zigbee message from 'Light Remote Lina's Room', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 16388
info 2022-06-13 18:57:26MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"on","action_group":16388,"linkquality":40}'
info 2022-06-13 18:57:26MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"","linkquality":40}'
info 2022-06-13 18:57:26MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room/action', payload 'on'
debug 2022-06-13 18:57:28Received Zigbee message from 'Light Remote Lina's Room', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 16388
info 2022-06-13 18:57:28MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"off","action_group":16388,"linkquality":76}'
info 2022-06-13 18:57:28MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room', payload '{"action":"","linkquality":76}'
info 2022-06-13 18:57:28MQTT publish: topic 'zigbee2mqtt/Light Remote Lina's Room/action', payload 'off'
```

### _color wheel right (red)_

```log
 2022-06-13 18:57:48Received Zigbee message from 'Light Remote Lina's Room', type 'commandMoveToColor', cluster 'lightingColorCtrl', data '{"colorx":45875,"colory":19660,"transtime":0}' from endpoint 1 with groupID 16388
```

### _color wheel top (violet)_
[... 43 lines cut ...]
```

### _right-top scene button (Disco ball)_

```log
[... cut ...]
debug 2022-06-13 19:10:20Received Zigbee message from 'Light Remote Lina's Room', type 'write', cluster 'genBasic', data '{"16389":6}' from endpoint 1 with groupID 16388
```

### _ceter-bottom scene button (Campfire)_

```log
debug 2022-06-13 19:11:09Received Zigbee message from 'Light Remote Lina's Room', type 'write', cluster 'genBasic', data '{"16389":4}' from endpoint 1 with groupID 16388
```

### _right-bottom scene button (Two hearts)_

```log
debug 2022-06-13 19:11:35Received Zigbee message from 'Light Remote Lina's Room', type 'write', cluster 'genBasic', data '{"16389":5}' from endpoint 1 with groupID 16388
```

RubenKelevra, 2022-06-13:
Two details:

- The light group button does not send a command itself, above are just other button presses to show how the selection influences the data.
- The white color bar has three physical buttons, the plus/minus bar for brightness has only two.

Koenkk, 2022-06-14:
Looks good, seems to be just a matter of adding the converters. See  https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions

The `commandMoveToColor` should be handled when adding the  `fz.command_move_to_color` converter.  The "right-bottom scene button" command looks custom so you have to write your own converter for this. The following external converter should give you a good start.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['tint-Remote-white'],
    model: '404022',
    description: 'Tint dim remote control',
    vendor: 'Müller Licht',
    fromZigbee: [fz.command_on, fz.command_off, fz.command_step, fz.command_move, fz.command_stop, fz.command_move_to_color_temp, fz.command_move_to_color],
    exposes: [e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down',
        'brightness_stop', 'color_temperature_move'])],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        device.powerSource = 'Battery';
        device.save();
    },
};

module.exports = definition;

```

RubenKelevra, 2022-07-23:
@Koenkk please reopen because this is not stale.

Why is the stale bot insisting on that? That doesn't make any sense.

I can't write JS, so I can only copy'n'paste something together and probably wrong. :)

francoisa43000, 2022-07-27:
Hello,
The remote control has the physical reference "404049C" but it is recognized as "404022", so information on zigbee2mqtt is missing. Would it be possible to modify? It looks identical to the classic "404049" it looks like a new version. 
Cordially

OMotornyi, 2022-08-31:
Please don't mark it stale... I am facing the issue and it would be nice to get this one running

Vumer4, 2022-09-14:
got the same problem

Koenkk, 2022-09-15:
I posted an external converter some time back: https://github.com/Koenkk/zigbee2mqtt/issues/12651#issuecomment-1155319765 but it seems nobody tried it yet. Can anyone try it or is the problem different?

francoisa43000, 2022-09-15:
Hello,
In the external converter the reference is 404022, my remote control has the reference 404049C, is it possible to modify the converter to make it work with this reference?
Cordially

Koenkk, 2022-09-15:
I've renamed the model to `404022/404049C`. Will be included in the next z2m release on 1 October.

RubenKelevra, 2022-09-15:
I've made a better image of the remote, if you like to include a picture of it:

[image]

License: CC-0

Koenkk, 2022-09-16:
@RubenKelevra thanks, updated the image (it will show both now)

RubenKelevra, 2022-09-16:
> @RubenKelevra thanks, updated the image (it will show both now)

I can test the converter this weekend. What info do you need? 🤔

Koenkk, 2022-09-19:
@RubenKelevra check if it fixes the points from https://github.com/Koenkk/zigbee2mqtt/issues/12651#issuecomment-1154176755

francoisa43000, 2022-09-22:
hello, 
I updated my Z2M to the latest dev version (1.27.2-dev commit), the remote control is well recognized, but some key does not work. Especially the color keys.
Cordially

Koenkk, 2022-09-22:
Can you provide the debug log when pressing the color keys?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

francoisa43000, 2022-09-22:
Hello,
Here are the logs from zigbee2mqtt for the remote when I press the color buttons:

2022-09-22 20:18:26Received MQTT message on 'homeassistant/device_automation/0xa4c13800aa302652/action_off/config' with data '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0xa4c13800aa302652"] ,"manufacturer":"Müller Licht","model":"Tint dim remote control (404022/404049C)","name":"Remote control Muller","sw_version":"MRM001"},"payload":"off ","subtype":"off","topic":"zigbee-espaly/Muller remote control/action","type":"action"}'
2022-09-22 20:21:29Received Zigbee message from 'Remote control Muller', type 'commandMoveToColor', cluster 'lightingColorCtrl', data '{"colorx":34493,"colory":29692,"transtime":0}' from endpoint 1 with groupID 16388
2022-09-22 20:21:53Received Zigbee message from 'Remote control Muller', type 'commandMoveToColor', cluster 'lightingColorCtrl', data '{"colorx":36633,"colory":27806,"transtime":0}' from endpoint 1 with groupID 16388
2022-09-22 20:21:55Received Zigbee message from 'Remote Control Muller', type 'commandMoveToColor', cluster 'lightingColorCtrl', data '{"colorx":39202,"colory":25542,"transtime":0}' from endpoint 1 with groupID 16388

The remote control is also poorly recognized in the photo:

[image]

Cordially

Koenkk, 2022-09-23:
@francoisa43000 this should be fixed when using the external converter from https://github.com/Koenkk/zigbee2mqtt/issues/12651#issuecomment-1155319765

francoisa43000, 2022-09-23:
with the external converter, I do not have all the functional buttons, I miss the 6 scene buttons. I must have made a bad adjustment, someone tried it on his side?

Koenkk, 2022-09-23:
What is logged in the debug log when you press the scene button?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

francoisa43000, 2022-09-23:
debug 2022-09-23 22:13:01Received Zigbee message from 'Telecommande Muller', type 'write', cluster 'genBasic', data '{"16389":3}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:01MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":null,"last_seen":"2022-09-23T20:13:00.715Z","linkquality":90}'
debug 2022-09-23 22:13:02Received Zigbee message from 'Telecommande Muller', type 'write', cluster 'genBasic', data '{"16389":1}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:02MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":null,"last_seen":"2022-09-23T20:13:01.664Z","linkquality":90}'
debug 2022-09-23 22:13:03Received Zigbee message from 'Telecommande Muller', type 'write', cluster 'genBasic', data '{"16389":2}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:03MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":null,"last_seen":"2022-09-23T20:13:02.501Z","linkquality":90}'
debug 2022-09-23 22:13:04Received Zigbee message from 'Telecommande Muller', type 'write', cluster 'genBasic', data '{"16389":6}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:04MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":null,"last_seen":"2022-09-23T20:13:03.375Z","linkquality":90}'
debug 2022-09-23 22:13:05Received Zigbee message from 'Telecommande Muller', type 'write', cluster 'genBasic', data '{"16389":4}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:05MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":null,"last_seen":"2022-09-23T20:13:04.123Z","linkquality":90}'
debug 2022-09-23 22:13:06Received Zigbee message from 'Telecommande Muller', type 'write', cluster 'genBasic', data '{"16389":5}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:06MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":null,"last_seen":"2022-09-23T20:13:04.993Z","linkquality":90}'
debug 2022-09-23 22:13:14Received Zigbee message from 'Telecommande Muller', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":43,"transtime":10}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:14MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"brightness_step_down","action_group":16388,"action_step_size":43,"action_transition_time":0.1,"last_seen":"2022-09-23T20:13:13.413Z","linkquality":90}'
info 2022-09-23 22:13:14MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"","last_seen":"2022-09-23T20:13:13.413Z","linkquality":90}'
info 2022-09-23 22:13:14MQTT publish: topic 'zigbee-espaly/Telecommande Muller/action', payload 'brightness_step_down'
debug 2022-09-23 22:13:15Received Zigbee message from 'Telecommande Muller', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":43,"transtime":10}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:15MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"brightness_step_up","action_group":16388,"action_step_size":43,"action_transition_time":0.1,"last_seen":"2022-09-23T20:13:14.603Z","linkquality":90}'
info 2022-09-23 22:13:15MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"","last_seen":"2022-09-23T20:13:14.603Z","linkquality":90}'
info 2022-09-23 22:13:15MQTT publish: topic 'zigbee-espaly/Telecommande Muller/action', payload 'brightness_step_up'
debug 2022-09-23 22:13:17Received Zigbee message from 'Telecommande Muller', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":200,"transtime":10}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:17MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"color_temperature_move","action_color_temperature":200,"action_group":16388,"action_transition_time":10,"last_seen":"2022-09-23T20:13:16.194Z","linkquality":90}'
info 2022-09-23 22:13:17MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"","last_seen":"2022-09-23T20:13:16.194Z","linkquality":90}'
info 2022-09-23 22:13:17MQTT publish: topic 'zigbee-espaly/Telecommande Muller/action', payload 'color_temperature_move'
debug 2022-09-23 22:13:17Received Zigbee message from 'Telecommande Muller', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":153,"transtime":10}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:17MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"color_temperature_move","action_color_temperature":153,"action_group":16388,"action_transition_time":10,"last_seen":"2022-09-23T20:13:16.842Z","linkquality":90}'
info 2022-09-23 22:13:17MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"","last_seen":"2022-09-23T20:13:16.842Z","linkquality":90}'
info 2022-09-23 22:13:17MQTT publish: topic 'zigbee-espaly/Telecommande Muller/action', payload 'color_temperature_move'
debug 2022-09-23 22:13:18Received Zigbee message from 'Telecommande Muller', type 'commandMoveToColor', cluster 'lightingColorCtrl', data '{"colorx":22156,"colory":10663,"transtime":0}' from endpoint 1 with groupID 16388
info 2022-09-23 22:13:18MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"color_move","action_color":{"x":0.338,"y":0.163},"action_group":16388,"action_transition_time":0,"last_seen":"2022-09-23T20:13:17.886Z","linkquality":87}'
[... cut ...]
info 2022-09-23 22:13:20MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"color_move","action_color":{"x":0.338,"y":0.163},"action_group":16388,"action_transition_time":0,"last_seen":"2022-09-23T20:13:19.835Z","linkquality":90}'
info 2022-09-23 22:13:20MQTT publish: topic 'zigbee-espaly/Telecommande Muller', payload '{"action":"","last_seen":"2022-09-23T20:13:19.835Z","linkquality":90}'
info 2022-09-23 22:13:20MQTT publish: topic 'zigbee-espaly/Telecommande Muller/action', payload 'color_move'

I pressed all the buttons on the remote control.

Koenkk, 2022-09-25:
Thanks added support for the scene buttons now. please check with https://github.com/Koenkk/zigbee2mqtt/issues/12651#issuecomment-1155319765

francoisa43000, 2022-09-25:
Hello, I updated the external converter but I still encounter the same problem, my remote control is a 404049C and not a 404022, in the external converter it is noted 404022. Is it possible to make a converter just for this model ?

Vumer4, 2022-09-25:
Hello,
404049C and 404022 are different devices. The 404049C is the successor of[MLI-404049](https://www.zigbee2mqtt.io/devices/MLI-404011_MLI-404049.html)

RubenKelevra, 2022-09-27:
> Hello, I updated the external converter but I still encounter the same problem, my remote control is a 404049C and not a 404022, in the external converter it is noted 404022. Is it possible to make a converter just for this model ?

As far as I understand, both remotes do report the same model number via Zigbee, so it's impossible to distinguish them from each other.

Keep in mind, they are all made to communicate to the light bulbs directly. This (likely) requires that the light bulbs know the model number and the key presses in advance. To control older bulbs, the model number and key numbers therefore needs to be the same on newer remotes. :)

RubenKelevra, 2022-09-27:
> Hello, 404049C and 404022 are different devices. The 404049C is the successor of [MLI-404049](https://www.zigbee2mqtt.io/devices/MLI-404011_MLI-404049.html)

Why do you think that this is not obvious to us?

francoisa43000, 2022-09-27:
> As far as I understand, both remotes do report the same model number via Zigbee, so it's impossible to distinguish them from each other.
>
> Keep in mind, they are all made to communicate to the light [...]

Hello,
yes I understand that the bulb knows in advance the commands of the remote control. However, I am only looking to help with the integration of this remote control. I don't have the ability to do the integrations myself.

RubenKelevra, 2022-09-27:
@francoisa43000 so what exactly your question then? :)

francoisa43000, 2022-09-27:
> @francoisa43000 so what exactly your question then? :)

The "scenes" buttons that still do not work

OMotornyi, 2022-10-03:
I was able to make scene button work but I had to define the external controller as following:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

function addActionGroup(payload, msg, definition) {
    const disableActionGroup = definition.meta && definition.meta.disableActionGroup;
    if (!disableActionGroup && msg.groupID) {
        payload.action_group = msg.groupID;
    }
}

const fzLocal = {
    scene_404022: {
        cluster: 'genBasic',
        type: 'write',
        convert: (model, msg, publish, options, meta) => {
            const payload = {action: `scene_${msg.data['16389']}`};
            addActionGroup(payload, msg, model);
            return payload;
        },
}
}
const definition = {
    zigbeeModel: ['tint-Remote-white'],
    model: '404022',
    description: 'Tint dim remote control',
    vendor: 'Müller Licht',
    fromZigbee: [fz.command_on, fz.command_off, fz.command_step, fz.command_move, fz.command_stop, fz.command_move_to_color_temp, fz.command_move_to_color, fzLocal.scene_404022],
    exposes: [e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down',
        'brightness_stop', 'color_temperature_move', 'scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6'])],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        device.powerSource = 'Battery';
        device.save();
    },
};

module.exports = definition;`

Here I used [this](https://github.com/Koenkk/zigbee-herdsman-converters/blob/725d7831ad11bfe4ce9bab7f565cefff396c2bd1/converters/fromZigbee.js#L3671) for fzlocal and I took  addActionGroup function from [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/725d7831ad11bfe4ce9bab7f565cefff396c2bd1/lib/utils.js#L108) in the same repo. I am not, however, sure if it's the best way to achieve the end goal

Vumer4, 2022-11-18:
Please don't mark it stale... The problem is not fixed

francoisa43000, 2022-12-21:
Hello,
Scene buttons not working. On Z2M, they are not recognized. Possibility to modify the configuration to make it work?
Cordially

OMotornyi, 2022-12-22:
> Hello, Scene buttons not working. On Z2M, they are not recognized. Possibility to modify the configuration to make it work? Cordially

https://github.com/Koenkk/zigbee2mqtt/issues/12651#issuecomment-1266096410 Have you tried to do what I have outlined in this comment?

francoisa43000, 2022-12-24:
Hello,
Yes, I created the converter but without success by deleting it and restarting z2m, the remote control works. However, z2m does not return all the information it seems. Attached is a screenshot of MQTT Explorer. Would it be possible to add them?
Cordially

[image]

Dunkelschunkel, 2023-01-03:
Hi,
my MLI Remote 404049C does not get recognized at all. 
[image]
I am on Zigbee2MQTT Version 1.29.0
Coordinator-Typ
zStack3x0
Coordinator-Version
20221102

When joining the device i get the error 
`Device '0xa4c13800aa4511bc' with Zigbee model 'Remote Control' and manufacturer name 'MLI' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html`

Sticker on the back of the remote states "tint Müllerlicht Art-NR 404049C"

I am kind of stuck on this one and some help would be greatly appreciated

Dunkelschunkel

Enemyscout, 2023-02-01:
Hello, same Issue as Dunkelschunkel has with my installation

 I am on Zigbee2MQTT Version 1.30.0 

Enemyscout

Enemyscout, 2023-02-01:
> I got it working by using the external converter posted here [#12651 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/12651#issuecomment-1266096410) And all I changed was the the vendor, from 'Müller Licht' to 'MLI' Sorry, I am on mobile. Not enough screen to make it a new issue for another device I hope this works with you Edit: It might have been the other converter. I'll check when at home Feb 1, 2023 8:33:27 PM Enemyscout ***@***.***>:
> […](#)

Thanks for the info. Sadly does not work for me. Error persists.

Enemyscout

Dunkelschunkel, 2023-02-01:
I had to delete and pair it again

Dunkelschunkel, 2023-02-02:
Try force removing and pairing it again. Also I like restarting after adding an external converter.
My external converter 404049c as promised:

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

function addActionGroup(payload, msg, definition) {
    const disableActionGroup = definition.meta && definition.meta.disableActionGro>
    if (!disableActionGroup && msg.groupID) {
        payload.action_group = msg.groupID;
    }
}

const fzLocal = {
    scene_404022: {
        cluster: 'genBasic',
        type: 'write',
        convert: (model, msg, publish, options, meta) => {
            const payload = {action: `scene_${msg.data['16389']}`};
            addActionGroup(payload, msg, model);
            return payload;
        },
}
}
const definition = {
    zigbeeModel: ['Remote Control'],
    model: '404022',
    description: 'Tint dim remote control',
    vendor: 'MLI',
    fromZigbee: [fz.command_on, fz.command_off, fz.command_step, fz.command_move, >
    exposes: [e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down',>
        'brightness_stop', 'color_temperature_move', 'scene_1', 'scene_2', 'scene_>
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        device.powerSource = 'Battery';
        device.save();
    },
};

module.exports = definition;
`

^G Hilfe     ^O Speichern ^W Wo ist    ^K Ausschneid^T Ausführen ^C Position
^X Beenden   ^R Datei öffn^\ Ersetzen  ^U Einfügen  ^J Ausrichten^_ Zu Zeile

Dunkelschunkel, 2023-02-02:
I was also able to bind the remote to the LED-Bulb it came bundled with while keeping both of them connected to Zigbee2MQTT. Even the group bindings as well as the 6 scene buttons are all working.

Dunkelschunkel, 2023-02-02:
@Enemyscout I hope it is working for you. If so we should raise it to the attention of @Koenkk

Enemyscout, 2023-02-02:
@Dunkelschunkel Testing it from remote. Less errors with the device. Will re-pair it when I‘m home. 

Your pasted Versions has some issues I fixed, mostly > or ### in odd places.

Thanks for the help, its progress :)

Enemyscout, 2023-02-02:
`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

function addActionGroup(payload, msg, definition) {
    const disableActionGroup = definition.meta && definition.meta.disableActionGroup;
    if (!disableActionGroup && msg.groupID) {
        payload.action_group = msg.groupID;
    }
}

const fzLocal = {
    scene_404022: {
        cluster: 'genBasic',
        type: 'write',
        convert: (model, msg, publish, options, meta) => {
            const payload = {action: `scene_${msg.data['16389']}`};
            addActionGroup(payload, msg, model);
            return payload;
        },
}
}
const definition = {
    zigbeeModel: ['Remote Control'],
    model: '404022',
    description: 'Tint dim remote control',
    vendor: 'MLI',
    fromZigbee: [fz.command_on, fz.command_off, fz.command_step, fz.command_move, fz.command_stop, fz.command_move_to_color_temp, fz.command_move_to_color, fzLocal.scene_404022],
    exposes: [e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down',
        'brightness_stop', 'color_temperature_move', 'scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6'])],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        device.powerSource = 'Battery';
        device.save();
    },
};

module.exports = definition;
`

ZigbeeModel and Vendor needed to be tweaked. @Dunkelschunkel helped a lot.

Dunkelschunkel, 2023-02-14:
@Koenkk would it be possible to ad this remote. This is basically the 404022/404049c just 
vendor = MLI
ZigbeeModel = Remote Control

Koenkk, 2023-02-14:
@Dunkelschunkel could you provide the data/database.db entry of this device?

Dunkelschunkel, 2023-02-15:
I had to read up on database.db and I am quite sure you are asking for this line

`{"id":12,"type":"EndDevice","ieeeAddr":"0xa4c13800aa4511bc","nwkAddr":7962,"manufId":4635,"manufName":"MLI","powerSource":"Battery","modelId":"Remote Control","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2048,"inClusterList":[0,3,4111],"outClusterList":[3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"hwVersion":0}}},"binds":[{"cluster":6,"groupID":16387,"type":"group"},{"cluster":8,"groupID":16387,"type":"group"},{"cluster":768,"groupID":16387,"type":"group"}],"configuredReportings":[],"meta":{"scenes":{}}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"MDRM001","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-150999557},"lastSeen":1675521628879,"defaultSendRequestWhen":"immediate"}
`
Dunkelschunkel

Koenkk, 2023-02-15:
Could you check if the device works with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{manufacturerName: 'MLI', modelID: 'Remote Control'}],
    zigbeeModel: ['tint-Remote-white'],
    model: '404022/404049C',
    description: 'Tint dim remote control',
    vendor: 'Müller Licht',
    fromZigbee: [fz.command_on, fz.command_off, fz.command_step, fz.command_move, fz.command_stop, fz.command_move_to_color_temp,
        fz.command_move_to_color, fz.tint_scene],
    exposes: [e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down',
        'brightness_stop', 'color_temperature_move', 'color_move', 'scene_1', 'scene_2', 'scene_3', 'scene_4', 'scene_5', 'scene_6'])],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        device.powerSource = 'Battery';
        device.save();
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if device works.

Dunkelschunkel, 2023-02-16:
It got recognized as the correct remote. There is no battery status indicator. But that was the same as with the external converter from this thread I used before [image]
Thank you

Koenkk, 2023-02-16:
@Dunkelschunkel I posted a new converter in https://github.com/Koenkk/zigbee2mqtt/issues/16693, let's continue there.

closed 2023-03-26 by github-actions[bot]