Koenkk/zigbee2mqtt #12652: [Feature request]: Device settings page needs a visual overhaul

RubenKelevra, 2022-05-30:
### Is your feature request related to a problem? Please describe

I find it pretty hard to parse the visuals of the device settings page (and other settings pages.

There's no rime or reason behind the boxes for values and the text around it - at least from the visual appearance:

[image]

### Describe the solution you'd like

Would be neat to have something like alternating background colors or other clear visual guides to separate the individual menu entries.

### Describe alternatives you've considered

None

### Additional context

None

sjorge, 2022-05-31:
You probably want to file that here https://github.com/nurikk/zigbee2mqtt-frontend/issues

Koenkk, 2022-05-31:
Please create issues related to the frontend here; https://github.com/nurikk/zigbee2mqtt-frontend/issues

closed 2022-05-31 by Koenkk