Koenkk/zigbee2mqtt #12655: Legrand 067726 Position set not working

cyrilphoenix71, 2022-05-30:
### What happened?

Legrand 067776 White model is pairing ok
State seems to be ok :
{
    "device": {
        "applicationVersion": 0,
        "dateCode": " \u000020220401\u0000\u0000\u0000\u0000\u0000",
        "friendlyName": "Volet Salle",
        "hardwareVersion": 1,
        "ieeeAddr": "0x0004740000b99330",
        "manufacturerID": 4129,
        "manufacturerName": " Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
        "model": "067776",
        "networkAddress": 12211,
        "powerSource": "Mains (single phase)",
        "softwareBuildID": "001a",
        "stackVersion": 65,
        "type": "Router",
        "zclVersion": 2
    },
    "last_seen": "2022-05-31T00:39:57+02:00",
    "linkquality": 255,
    "position": 66,
    "action": ""
}
But we cannot play with position command, only 100% to open completely works and any value from 0 to 99% goes to close the shutter completely.

### What did you expect to happen?

I want to send (with z2m position slider) 50% in position to open the shutter to 50% level

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

26660700

### Adapter

Conbee2

### Debug log

Info 2022-05-31 00:47:09Succesfully changed options
Info 2022-05-31 00:47:09MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"lq8sz-2"}'
Debug 2022-05-31 00:47:22Received MQTT message on 'zigbee2mqtt/Volet Salle/set' with data '{"position":100}'
Debug 2022-05-31 00:47:22Publishing 'set' 'position' to 'Volet Salle'
Info 2022-05-31 00:47:22MQTT publish: topic 'zigbee2mqtt/Volet Salle', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020220401\u0000\u0000\u0000\u0000\u0000","friendlyName":"Volet Salle","hardwareVersion":1,"ieeeAddr":"0x0004740000b99330","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067776","networkAddress":12211,"powerSource":"Mains (single phase)","softwareBuildID":"001a","stackVersion":65,"type":"Router","zclVersion":2},"last_seen":"2022-05-31T00:47:19+02:00","linkquality":95,"position":48}'
Info 2022-05-31 00:47:22MQTT publish: topic 'zigbee2mqtt/Volet Salle', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020220401\u0000\u0000\u0000\u0000\u0000","friendlyName":"Volet Salle","hardwareVersion":1,"ieeeAddr":"0x0004740000b99330","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067776","networkAddress":12211,"powerSource":"Mains (single phase)","softwareBuildID":"001a","stackVersion":65,"type":"Router","zclVersion":2},"last_seen":"2022-05-31T00:47:19+02:00","linkquality":95,"position":100}'
Debug 2022-05-31 00:47:28Received MQTT message on 'zigbee2mqtt/Volet Salle/set' with data '{"position":46}'
Debug 2022-05-31 00:47:28Publishing 'set' 'position' to 'Volet Salle'
Info 2022-05-31 00:47:28MQTT publish: topic 'zigbee2mqtt/Volet Salle', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020220401\u0000\u0000\u0000\u0000\u0000","friendlyName":"Volet Salle","hardwareVersion":1,"ieeeAddr":"0x0004740000b99330","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067776","networkAddress":12211,"powerSource":"Mains (single phase)","softwareBuildID":"001a","stackVersion":65,"type":"Router","zclVersion":2},"last_seen":"2022-05-31T00:47:25+02:00","linkquality":247,"position":100}'
Info 2022-05-31 00:47:28MQTT publish: topic 'zigbee2mqtt/Volet Salle', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020220401\u0000\u0000\u0000\u0000\u0000","friendlyName":"Volet Salle","hardwareVersion":1,"ieeeAddr":"0x0004740000b99330","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067776","networkAddress":12211,"powerSource":"Mains (single phase)","softwareBuildID":"001a","stackVersion":65,"type":"Router","zclVersion":2},"last_seen":"2022-05-31T00:47:25+02:00","linkquality":247,"position":46}'
Debug 2022-05-31 00:48:09Received MQTT message on 'zigbee2mqtt/Volet Salle/set' with data '{"position":68}'
Debug 2022-05-31 00:48:09Publishing 'set' 'position' to 'Volet Salle'
Info 2022-05-31 00:48:10MQTT publish: topic 'zigbee2mqtt/Volet Salle', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020220401\u0000\u0000\u0000\u0000\u0000","friendlyName":"Volet Salle","hardwareVersion":1,"ieeeAddr":"0x0004740000b99330","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067776","networkAddress":12211,"powerSource":"Mains (single phase)","softwareBuildID":"001a","stackVersion":65,"type":"Router","zclVersion":2},"last_seen":"2022-05-31T00:48:06+02:00","linkquality":215,"position":46}'
Info 2022-05-31 00:48:10MQTT publish: topic 'zigbee2mqtt/Volet Salle', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020220401\u0000\u0000\u0000\u0000\u0000","friendlyName":"Volet Salle","hardwareVersion":1,"ieeeAddr":"0x0004740000b99330","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067776","networkAddress":12211,"powerSource":"Mains (single phase)","softwareBuildID":"001a","stackVersion":65,"type":"Router","zclVersion":2},"last_seen":"2022-05-31T00:48:06+02:00","linkquality":215,"position":68}'
Debug 2022-05-31 00:48:46Received MQTT message on 'zigbee2mqtt/Volet Salle/set' with data '{"position":88}'
Debug 2022-05-31 00:48:46Publishing 'set' 'position' to 'Volet Salle'
Info 2022-05-31 00:48:47MQTT publish: topic 'zigbee2mqtt/Volet Salle', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020220401\u0000\u0000\u0000\u0000\u0000","friendlyName":"Volet Salle","hardwareVersion":1,"ieeeAddr":"0x0004740000b99330","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067776","networkAddress":12211,"powerSource":"Mains (single phase)","softwareBuildID":"001a","stackVersion":65,"type":"Router","zclVersion":2},"last_seen":"2022-05-31T00:48:44+02:00","linkquality":215,"position":68}'
Info 2022-05-31 00:48:47MQTT publish: topic 'zigbee2mqtt/Volet Salle', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020220401\u0000\u0000\u0000\u0000\u0000","friendlyName":"Volet Salle","hardwareVersion":1,"ieeeAddr":"0x0004740000b99330","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067776","networkAddress":12211,"powerSource":"Mains (single phase)","softwareBuildID":"001a","stackVersion":65,"type":"Router","zclVersion":2},"last_seen":"2022-05-31T00:48:44+02:00","linkquality":215,"position":88}'
Debug 2022-05-31 00:48:49Received MQTT message on 'zigbee2mqtt/Volet Salle/set' with data '{"position":95}'
Debug 2022-05-31 00:48:49Publishing 'set' 'position' to 'Volet Salle'
Info 2022-05-31 00:48:49MQTT publish: topic 'zigbee2mqtt/Volet Salle', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020220401\u0000\u0000\u0000\u0000\u0000","friendlyName":"Volet Salle","hardwareVersion":1,"ieeeAddr":"0x0004740000b99330","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067776","networkAddress":12211,"powerSource":"Mains (single phase)","softwareBuildID":"001a","stackVersion":65,"type":"Router","zclVersion":2},"last_seen":"2022-05-31T00:48:46+02:00","linkquality":215,"position":88}'
[... cut ...]
Info 2022-05-31 00:52:01MQTT publish: topic 'zigbee2mqtt/Volet Salle', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020220401\u0000\u0000\u0000\u0000\u0000","friendlyName":"Volet Salle","hardwareVersion":1,"ieeeAddr":"0x0004740000b99330","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067776","networkAddress":12211,"powerSource":"Mains (single phase)","softwareBuildID":"001a","stackVersion":65,"type":"Router","zclVersion":2},"last_seen":"2022-05-31T00:51:58+02:00","linkquality":231,"position":44}'

bemble, 2022-06-18:
I think to be able to have it work we need the `lift_duration` to be set or find a way to automatically calibrate the module (my enocean shutter modules were able to automatically calibrate the lift duration).

Another thing, we are not able to set how the led behave. When I take a look at the converter ([legrand.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/legrand.js#L109)), we should be able to set how the led behaves.

bemble, 2022-06-19:
Setting position is not even possible in legrand app, we have to wait a firmware update from legrand and hope it would be available.

closed 2022-07-28 by github-actions[bot]