Koenkk/zigbee2mqtt #12660: TuYa ZM-35H-Q sending duplicate "occupancy: true" message after exactly 60 seconds

diagonali, 2022-05-31:
### What happened?

I've set up this device as an entry motion sensor. It sends a message when motion is detected but then activates the alarm unexpectedly after it is armed. Upon investigating, I saw that it sends another message exactly 60 seconds after the first which contains a duplicate "occupancy: true" value.

This behaviour makes this sensor almost unusable for an alarm system since it could always send an "occupancy: true" message within 60 seconds, activating an alarm prematurely.

### What did you expect to happen?

I expected it to only send one message upon occupancy and then no more until _after_ 60 seconds had elapsed and motion was detected again.

### How to reproduce it (minimal and precise)

Simply set up the sensor, trigger it and wait 60 seconds to see the second "occupancy: true" message appear in the logs.

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

diagonali, 2022-06-02:
> Looks that the sensor just sends the occupancy true message twice meaning z2m will sent it twice to MQTT.
> 
> ```
> debug 2022-06-01 18:10:37: Received Zigbee message from 'house/downstairs/hall/motion/wall', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
> ...
> debug 2022-06-01 18:11:37: Received Zigbee message from 'house/downstairs/hall/motion/wall', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
> ```
> 
> So behaviour looks ok to me.

Ah, that's a real shame. As you say, the behaviour of the sensor itself is clearly the problem and unfortunately quite a big bug as far as my use case is concerned, since it triggers an alarm with the second (false) message. I really can't understand why the sensor would do this - whether it's intentional or an actual bug.

Unless there's a firmware update, I don't think it's something that's possible to workaround so I'll have to replace all the (8) sensors I bought. This might be the reason they were cheap compared to buying Aqara motion sensors...

Thanks for looking into this though. Appreciate your help.

Koenkk, 2022-06-03:
> EDIT: I'm wondering if changing the min reporting interval for this device could solve the problem? Maybe make it 62 seconds?

This is an ias device, meaning no configured reporting is setup.

You should adapter your automations to respond on the state change e.g. false -> true, not the single message. Even with Xiaomi sensors you can get this. If e.g. occupancy is true and the device report temperature, the following will be send to MQTT: `{temperature: 25, occupancy: true}`

diagonali, 2022-06-04:
> This is an ias device, meaning no configured reporting is setup.
>
> You should adapter your automations to respond on the state change e.g. false -> true, not the single message. Even with Xiaomi [...]

That's very useful thanks for the tip. I did think to adapt my automations for monitoring change only, but didn't realise Xiaomi sensors can also do this (I've never seen it yet) so I thought the duplicates were a bug. It will also make my automations more reliable to check for change only, so I'll definitely try that now instead of buying all new Aqara sensors.

Thanks for your help!

Koenkk, 2022-06-04:
> I did think to adapt my automations for monitoring change only, but didn't realise Xiaomi sensors can also do this (I've never seen it yet) so I thought the duplicates were a bug. 

The chance of this happening is low, but it can happen. So better to make your automations more robust.

> Is there anything in zigbee2mqtt itself which can be accessed and only represents a state change for a topic value?

No, you have to do this in your automation

diagonali, 2022-06-06:
> 

Ok thanks @Koenkk 

For anyone else who finds this, I managed to account for the duplicate message from my motion sensor (I use Node Red) by setting a flow variable which is updated by all and any messages from any motion sensors, set to the current "occupancy" value. I then reference this value from within a function node which decides how to respond and only allow triggers if it changes from false > true. Would be happy to share if needed.

Koenkk, 2022-06-07:
Cool, asuming this can be closed now.

closed 2022-06-07 by Koenkk

lich2000117, 2023-11-26:
Can confirm, my motion sensor does the same thing. It is probably a bug from the device itself.