Koenkk/zigbee2mqtt #12678: No Illuminance entity on home assistant for the new TuYa ZG-204ZL

rosaldanha, 2022-06-01:
### What happened?

The Illuminance values, for device TuYa ZG-204ZL, are not showing on home assistant, there is no entity to show it. All the others values (occupancy, battery and others) are ok.

Thanks for the fast implementation of this end device.

### What did you expect to happen?

I want to see the Illuminance values on home assistant, they are already showing into zigbee2mqtt.

### How to reproduce it (minimal and precise)

1) install the device TuYa ZG-204ZL on zigbee2mqtt
2) after that you will see only this 4 related entities at home assistant (where mlfb87sala is the device name):

sensor.mlfb87sala_battery
select.mlfb87sala_keep_time
binary_sensor.mlfb87sala_occupancy
select.mlfb87sala_sensitivity

There is no "Illuminance" entity. 

### Zigbee2MQTT version

1.25.1-dev commit: 4481543

### Adapter firmware version

20220219

### Adapter

zStack3x0

Pferdebockwurst, 2022-06-06:
Off-topic: where did you buy these, can you provide a link?

rosaldanha, 2022-06-06:
Hello ! Off course:  https://www.aliexpress.com/item/3256803771958581.html?spm=a2g0o.order_list.0.0.3ceecaa4HofuMT&gatewayAdapt=4itemAdapt

rosaldanha, 2022-06-20:
After updating to version 1.25.2-dev commit: [820c20a] is working, I just have to enable the entity. Thanks a lot !

o0Scope0o, 2023-06-16:
> After updating to version 1.25.2-dev commit: [[820c20a](https://github.com/Koenkk/zigbee2mqtt/commit/820c20af806ba2b43c90d09dd9ffe7ca7b930fe2)] is working, I just have to enable the entity. Thanks a lot !

Dude, happy you fixed this, but sorry to say im having the exact same issue, could you please help me do the same? Im a bit lost tbh.

o0Scope0o, 2023-06-16:
> Dude, happy you fixed this, but sorry to say im having the exact same issue, could you please help me do the same? Im a bit lost tbh.

All sorted now thanks, figured out what I needed to do based on your post! Appreciate it!! :D

Volkir, 2023-07-24:
> After updating to version 1.25.2-dev commit: [[820c20a](https://github.com/Koenkk/zigbee2mqtt/commit/820c20af806ba2b43c90d09dd9ffe7ca7b930fe2)] is working, I just have to enable the entity. Thanks a lot !

Hi. Can you tell me how you "enable the entity"?

Gil80, 2023-08-27:
@Volkir did you end up enabling the entity? I have the same problem.

Volkir, 2023-09-08:
> Hi Gil80
I finally enabled this entity...
Check here -> https://github.com/Koenkk/zigbee2mqtt/discussions/18631

lich2000117, 2024-03-09:
Came here to say thank you! We need to specifically enable that entity to make it work😁