Koenkk/zigbee2mqtt #12686: Xiaomi door/window sensor stopped exposing temperature

Mithrox, 2022-06-02:
### What happened?

This morning I upgraded to HA 2022.6 and Zigbee2mqtt 1.25.2-1 and after upgrade my Xiaomi MCCGQ11LM all stopped reporting on temperature. Device_temperature just shows N/A and it does not help to trigger communication through opening/closing the sensor either. 

[image]

I am not sure what is the firmware version of the Conbee2, but I have pasted in the info I guess through z2m interface.

[image]

### What did you expect to happen?

Continue working as before update

### How to reproduce it (minimal and precise)

Update to newest version and then restart Zigbee2mqtt, the issue comes instantly and for all devices.

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

0x26720700

### Adapter

Conbee2

Otnow, 2022-06-02:
Xiaomi/Aqara battery-powered sensors report service information (including `device_temperature`) once _every ~55 minutes_, so you just need to wait.

To get the value immediately, you need to _press the pairing button once_.

Mithrox, 2022-06-02:
Hello.

I did try to trigger dem by pressing the button, and I also triggered update to the open/close by normal means. Weirdly enough, one of them is now reporting, but the rest of them (I have 5) is still not reporting after pressing button and waiting several hours. And when I have updated both HA and Z2M before they have been ready within 30 seconds (they are white first, then comes online normally with correct state).

Otnow, 2022-06-02:
Please write a [debug log](https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging) for the problematic sensor, where will be visible messages at the moment the pairing button is pressed once.

Mithrox, 2022-06-03:
I have enabled debug log in Z2M and I have pressed the buttons, repeatedly on the device (one of them) that do not show temperature as they should, but nothing happens in the debug log. A lot of the other Zigbee devices are showing entries, but it does not do anything when I press the button and the blue light blinks.

It seems clear some of the devices are lost without a route, this did happen after the upgrade (look at the image). Is there a way to force these to get back with their "path-devices" without having to remove them completely and include them back in?

[image]

Mithrox, 2022-06-03:
Ok, so I did enable join and did the procedure for join without removing them first, and all nodes are now back in and reporting. I will close this case as I have resolved my issue, but it would be good to know if this was something isolated to my system, or if others experienced the same. It seems there was something that kicked out at least 5 nodes in my setup, I believe it was more but somehow some of them found it's way back in.

Anyways, thanks for the input, sorry for the trouble.

Mithrox, 2022-06-03:
As a little extra information. I also found out that some of my Smart plugs had to be rejoined as well, they would just turn off or on once, and then they hung. So now I have more or less rejoined everything, and it looks to be working as it should. Prior to this I also restarted the NUC that runs HA completely to make sure it was power cycled, but something weird definitely did happen during upgrade.

SaturnusDJ, 2022-12-16:
Just now I see that after upgrading the zzh CC2652R stick and/or after upgrading Z2M to 1.25.2, the door/window sensors in my setup also stopped reporting temperature. Updating Z2M to latest now. Re-pair I can only do 2 weeks from now.