Koenkk/zigbee2mqtt #12687: Aqara wall switches (with neutra) - missing "energy" in HA

quarcko, 2022-06-02:
### What happened?

Aqara light wall switches previosly had consumtion / energy values.
I can still see the values in Z2M control panel inside "state" tab of the devices.
But entity is missing and not populated into HA?

Do i miss anything?

Here is "energy":
[image]

But not "exposed":
[image]

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

latest

### Adapter

zig-a-zig-ah!

Koenkk, 2022-06-02:
Please provide the model number of this switch

quarcko, 2022-06-14:
Hi,

sorry for a delay, i have 2 types (single and double ones):
[QBKG11LM]
[QBKG12LM]

And i for sure remmeber some time ago "consumtion" or "energy" attributes (or entities) were available for these switches.

Koenkk, 2022-06-14:
Added energy for both now, consumption is deprecated (and the same value as energy) so I didn't expose that.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gerald732, 2023-12-16:
hi @Koenkk, +1 to @donghyun-lees. Can we also include QBKG26LM and QBKG24LM too please? `Energy` is present for the single gang, QBKG23LM, but not the 2 and 3 gangs switches.