Koenkk/zigbee2mqtt #12688: [Feature request]: Aqara T1 smart LED bulb (ZNLDP13LM) - add support for the `power_outage_memory` option

gorbunovav, 2022-06-02:
### Is your feature request related to a problem? Please describe

According to reviews of this bulb with miHome / AqaraHome apps it is possible to configure the power outage memory behaviour (at least for the `lumi.light.acn014` modification).

The way it was handled for `ZNLDP12LM` doesn't work for `ZNLDP13LM`, so probably the command has changed.

### Describe the solution you'd like

Figure out the new command and add the config option to the device converter.

### Describe alternatives you've considered

None

### Additional context

The feature is important, especially if you have a lot of these bulbs in the house 🤔

gorbunovav, 2022-06-29:
Sniffed packets

Turn-off:
```
Destination Endpoint: 1
Cluster: Manufacturer Specific (0xfcc0)
Command: Unknown (0x01)
Data (6 bytes)
    Data: 050302050100
    [Length: 6]
```

Turn-on:
```
Destination Endpoint: 1
Cluster: Manufacturer Specific (0xfcc0)
Command: Unknown (0x01)
Data (6 bytes)
    Data: 050302050101
    [Length: 6]
```

gorbunovav, 2022-06-29:
@Koenkk could you advise how to proceed?

Koenkk, 2022-07-01:
- Add the command [here](https://github.com/Koenkk/zigbee-herdsman/blob/359279ab2c14e4f9a7e8f5db383e63785b4f70d3/src/zcl/definition/cluster.ts#L4376):

```js
        commands: {
            command1: {
                ID: 1,
                parameters: [
                    {name: 'data', type: BuffaloZclDataType.BUFFER},
                ],
            },
        },
```

- Then send the command in toZigbee with the following code:

```js
let enabled = true;
await entity.command('aqaraOpple', 'command1', {data: Buffer.from([0x05, 0x03, 0x02, 0x05, 0x01, enabled ? 0x01 : 0x00])});
```

gorbunovav, 2022-07-02:
@Koenkk thank you!

I was able to send the command (I've also added `manufacturerOptions.xiaomi` as a third parameter).

But it doesn't work 🤔

My current observations:

1. I was sniffing commands from the `Xiaomi Gateway` (`ZNDMWG02LM`) and `MiHome` app in Mainland China region.
2. I've noticed that there are no normal commands in the sniffed traffic. So e.g. for the on/off I see the same `0x01` command with a bit different payload (`0x05, 0x02, 0x01, 0x05, 0x01, 0x01`). I've tried to send it from the `Z2M`, but there was no reaction also.
3. I was able to pair the bulb with the `Tuya Smart Hub` - I see regular commands there, but there is no `power memory` config option sadly.
4. If the bulb is paired with `Z2M` I see normal commands. Attempt to send this `0x01` command yields strange response from the device with the status `Ota Abort (0x95)`.
5. I've compared sent commands details in Wireshark - I see only small difference on the `ZigBee Application Support Layer Data` level -> `Frame Control Field` -> `Acknowledgement Request` (it is enabled in the original command and is `false` in `Z2M`)

My guess is that maybe the bulb is pairing in some different mode with the `Xiaomi Gateway` and uses some different command set?

Koenkk, 2022-07-02:
What coordinator and firmware are you running?

gorbunovav, 2022-07-02:
@Koenkk it is a USB CC2652P Stick (https://aliexpress.ru/item/1005002823262979.html) with zStack3x0 20220219

Koenkk, 2022-07-02:
To be sure, did you repair the device after upgrading to this firmware? This will trick the device thinking it is connected to a Xiaomi gateway.

gorbunovav, 2022-07-02:
@Koenkk just tried to re-pair it again. The behavior is still the same.

Otnow, 2022-07-04:
I have a suspicion that the _Xiaomi_ and _Aqara_ hubs use different commands, and since. when paired with _z2m_, the device thinks that it is connected to the _Aqara_ hub, then it probably does not accept commands intended for the mode of operation with the _Xiaomi_ hub.

Therefore, I suggest trying to use the `[aqaraOpple] 0x0201` attribute in `tz.xiaomi_switch_power_outage_memory`, changing it by adding the light bulb model to the appropriate conditions:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/052f2536d53c1571313361a48a12e56ec31c94af/converters/toZigbee.js#L2165-L2201

gorbunovav, 2022-07-05:
@Otnow Bingo! It works for both `lumi.light.cwac02` and `lumi.light.acn014`.
Thank you!

I will prepare PR tomorrow.

closed 2022-07-07 by Koenkk