Koenkk/zigbee2mqtt #12694: Following #11479, OnOff cluster is not available for bind

IsThisUsernameFree, 2022-06-02:
### What happened?

After pairing a TS0001 (_TZ3000_5ng23zjs), I'm not able to bind the OnOff cluster (it's not available). Even if I unbind it from the coordinator.

[image]

### What did you expect to happen?

OnOff cluster is available to sync the module with the LED bulb

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20210708

### Adapter

Sonoff ZB-P

IsThisUsernameFree, 2022-06-02:
Filling this issue allowed me to spot that my firmware was outdated. 

Coordinator version 20220219, not working either.

Koenkk, 2022-06-02:
> OnOff cluster is available to sync the module with the LED bulb

What do you mean with "available"?

IsThisUsernameFree, 2022-06-02:
I meant available to select to perform the bind

Koenkk, 2022-06-03:
Can you provide the data/database.db entry of the TS0001 ?

IsThisUsernameFree, 2022-06-03:
Here it is 

{"id":24,"type":"Router","ieeeAddr":"0xcc86ecfffe0d9415","nwkAddr":59149,"manufId":4098,"manufName":"_TZ3000_5ng23zjs","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TZ3000_5ng23zjs","stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e33321","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1654246868493,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-06-04:
If I understand correctly, you want to control the light bulb directly from the switch. This switch does not support that because `genOnOff` (6) is not an output cluster of this device. (only input)

IsThisUsernameFree, 2022-06-04:
Yes I would've liked to ignore the relay of the module and use it as a switch (in fact, I wanted to use it's capability to wire a dumb switch and bind it to control the lights without requiring any automation system)

Koenkk, 2022-06-04:
This is not possible with this switch.

closed 2022-07-13 by github-actions[bot]