Koenkk/zigbee2mqtt #12697: No new tagged Docker image on ghcr.io/koenkk/zigbee2mqtt

Dennis14e, 2022-06-03:
### What happened?

I use [ghcr.io/koenkk/zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt/pkgs/container/zigbee2mqtt) as Docker image source and the latest stable release 1.25.2 was not uploaded/tagged there.

### What did you expect to happen?

A new 1.25.2 tagged version of zigbee2mqtt.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

-

### Adapter

-

Koenkk, 2022-06-03:
Latest images are currently not available on ghcr due to an issue with ghcr itself.

- Build log: https://github.com/Koenkk/zigbee2mqtt/runs/6709993727?check_suite_focus=true
- https://github.community/t/buildx-failed-with-error-cannot-reuse-body-request-must-be-retried/253178/2
- https://github.com/felddy/foundryvtt-docker/issues/362

I will periodically check if it works again.

onedr0p, 2022-06-07:
I believe GHCR is working now, @Koenkk can you retry the build?

Koenkk, 2022-06-08:
@onedr0p thanks for letting me know! Images are available now.

duckfullstop, 2022-06-09:
I seem to be getting an issue pulling the amd64 version of 1.25.2 on containerd:

```
Failed to pull image "ghcr.io/koenkk/zigbee2mqtt:1.25.2": rpc error: code = NotFound desc = failed to pull and unpack image "ghcr.io/koenkk/zigbee2mqtt:1.25.2": failed to copy: httpReadSeeker: failed open: content at https://ghcr.io/v2/koenkk/zigbee2mqtt/manifests/sha256:63812d7a77ad6ab414cfb597ae8934dcb9d1bc665515b32db22b56b5298a6249 not found: not found
```

The `1.25.2` tag seems to be resolving to the `linux/amd64` manifest as it should, but for whatever reason the manifest tag itself is not present on GHCR (pulling the manifest explicitely with that sha256 (which is the same as the one on the Packages page) also doesn't work). Am I going insane here? Is this working for everyone else?

Dennis14e, 2022-06-09:
I can at least replicate the following on arm64 @duckfullstop :
```
root@rpi4-1:~# docker pull ghcr.io/koenkk/zigbee2mqtt:1.25.2
1.25.2: Pulling from koenkk/zigbee2mqtt
manifest unknown

root@rpi4-1:~# docker pull koenkk/zigbee2mqtt:1.25.2
1.25.2: Pulling from koenkk/zigbee2mqtt
9981e73032c8: Already exists
20503679020f: Pull complete
e634e75aa9f0: Pull complete
affaf0df33eb: Pull complete
87eab9f79b9e: Pull complete
988f8f018472: Pull complete
485876a066a9: Pull complete
90c4261ed7a1: Pull complete
56c4e56d97f3: Pull complete
58194e0ea353: Pull complete
29b50d92ae29: Pull complete
bb4e242cd83b: Pull complete
80b3f1c9c36c: Pull complete
39beee936da9: Pull complete
Digest: sha256:94b3a150bb3733cda15ffa3541eea75fe43d8882573b09d8b294930be3975fca
Status: Downloaded newer image for koenkk/zigbee2mqtt:1.25.2
docker.io/koenkk/zigbee2mqtt:1.25.2
```

So something seems to be different with GHCR compared to Docker Hub.

onedr0p, 2022-06-09:
I was able to pull it yesterday, maybe GHCR is having issue.

Koenkk, 2022-06-11:
GHCR is indeed acting strange lately, the `linux/arm64` image is there for sure: https://github.com/Koenkk/zigbee2mqtt/pkgs/container/zigbee2mqtt/24620318?tag=1.25.2

Koenkk, 2022-06-15:
@mitsumaui can you check if it works now?