Koenkk/zigbee2mqtt #12699: Perenio PEHPL0X "not support"

drako86, 2022-06-03:
### What happened?

Hi ! after the last update, I saw that finally this socket was put on the list of supported devices ... so I took it back from the drawer and tried to pair it ... but again, it tells me that it is not supported ... Help me pls !

### How to reproduce it (minimal and precise)

I tried to :
1) update the coordinator firmware
2) update zigbee2mqtt from scratch
3) restarted every machine on the network

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

20220219

### Adapter

Z-Bee Duo (eth)

### Debug log

Info 2022-06-03 13:03:16Device '0xf4ce364f116817e5' joined
Info 2022-06-03 13:03:16MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4ce364f116817e5","ieee_address":"0xf4ce364f116817e5"},"type":"device_joined"}'
Info 2022-06-03 13:03:16MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xf4ce364f116817e5"},"type":"device_connected"}'
Info 2022-06-03 13:03:16Starting interview of '0xf4ce364f116817e5'
Info 2022-06-03 13:03:16MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4ce364f116817e5","ieee_address":"0xf4ce364f116817e5","status":"started"},"type":"device_interview"}'
Info 2022-06-03 13:03:16MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xf4ce364f116817e5"},"type":"pairing"}'
Debug 2022-06-03 13:03:16Device '0xf4ce364f116817e5' announced itself
Info 2022-06-03 13:03:16MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4ce364f116817e5","ieee_address":"0xf4ce364f116817e5"},"type":"device_announce"}'
Info 2022-06-03 13:03:16MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xf4ce364f116817e5"},"type":"device_announced"}'
Debug 2022-06-03 13:03:52Device '0xf4ce364f116817e5' announced itself
Info 2022-06-03 13:03:52MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4ce364f116817e5","ieee_address":"0xf4ce364f116817e5"},"type":"device_announce"}'
Info 2022-06-03 13:03:52MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xf4ce364f116817e5"},"type":"device_announced"}'
Error 2022-06-03 13:05:34Failed to interview '0xf4ce364f116817e5', device has not successfully been paired
Info 2022-06-03 13:05:34MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xf4ce364f116817e5","ieee_address":"0xf4ce364f116817e5","status":"failed"},"type":"device_interview"}'
Info 2022-06-03 13:05:34MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0xf4ce364f116817e5"},"type":"pairing"}'

Koenkk, 2022-06-04:
`Error 2022-06-03 13:05:34Failed to interview '0xf4ce364f116817e5', device has not successfully been paired` -> interview failed, try to pair the device close to the coordinator.

drako86, 2022-06-04:
hello, most likely it is my socket that has problems (or is it the low quality socket) it connects and creates 1000 more errors it detaches from the coordinator ... I throw it and I get over it!

closed 2022-06-04 by drako86