Koenkk/zigbee2mqtt #12701: A light used to work, now is unsupported

auanasgheps, 2022-06-03:
### What happened?

Hi, I have purchased a Zigbee bulb from a local electric store and found out was supported, being recognized as [this one](https://www.zigbee2mqtt.io/devices/50064.html) and worked perfectly well for months.

This until some days ago: the bulb has changed manufacturer, model and now shows as UNSUPPORTED.

I have deleted the bulb and re-paired it, still shows up as unsupported. Doesn't make any sense to me. 

The "Paulmann 50064" model albeit being generic worked perfectly fine. 

[image]

I checked the logs the Addon does save in `/config/zigbee2mqtt/log` and the issue started to appear when Zigbee2MQTT got updated to 1.25.2.
As soon as communication with devices was initiated after the update, this appeared

```log
warn  2022-06-02 08:50:56: Received message from unsupported device with Zigbee model 'CCT Light' and manufacturer name 'Astuta/ZB-CCT'
warn  2022-06-02 08:50:56: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

My setup:
- Raspberry Pi4 4GB
- Home Assistant 2022.5.5 with HASS OS 
- SSD with long extension cable
- Raspbee2

### What did you expect to happen?

Surely not a manufacturer and model change.
The bulb was working perfectly fine up to 1.25.1

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

0x26720700

### Adapter

RaspBee2

### Debug log

As of now I'm having trouble to complete pairing for this device. Looks like is unable to complete the interview.
Before opening this case I was able to re-pair it multiple times (screenshot above)

[z2mlog.txt](https://github.com/Koenkk/zigbee2mqtt/files/8832428/z2mlog.txt)

Koenkk, 2022-06-04:
Could you provide the data/database.db entry of this device? Then I can re-add support for it.

auanasgheps, 2022-06-04:
Hi @Koenkk can you please help me where I can find it? Since I'm using the add-on on a Supervised HA installation, I don't know where to look for that file.

auanasgheps, 2022-06-04:
Nevermind, found it.

```
{"id":4,"type":"Router","ieeeAddr":"0x00158d00012f3c8a","nwkAddr":60687,"manufId":4137,"manufName":"Astuta/ZB-CCT","modelId":"CCT Light","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,4,3,6,8,5,768,65535,65535,4096,4111],"outClusterList":[25,4111],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genBasic":{"attributes":{"modelId":"CCT Light","manufacturerName":"Astuta/ZB-CCT","zclVersion":3,"appVersion":1,"stackVersion":1,"hwVersion":21,"swBuildId":"00.00.01"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":21,"swBuildId":"00.00.01","zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1654334876120,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-06-04:
Added it back now (it will be recognized as a Ecodim ED-10041 since this looks to be a whitelabel of it)
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

If you don't want to switch to the dev branch you can use this external converter until the next release:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'CCT Light', manufacturerName: 'ZigBee/CCT', manufacturerID: 4137},
        {modelID: 'CCT Light', manufacturerName: 'Astuta/ZB-CCT', manufacturerID: 4137}],
    model: 'ED-10041',
    vendor: 'EcoDim',
    description: 'Zigbee LED filament light dimmable E27, edison ST64, flame 2200K',
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}),
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

closed 2022-06-04 by Koenkk

nagi121, 2022-06-05:
Hi,
I have the same problem with "CCT light", but with other manufacturer. This is the database entry of my device.

`{"id":2,"type":"Router","ieeeAddr":"0x00158d0003147f91","nwkAddr":19226,"manufId":4509,"manufName":"Paulmann Licht GmbH ","powerSource":"Mains (single phase)","modelId":"CCT light ","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,4,3,6,8,5,768,4096],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160301","swBuildId":"1218-0909","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1654440090113,"defaultSendRequestWhen":"immediate"}`

Several products exist on the market with this controller, branded by different manufacturer. Paulmann seems to be the original manufacturer of this controller.

This is the homepage of the controller:

[Paulmann 500.46](https://www.lightkontor.de/Paulmann_No_50046_SmartHome_Zigbee_MaxLED_Tunable_White_Controller_Weiss.html?gclid=Cj0KCQjwqPGUBhDwARIsANNwjV5iUz9Blm61-dSC4u4cfRkCamfqRMIjpbyRugpDPBUrrWPVKOQBzLAaAu7AEALw_wcB)

I tried the external converter @Koenkk has posted above, but with no success. I also tried to adapt the converter to my values, but also without success.

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'CCT Light', manufacturerName: 'Paulmann Licht GmbH', manufacturerID: 4509},
        {modelID: 'CCT Light ', manufacturerName: 'Paulmann Licht GmbH ', manufacturerID: 4509}
    ],
    model: '500.46',
    vendor: 'Paulmann',
    description: 'MaxLED 500 Controller Smart Home Zigbee Tunable White DC 24V',
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [200, 370]}),
};

module.exports = definition;
`

What to do know?

nagi121, 2022-06-05:
Okay, I got it. Sometimes it helps to respect the upper/lower writing.

After changing 'CCT Light' to ' CCT light' everything works as expected.

Thanks anyway.