Koenkk/zigbee2mqtt #12702: [Feature request]: Default transition period for groups

RubenKelevra, 2022-06-03:
### Is your feature request related to a problem? Please describe

I have three zigbee devices in a group in Zigbee2MQTT, all are configured to transition 4 seconds.

They do this fine when individually addressed, but when I "toggle" the group in HA I get an instant response from them.

### Describe the solution you'd like

I just want always my 4 seconds transition, but for the whole group. Is that possible protocol wise? 🤔 

### Describe alternatives you've considered

Since toggle in HA doesn't allow to set a transition, I would need to complicate things by using an if regarding the current state, which is unnecessarily complicated. 

### Additional context

None

Koenkk, 2022-06-04:
Did you set the `transition` for the group in the `configuration.yaml`?

RubenKelevra, 2022-06-04:
> Did you set the `transition` for the group in the `configuration.yaml`?

Well, I don't configure Zigbee2MQTT in the config files, but via GUI. If that's an option, it's currently not exposed to the GUI, I guess :)

Koenkk, 2022-06-05:
Does it work when you set it in the `configuration.yaml`?

closed 2022-07-14 by github-actions[bot]