Koenkk/zigbee2mqtt #12710: Issue with PIR model NAS-PD07

patrick-foubert-56770, 2022-06-04:
### What happened?

Hello, 

This moorning i've 2 devices (NAS-PD07) where model has changed into zigbeemqtt is show as alarm model  NAS-AB02B2.
I remove them and add an spare (never added to my config) is also detected as NAS-AB02B2

### What did you expect to happen?

No idea.... DB corrupt ?
Mqtt / ha restarted, same issue

### How to reproduce it (minimal and precise)

Just try pairing again, same issue 

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Debug log

Log file for pairing: 

info 2022-06-04 10:32:50Device '0xa4c13808ff9ca3a0' joined
info 2022-06-04 10:32:51MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13808ff9ca3a0","ieee_address":"0xa4c13808ff9ca3a0"},"type":"device_joined"}'
info 2022-06-04 10:32:51MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c13808ff9ca3a0"},"type":"device_connected"}'
info 2022-06-04 10:32:51Starting interview of '0xa4c13808ff9ca3a0'
info 2022-06-04 10:32:51MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13808ff9ca3a0","ieee_address":"0xa4c13808ff9ca3a0","status":"started"},"type":"device_interview"}'
info 2022-06-04 10:32:51MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c13808ff9ca3a0"},"type":"pairing"}'
debug 2022-06-04 10:32:52Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0202"}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:52Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_kmh5qpmb"}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:53Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:53Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:54Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'genBasic', data '{"appVersion":67}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:54Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:55Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:55Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:55Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:55Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:56Received Zigbee message from '0xa4c13808ff9ca3a0', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:56Received Zigbee message from '0xa4c13808ff9ca3a0', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID null
debug 2022-06-04 10:32:57Received Zigbee message from '0xa4c13808ff9ca3a0', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00212effff07e634","zoneState":1}' from endpoint 1 with groupID null
info 2022-06-04 10:32:57Successfully interviewed '0xa4c13808ff9ca3a0', device has successfully been paired
info 2022-06-04 10:32:57Device '0xa4c13808ff9ca3a0' is supported, identified as: Neo Alarm (NAS-AB02B2)
info 2022-06-04 10:32:57MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Alarm","exposes":[{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"alarm","property":"alarm","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"melody","property":"melody","type":"enum","values":["1","2","3","4","5","6","7","8","9","10","11","12","13","14","15","16","17","18"]},{"access":3,"name":"duration","property":"duration","type":"numeric","unit":"second","value_max":1800,"value_min":0},{"access":3,"name":"volume","property":"volume","type":"enum","values":["low","medium","high"]},{"access":1,"name":"battpercentage","property":"battpercentage","type":"numeric","unit":"%"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"NAS-AB02B2","options":[],"supports_ota":false,"vendor":"Neo"},"friendly_name":"0xa4c13808ff9ca3a0","ieee_address":"0xa4c13808ff9ca3a0","status":"successful","supported":true},"type":"device_interview"}'
info 2022-06-04 10:32:58Configuring '0xa4c13808ff9ca3a0'

patrick-foubert-56770, 2022-06-04:
Additional info for this pir device (neo coolcam):  Zigbee manufacter:   _TZ3000_kmh5qpmb

Koenkk, 2022-06-04:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_7hfcudw5'}, {modelID: 'TS0202', manufacturerName: '_TZ3000_kmh5qpmb'}],
    model: 'NAS-PD07',
    vendor: 'Neo',
    description: 'Motion, temperature & humidity sensor CUSTOM',
    fromZigbee: [fz.neo_nas_pd07, fz.ignore_tuya_set_time],
    toZigbee: [tz.neo_nas_pd07],
    onEvent: tuya.onEventSetTime,
    exposes: [e.occupancy(), e.humidity(), e.temperature(), e.tamper(), e.battery_low(),
        exposes.enum('power_type', ea.STATE, ['battery_full', 'battery_high', 'battery_medium', 'battery_low', 'usb']),
        exposes.enum('alarm', ea.STATE, ['over_temperature', 'over_humidity', 'below_min_temperature', 'below_min_humdity', 'off'])
            .withDescription('Temperature/humidity alarm status'),
        exposes.numeric('temperature_min', ea.STATE_SET).withUnit('°C').withValueMin(-20).withValueMax(80),
        exposes.numeric('temperature_max', ea.STATE_SET).withUnit('°C').withValueMin(-20).withValueMax(80),
        exposes.binary('temperature_scale', ea.STATE_SET, '°C', '°F').withDescription('Temperature scale (°F/°C)'),
        exposes.numeric('humidity_min', ea.STATE_SET).withUnit('%').withValueMin(1).withValueMax(100),
        exposes.numeric('humidity_max', ea.STATE_SET).withUnit('%').withValueMin(1).withValueMax(100),
        // exposes.binary('unknown_111', ea.STATE_SET, 'ON', 'OFF').withDescription('Unknown datapoint 111 (default: ON)'),
        // exposes.binary('unknown_112', ea.STATE_SET, 'ON', 'OFF').withDescription('Unknown datapoint 112 (default: ON)')
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await endpoint.command('manuSpecificTuya', 'mcuVersionRequest', {'seq': 0x0002});
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

patrick-foubert-56770, 2022-06-04:
Hi, first try error, second, it's works.... i can't understand why i successfully paired them and config change after :(

alb1981, 2022-06-05:
The device was correctly identified. But it does not detect movement.

patrick-foubert-56770, 2022-06-06:
Hi same issue, detected but in zigbeemqtt:

occupancy
Indique si l'appareil a détecté une occupation (pir detection)
N/A

If i expose PIR detector, nothing in logs

patrick-foubert-56770, 2022-06-06:
Perhaps better info:

debug 2022-06-06 11:45:39Received Zigbee message from 'capteur_pir_studio', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID null
debug 2022-06-06 11:45:39No converter available for 'NAS-PD07' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'

patrick-foubert-56770, 2022-06-06:
Perhaps more info for a solution...

Log result.... if i expose pir to body and leave after without exposition

debug 2022-06-06 11:50:04Received Zigbee message from 'capteur_pir_studio', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID null
debug 2022-06-06 11:51:04Received Zigbee message from 'capteur_pir_studio', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID null
debug 2022-06-06 11:51:12Received Zigbee message from 'capteur_pir_studio', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID null

zonestatus seems corresponding to "expose" ?

Tested and confirmed on a second device, same response

patrick-foubert-56770, 2022-06-10:
Hello, daily update @Koenkk , on a fresh install (VM), same adapter (Conbee 2), same issue.

I try with ZHA add on, it's works, correctly detected, pir motion works also...

Koenkk, 2022-06-11:
Can you try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_7hfcudw5'}, {modelID: 'TS0202', manufacturerName: '_TZ3000_kmh5qpmb'}],
    model: 'NAS-PD07',
    vendor: 'Neo',
    description: 'Motion, temperature & humidity sensor CUSTOM',
    fromZigbee: [fz.neo_nas_pd07, fz.ignore_tuya_set_time, fz.ias_occupancy_alarm_1],
    toZigbee: [tz.neo_nas_pd07],
    onEvent: tuya.onEventSetTime,
    exposes: [e.occupancy(), e.humidity(), e.temperature(), e.tamper(), e.battery_low(),
        exposes.enum('power_type', ea.STATE, ['battery_full', 'battery_high', 'battery_medium', 'battery_low', 'usb']),
        exposes.enum('alarm', ea.STATE, ['over_temperature', 'over_humidity', 'below_min_temperature', 'below_min_humdity', 'off'])
            .withDescription('Temperature/humidity alarm status'),
        exposes.numeric('temperature_min', ea.STATE_SET).withUnit('°C').withValueMin(-20).withValueMax(80),
        exposes.numeric('temperature_max', ea.STATE_SET).withUnit('°C').withValueMin(-20).withValueMax(80),
        exposes.binary('temperature_scale', ea.STATE_SET, '°C', '°F').withDescription('Temperature scale (°F/°C)'),
        exposes.numeric('humidity_min', ea.STATE_SET).withUnit('%').withValueMin(1).withValueMax(100),
        exposes.numeric('humidity_max', ea.STATE_SET).withUnit('%').withValueMin(1).withValueMax(100),
        // exposes.binary('unknown_111', ea.STATE_SET, 'ON', 'OFF').withDescription('Unknown datapoint 111 (default: ON)'),
        // exposes.binary('unknown_112', ea.STATE_SET, 'ON', 'OFF').withDescription('Unknown datapoint 112 (default: ON)')
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await endpoint.command('manuSpecificTuya', 'mcuVersionRequest', {'seq': 0x0002});
    },
};

module.exports = definition;
```

patrick-foubert-56770, 2022-06-11:
Thanks, gracias, merci... bedankt... :)) It's works

Koenkk, 2022-06-11:
Do the other features also work? (e.g. `alarm`, `temperature_max`)

patrick-foubert-56770, 2022-06-17:
Hi Koen,

My model has only PIR / Battery sensor, temperature and alarm are optionals 
;(

I try to order one from same seller with tamper / temperature for testing

Patrick Foubert
2 Locmaria
56770 Plouray

https://www.3cis.fr
Mobile: +33 6 24 85 15 27
Immatriculé au RCS de Lorient
N° Siren: 893 012 732
N° Siret:  893 012 732 00010
TVA: FR 58893012732

#12710)

sihui62, 2022-07-02:
Same issue here, NAS-PD07 is detected as NAS-AB02B2.
External converter from above
https://github.com/Koenkk/zigbee2mqtt/issues/12710#issuecomment-1152875551
fixed it partly.

> Do the other features also work? (e.g. alarm, temperature_max)
My NAS-PD07 has temperature and humidity (at least according to the manual), but only
occupancy, 
tamper, 
battery_low, 
last_seen and 
linkquality
are shown via external converter.

Z2M version: 1.26.0-dev latest from today
Adapter: ITead Sonoff Zigbee 3.0 USB Dongle Plus

Koenkk, 2022-07-02:
Great, pushed the fix now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

heggink, 2022-07-11:
> Thanks, gracias, merci... bedankt... :)) It's works

Hey Patrick,
Did you do anything special to make it work? I just got 2 of these (without the temp/hum). Using the latest dev branch, they get recognised correctly but no PIR motion is being detected :-(
H

closed 2022-08-18 by github-actions[bot]

denisov-vlad, 2022-10-04:
Having the same issue, PIR sensor stopped working both via original and dev addons.

[image]

wolph, 2022-12-13:
The suggested fix in https://github.com/Koenkk/zigbee2mqtt/issues/12710#issuecomment-1152875551 does work for occupancy. Temperature and humidity are not working for me however.

My sensor should have those, but it was ordered from AliExpress I'm not a 100% certain about that ;)

10der, 2023-06-02:
btw, got 5 pcs with temperature and humidity sensors (double check with dissemble module!)
no  temperature and humidity