Koenkk/zigbee2mqtt #12711: Discrepance between home assistant configuration tab and configuration.yaml

jlo88, 2022-06-04:
### What happened?

I'm trying to understand the connection between configuration.yaml and the home assistant addon configuration tab. When I make changes in the home assistant configuration tab and save it I don't see the updates in the yaml file and vise versa. Which one is leading?

### What did you expect to happen?

home assistant addon configuration tab to have the same content as configuration.yaml

### How to reproduce it (minimal and precise)

Configuration.yaml:
```
{
  "external_converters": [],
  "devices": ["devices.yaml"],
  "groups": ["groups.yaml"],
  "homeassistant": true,
  "permit_join": false,
  "mqtt":
    {
      "base_topic": "zigbee2mqtt",
      "server": "mqtt://core-mosquitto",
      "user": "XXX",
      "password": "XXX",
    },
  "serial": { "port": "/dev/ttyACM0" },
  "advanced":
    {
      "log_level": "info",
      "pan_id": 6753,
      "channel": 11,
      "network_key": [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13],
      "availability_blocklist": [],
      "availability_passlist": [],
    },
  "device_options": {},
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "frontend": { "port": 8099 },
  "experimental": {},
}
```

home assistant addon in yaml: 

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto
  user: XXX
  password: XXX
serial:
  port: /dev/ttyACM0
```

Many options from the configuration.yaml are not visible in the addon. When I make a change in the addon the file on the disk is not updated and the other way around. Removing configuration.yaml results in an error after restarting and recreating the file. So I know I'm looking at the correct file. 

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20191106

### Adapter

20191106

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8837509/log.txt)

Koenkk, 2022-06-04:
With the release of the 1.25.0-1 addon, only the serial and MQTT configuration from the HA dashboard are synced to the z2m `configuration.yaml`. The z2m `configuration.yaml` is leading and can also be updated via the z2m frontend.

jlo88, 2022-06-04:
Okay, that makes sense. Does this also work the other way around? For example now I changed the yaml file mqtt settings but when I look in the addon the changes are not visible.

Koenkk, 2022-06-05:
> Does this also work the other way around? For example now I changed the yaml file mqtt settings but when I look in the addon the changes are not visible.

No, only serial/MQTT is synced to z2m configuration.yaml, nothing is synced back

> The z2m frontend also doesn't show the changes. How do I know they are applied?

Then they are not applied.

callacomp, 2022-06-06:
What about herdsman debug option?

jlo88, 2022-06-06:
@callacomp You mean with the debug option on? I've attached another log.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8846716/log.txt)
 
@Koenkk Maybe a bit more background. I'm trying to use a secret.yaml file. I've placed it next to the config yaml file. Config file: 

```
{
  "external_converters": [],
  "devices": ["devices.yaml"],
  "groups": ["groups.yaml"],
  "homeassistant": true,
  "permit_join": false,
  "mqtt":
    {
      "base_topic": "zigbee2mqtt",
      "server": "mqtt://core-mosquitto",
      "user": "!secret mqtt_user",
      "password": "!secret mqqt_password",
    },
  "serial": { "port": "/dev/ttyACM0" },
  "advanced":
    {
      "log_level": "info",
      "pan_id": 6753,
      "channel": 11,
      "network_key": [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13],
      "availability_blocklist": [],
      "availability_passlist": [],
    },
  "device_options": {},
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "frontend": { "port": 8099 },
  "experimental": {},
}
```

The frontend shows my username but no password. Home assistant shows both username and password. Does it mean it took them from the config file?

Koenkk, 2022-06-07:
@jlo88 I'm a bit confused, you need to put MQTT connection details in the MQTT section of the addon config, can you make a screenshot of the addon config? I expect something like this:

[image]

Koenkk, 2022-06-08:
Interesting, getting the same error. I think this may be a feature of Home Assistant. Seems it tries to parse the secret itself instead of letting zigbee2mqtt do it. Can you add the secret to the Home Assistant `secrets.yaml` instead?

jlo88, 2022-06-11:
@Koenkk sorry for the late response. I tried this out but this also didn't help. 

I adjusted the `secret.yaml` from z2m to make sure there can be no duplicate definitions:

```
mqtt_user_z2m: XXX
mqqt_password_z2m: YYY
```

in my home assistant `secrets.yaml` I have this: 

```
mqtt_user: XXX
mqqt_password: YYY
```

I did three tests: 

### Test 1

```
user: '!secret mqtt_user_z2m'
password: '!secret mqqt_password_z2m'
```

gives me 

```

Failed to save add-on configuration, Unknown secret 'mqtt_user_z2m' in Zigbee2mqtt (45df7312_zigbee2mqtt). Got {'data_path': '/config/zigbee2mqtt', 'socat': {'enabled': False, 'master': 'pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777', 'slave': 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5', 'options': '-d -d', 'log': False}, 'mqtt': {'base_topic': 'zigbee2mqtt', 'server': 'mqtt://core-mosquitto', 'user': '!secret mqtt_user_z2m', 'password': '!secret mqqt_password_z2m'}, 'serial': {'port': '/dev/ttyACM0'}, 'zigbee_herdsman_debug': True, 'zigbee_shepherd_devices': True}
```

so it doesn't even seem to find the username. 

### Test 2

```
user: '!secret mqtt_user'
password: '!secret mqtt_password'
```

gives me 

```
Failed to save add-on configuration, Unknown secret 'mqtt_password' in Zigbee2mqtt (45df7312_zigbee2mqtt). Got {'data_path': '/config/zigbee2mqtt', 'socat': {'enabled': False, 'master': 'pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777', 'slave': 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5', 'options': '-d -d', 'log': False}, 'mqtt': {'base_topic': 'zigbee2mqtt', 'server': 'mqtt://core-mosquitto', 'user': '!secret mqtt_user', 'password': '!secret mqtt_password'}, 'serial': {'port': '/dev/ttyACM0'}, 'zigbee_herdsman_debug': True, 'zigbee_shepherd_devices': True}
```

### Test 3

```
user: '!secret mqtt_user'
password: YYY
```

succes!

Seems to me:

1. It does not look at my `secret.yaml` in the z2m folder.
1. It can find the username in `secrets.yaml` from home assistant. 
1. It cannot find the password in `secrets.yaml` from home assistant even though it is defined there.

Koenkk, 2022-06-12:
I've just checked with a HA developer and it indeed looks from the Home Assistant `secrets.yaml` (there is no way to disable this behaviour). It works for me (maybe it requires restarting HA?):

[image]

[image]

jlo88, 2022-06-14:
I tried a home assistant restart and now I can use the password from the hass secrets.yaml. Thanks!

Koenkk, 2022-06-15:
Great, added this info the to readme: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt

Assuming this can be closed now.

closed 2022-06-15 by Koenkk