Koenkk/zigbee2mqtt #12719: M515egbzn (TS0601 _TZE200_xuzcvlku) blind motor doesn't get state and position

VikeDragon, 2022-06-05:
### What happened?

Greetings!

I had to reconnect one of the two devices M515egbzn (TS0601 _TZE200_xuzcvlku) blind motor after updating 1.25.2. 
The status and position  are not accepted from the device. 
Connected converters Tuya and Moes - does not help. 
Another device that has not reconnected works great.
Here is the last state of the problematic device after reconnection :
{
"last_seen": "2022-06-05T09:57:39+05:00",
" communication quality": 149,
"running": true
}
, and should be reflected like this:
{
"last_seen": "2022-06-05T09:47:02+05:00",
" communication quality": 102,
"position": 0,
"in progress": true,
"status": "CLOSE"
}

### What did you expect to happen?

The status and position  are accepted from the device and should be reflected like this:
{
"last_seen": "2022-06-05T09:47:02+05:00",
" communication quality": 102,
"position": 0,
"in progress": true,
"status": "CLOSE"
}

### How to reproduce it (minimal and precise)

reconnect M515egbzn (TS0601 _TZE200_xuzcvlku) blind motor

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus (zStack3x0)

### Debug log

2022-06-05 11:14:04MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec20a3c', payload '{"last_seen":"2022-06-05T11:14:14+05:00","linkquality":94,"running":null}'
2022-06-05 11:17:06Received Zigbee message from '0x5c0272fffec20a3c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":1}],"seq":700}' from endpoint 1 with groupID 0
2022-06-05 11:17:06MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec20a3c', payload '{"last_seen":"2022-06-05T11:17:16+05:00","linkquality":138,"running":true}'
2022-06-05 11:17:07Received Zigbee message from '0x5c0272fffec20a3c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,2],"type":"Buffer"},"datatype":2,"dp":2}],"seq":956}' from endpoint 1 with groupID 0
2022-06-05 11:17:07MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec20a3c', payload '{"last_seen":"2022-06-05T11:17:16+05:00","linkquality":138,"running":true}'
2022-06-05 11:17:08Received Zigbee message from '0x5c0272fffec20a3c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,7],"type":"Buffer"},"datatype":2,"dp":2}],"seq":956}' from endpoint 1 with groupID 0
2022-06-05 11:17:08MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec20a3c', payload '{"last_seen":"2022-06-05T11:17:17+05:00","linkquality":138,"running":true}'
2022-06-05 11:17:09Received Zigbee message from '0x5c0272fffec20a3c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,11],"type":"Buffer"},"datatype":2,"dp":2}],"seq":956}' from endpoint 1 with groupID 0
2022-06-05 11:17:09MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec20a3c', payload '{"last_seen":"2022-06-05T11:17:18+05:00","linkquality":138,"running":true}'
2022-06-05 11:17:10Received Zigbee message from '0x5c0272fffec20a3c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,16],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1212}' from endpoint 1 with groupID 0
2022-06-05 11:17:10MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec20a3c', payload '{"last_seen":"2022-06-05T11:17:19+05:00","linkquality":142,"running":true}'
2022-06-05 11:17:11Received Zigbee message from '0x5c0272fffec20a3c', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,20],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1212}' from endpoint 1 with groupID 0
2022-06-05 11:17:11MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec20a3c', payload '{"last_seen":"2022-06-05T11:17:20+05:00","linkquality":142,"running":true}'

ifajardoj, 2022-06-07:
Same problem here with two devices (M515EGBZN TS0601 _TZE200_xuzcvlku), roller blinds position not updating with zigbee2mqtt release 1.25.2-1. Going back to release 1.25.1-2 or before solve the problem. I'm using ROUTER_CC2530_CC2591.

Thanks.

kristofejro, 2022-06-07:
For me the problem is the same.
I have two TS0601 (_TZE200_xuzcvlku). With 1.25.2 position physicaly is changing but its value is not updating in the software. With 1.25.1 it is working fine. I am on zzh.

leroyloren, 2022-06-08:
1) **Physically press a button on the device**
```
Debug 2022-06-08 10:35:25Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,98],"type":"Buffer"},"datatype":2,"dp":2}],"seq":3}' from endpoint 1 with groupID 0
Info 2022-06-08 10:35:25MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:35:26+02:00","linkquality":123,"position":3,"running":true,"state":"OPEN"}'
Debug 2022-06-08 10:35:26Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2}],"seq":3}' from endpoint 1 with groupID 0
Info 2022-06-08 10:35:26MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:35:27+02:00","linkquality":127,"position":3,"running":true,"state":"OPEN"}'
Debug 2022-06-08 10:35:29Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,98],"type":"Buffer"},"datatype":2,"dp":2}],"seq":3}' from endpoint 1 with groupID 0
Info 2022-06-08 10:35:29MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:35:30+02:00","linkquality":131,"position":3,"running":true,"state":"OPEN"}'
Debug 2022-06-08 10:35:30Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,97],"type":"Buffer"},"datatype":2,"dp":2}],"seq":3}' from endpoint 1 with groupID 0
Info 2022-06-08 10:35:30MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:35:31+02:00","linkquality":127,"position":3,"running":true,"state":"OPEN"}'
```

2) Interface control (button down)
```
Debug 2022-06-08 10:37:00Received MQTT message on 'zigbee2mqtt/Roleta pokoj/set' with data 'CLOSE'
Debug 2022-06-08 10:37:00Publishing 'set' 'state' to 'Roleta pokoj'
Debug 2022-06-08 10:37:01Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":1}],"seq":4}' from endpoint 1 with groupID 0
Info 2022-06-08 10:37:01MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:37:02+02:00","linkquality":112,"position":3,"running":true,"state":"OPEN"}'
Debug 2022-06-08 10:37:02Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,98],"type":"Buffer"},"datatype":2,"dp":2}],"seq":4}' from endpoint 1 with groupID 0
Info 2022-06-08 10:37:02MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:37:03+02:00","linkquality":120,"position":3,"running":true,"state":"OPEN"}'
Debug 2022-06-08 10:37:03Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2}],"seq":4}' from endpoint 1 with groupID 0
Info 2022-06-08 10:37:03MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:37:04+02:00","linkquality":112,"position":3,"running":true,"state":"OPEN"}'
```

3) Interface control (button up)
```
Debug 2022-06-08 10:37:47Received MQTT message on 'zigbee2mqtt/Roleta pokoj/set' with data 'OPEN'
Debug 2022-06-08 10:37:47Publishing 'set' 'state' to 'Roleta pokoj'
Debug 2022-06-08 10:37:48Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":5}' from endpoint 1 with groupID 0
Info 2022-06-08 10:37:48MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:37:49+02:00","linkquality":116,"position":3,"running":true,"state":"OPEN"}'
Debug 2022-06-08 10:37:49Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,99],"type":"Buffer"},"datatype":2,"dp":2}],"seq":5}' from endpoint 1 with groupID 0
Info 2022-06-08 10:37:49MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:37:50+02:00","linkquality":112,"position":3,"running":true,"state":"OPEN"}'
Debug 2022-06-08 10:37:50Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,98],"type":"Buffer"},"datatype":2,"dp":2}],"seq":5}' from endpoint 1 with groupID 0
Info 2022-06-08 10:37:50MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:37:51+02:00","linkquality":109,"position":3,"running":true,"state":"OPEN"}'
Debug 2022-06-08 10:37:51Received Zigbee message from 'Roleta pokoj', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,97],"type":"Buffer"},"datatype":2,"dp":2}],"seq":5}' from endpoint 1 with groupID 0
Info 2022-06-08 10:37:51MQTT publish: topic 'zigbee2mqtt/Roleta pokoj', payload '{"last_seen":"2022-06-08T10:37:52+02:00","linkquality":109,"position":3,"running":true,"state":"OPEN"}'
```

ifajardoj, 2022-06-08:
Here is my log when closing the device "ESTOR SALON", marked with bold text the info message.

**Zigbee2MQTT:info  2022-06-08 12:35:24: MQTT publish: topic 'zigbee2mqtt/ESTOR SALON', payload '{"linkquality":93,"position":100,"running":true,"state":"OPEN"}'**
2022-06-08T10:35:24.547Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-06-08T10:35:24.547Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-06-08T10:35:24.547Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-06-08T10:35:24.547Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-06-08T10:35:24.548Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-06-08T10:35:24.552Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,59,253]
2022-06-08T10:35:24.553Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,59,253]
2022-06-08T10:35:24.553Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,59] - 253
2022-06-08T10:35:24.553Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":59}
2022-06-08T10:35:24.553Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-06-08T10:35:25.193Z zigbee-herdsman:controller:endpoint Command 0x804b50fffe4e49df/1 manuSpecificTuya.dataRequest({"seq":3,"dpValues":[{"dp":1,"datatype":4,"data":[1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-06-08T10:35:25.193Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x804b50fffe4e49df:33293/1 (0,0,1)
2022-06-08T10:35:25.194Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":33293,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":60,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[17,25,0,3,0,1,4,0,1,1]}}
2022-06-08T10:35:25.194Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,20,36,1,13,130,1,1,0,239,60,0,30,10,17,25,0,3,0,1,4,0,1,1,119]
2022-06-08T10:35:25.206Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-06-08T10:35:25.207Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-06-08T10:35:25.207Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-06-08T10:35:25.208Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-06-08T10:35:25.208Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-06-08T10:35:25.212Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,60,250]
2022-06-08T10:35:25.213Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,60,250]
2022-06-08T10:35:25.213Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,60] - 250
2022-06-08T10:35:25.213Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":60}
2022-06-08T10:35:25.213Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-06-08T10:35:25.472Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,13,130,1,183,143,50]
2022-06-08T10:35:25.473Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,13,130,1,183,143,50]
2022-06-08T10:35:25.474Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [13,130,1,183,143] - 50
2022-06-08T10:35:25.474Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":33293,"relaycount":1,"relaylist":[36791]}
2022-06-08T10:35:25.475Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-06-08T10:35:25.519Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,33,68,129,0,0,0,239,13,130,1,1,0,90,0,194,215,20,0,0,13,9,69,1,2,0,2,2,0,4,0,0]
2022-06-08T10:35:25.520Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,33,68,129,0,0,0,239,13,130,1,1,0,90,0,194,215,20,0,0,13,9,69,1,2,0,2,2,0,4,0,0]
2022-06-08T10:35:25.521Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,13,183,143,28,176]
2022-06-08T10:35:25.522Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,33,68,129,0,0,0,239,13,130,1,1,0,90,0,194,215,20,0,0,13,9,69,1,2,0,2,2,0,4,0,0,0,13,183,143,28,176]
2022-06-08T10:35:25.522Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,13,130,1,1,0,90,0,194,215,20,0,0,13,9,69,1,2,0,2,2,0,4,0,0,0,13,183,143,28] - 176
2022-06-08T10:35:25.523Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":33293,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":90,"securityuse":0,"timestamp":1365954,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,69,1,2,0,2,2,0,4,0,0,0,13]}}
2022-06-08T10:35:25.526Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":69,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"seq":2,"dpValues":[{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,13]}}]},"Command":{"ID":1,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataResponse"}},"address":33293,"endpoint":1,"linkquality":90,"grou [... cut ...]
2022-06-08T10:35:25.850Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,62,248]
2022-06-08T10:35:25.851Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,62,248]
2022-06-08T10:35:25.851Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,62] - 248
2022-06-08T10:35:25.852Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":62}
2022-06-08T10:35:25.852Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

Koenkk, 2022-06-08:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-06-08 by Koenkk

leroyloren, 2022-06-08:
I confirm that it works again ;-) Thanks.