Koenkk/zigbee2mqtt #12721: temperature sensor gone, now suddenly device temperature?

atv2016, 2022-06-05:
### What happened?

see above

### What did you expect to happen?

Just want to know if this is normal/expected to break in this release, seems to happen after i just updated to the latest.

### How to reproduce it (minimal and precise)

see above

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211210

### Adapter

conbee-ii

### Debug log

n/a

Otnow, 2022-06-05:
That's the way it should be.

Please read: https://github.com/Koenkk/zigbee2mqtt/issues/12613 and https://github.com/Koenkk/zigbee2mqtt/issues/12700.

atv2016, 2022-06-05:
Ok gotcha, thanks!

So climactic temperature and chip temperature. So i guess i just need to update all my dashboards to use that entity now ? Is there any quick way of doing this as this breaks a lot of dashboards.

Otnow, 2022-06-05:
> So i guess i just need to update all my dashboards to use that entity now ? Is there any quick way of doing this as this breaks a lot of dashboards.

I don't quite understand what it's about. Home Assistant?

atv2016, 2022-06-05:
Ah yes sorry. Home assistant.

Otnow, 2022-06-05:
> Ah yes sorry. Home assistant.

You can try using the replace function in the [File editor](https://github.com/home-assistant/addons/tree/master/configurator) addon:

[image]