Koenkk/zigbee2mqtt #12722: Feedback from Aqara E1 without neutral (QBKG39LM) not work

YuriyDY, 2022-06-05:
### What happened?

Hello everyone,

I have same issue that was with QBKG38LM one year ago:
https://github.com/Koenkk/zigbee2mqtt/issues/7625

But in my case I have switch with two gang without neutral (QBKG39LM). The symptoms are same. I can pair switch with Z2M, I can send commands from Z2M to switch, but switch cannot send commands to Z2M (for example, when I manually switching states).

Could you please give me some advices how to fix that issue?

### Zigbee2MQTT version

1.25.1-1

### Adapter firmware version

20210319

### Adapter

DIY CC2652 based

Otnow, 2022-06-13:
You need to update the firmware of the coordinator and re-pair problem devices:

https://www.zigbee2mqtt.io/devices/QBKG39LM.html#adapter-firmware

YuriyDY, 2022-06-14:
thanks a lot sir! It works now

closed 2022-06-14 by YuriyDY