Koenkk/zigbee2mqtt #12726: [New device support]: Niko connectable dimmer 552-72201

norbertbxl, 2022-06-05:
### Link

https://www.niko.eu/en/products/niko-home-control/products-for-traditional-wiring/connected-dimmer-3-200-w-2-wire-zigbee-productmodel-niko-e61c5693-2d0c-5be5-8544-461983048049

### Database entry

Entry: {"id":23,"type":"Router","ieeeAddr":"0xcc86ecfffe25d5b5","nwkAddr":15836,"manufId":4703,"manufName":"NIKO NV","powerSource":"Mains (single phase)","modelId":"Connectable dimmer,3-200W,2-wire","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":267,"inClusterList":[0,3,4,5,6,8,2821,64512,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Connectable dimmer,3-200W,2-wire","manufacturerName":"NIKO NV","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20201010","swBuildId":"313-689-00"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20201010","swBuildId":"313-689-00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648237398277,"useImplicitCheckin":true}

### Comments

Hello
Can someone help me in adding this device

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Connectable dimmer,3-200W,2-wire'],
    model: '552-72201',
    vendor: 'Niko',
    description: 'Connectable dimmer',
    fromZigbee: [fz.temperature], // <-- added here all clusters reported from zigbee
    toZigbee: [], // <-- add here all clusters to send commands to zigbee
    exposes: [e.battery(), e.temperature(), e.humidity()], // <-- this will define which fields will be exposed in the definition message to configure a front end (e.g. the z2m frontend, Home Assistant, Domoticz)
};

module.exports = definition;
```

sjorge, 2022-06-05:
Try with something like this, not super confident as all the Niko stuff I messed with were a bit non-standard

```node
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Connectable dimmer,3-200W,2-wire'],
    model: '552-72201',
    vendor: 'Niko',
    description: 'Connectable dimmer',
    fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.command_move, fz.command_stop],
    toZigbee: [tz.light_onoff_brightness, tz.level_config], 
    exposes: [e.light_brightness().withLevelConfig()],
};

module.exports = definition;
```

norbertbxl, 2022-06-05:
Yes
Thank you very much Sjorge !!!
Works as it should.

sjorge, 2022-06-05:
I'll do a PR so it gets added in the next release.

norbertbxl, 2022-06-05:
Heel erg bedankt
Merci

norbertbxl, 2022-06-05:
I noticed a little problem
The state is not updated when switching with the switch

sjorge, 2022-06-05:
Do you know how to configure reporting using the frontend?

If so try adding OnOff.OnOff and LevelCtl.currentLevel.

If not I can whip up a modified device definition.

norbertbxl, 2022-06-05:
if I can ask for modifications it would be great

sjorge, 2022-06-05:
```node
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Connectable dimmer,3-200W,2-wire'],
    model: '552-72201',
    vendor: 'Niko',
    description: 'Connectable dimmer',
    fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.command_move, fz.command_stop],
    toZigbee: [tz.light_onoff_brightness, tz.level_config],
    exposes: [e.light_brightness().withLevelConfig()],
    configure: async (device, coordinatorEndpoint, logger) => {
       const endpoint = device.getEndpoint(1);
       await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
       await reporting.onOff(endpoint);
       await reporting.brightness(endpoint);
   },
};

module.exports = definition;
```

sjorge, 2022-06-06:
Oh you might need to reconfigure it after updating. Let me know if this helped.

norbertbxl, 2022-06-06:
Hello Sjorge
At the moment it looks like everything is working as it should.

norbertbxl, 2022-06-25:
Hello Sjorge 

For some time I noticed that turning on the light with the switch the brightness is set to zero
I think it may have something to do with the current_level_startup settings but when I want to change the setting I get an error message

sjorge, 2022-06-25:
Is the actual brightness of the bulb(s) set to 0 ? Or does mqtt report brightness=0 for the dimmer?

norbertbxl, 2022-06-25:
the actual brightness of the bulb is set to zero, but only with a toggle device

norbertbxl, 2022-06-25:
if I turn on the bulb remotely, it remembers the last brightness
if I turn on with the switch it is set to zero

sjorge, 2022-06-25:
What error messages do you get when trying to set current_level_startup ?

norbertbxl, 2022-06-25:
Publish 'set' "level config' to
W
"Salon Test' failed: Error: Write
Oxcc86ecfffe 25d638/1
122
genLevelCtri(("startUpCurrentLevel*
("sendWhen":"immediate""timeout"
failed (Status
'UNSUPPORTED ATTRIBUTE")'

sjorge, 2022-06-25:
Try with setting `on_level`

norbertbxl, 2022-06-25:
there are no settings on_level

sjorge, 2022-06-25:
> there are no settings on_level

It's not in expose yes, but you should be able to set it using mqtt directly

norbertbxl, 2022-06-25:
unfortunately I can't handle it

norbertbxl, 2022-07-07:
Hello Sjorge

After the update I added a few switches and initially they worked ok but after two days I noticed that already two of them stopped reporting the state when I used the physical  switch.

Do you know why this is happening

sjorge, 2022-07-07:
Do they start to report again if you hit reconfigure ?

norbertbxl, 2022-07-07:
One was enough to reset and the other had to be removed and added again

closed 2022-08-14 by github-actions[bot]