Koenkk/zigbee2mqtt #12732: 100% CPU after some time, and then forever until restart

Tsury, 2022-06-06:
### What happened?

After some time, `node` will consume 100% CPU forever

### What did you expect to happen?

Normal operation

### How to reproduce it (minimal and precise)

I'm not sure.

Using `strace` on the node process yields endless spam which looks like this:

```
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344385, tv_nsec=864511492}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344385, tv_nsec=864693101}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344385, tv_nsec=864866006}) = 0
epoll_wait(14, [{EPOLLIN, {u32=17, u64=17}}], 1024, 0) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344386, tv_nsec=941712480}) = 0
read(17, "\1\0\0\0\0\0\0\0", 1024)      = 8
write(23, "\306bL\374e\321\321\7\300\t\324?\2357.\322v\371s0\336\377\2\257\277\2620\6\341(\253\313"..., 65536) = 65536
futex(0x3c3a2c8, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
writev(24, [{iov_base="HTTP/1.1 200 OK\r\nContent-Type: t"..., iov_len=342}, {iov_base="\37\213\10\0\0\0\0\0\2\3\354\275{\217+7\222'\372\377\375\24\232c\30>\325\226\344\314\224R"..., iov_len=60389}], 2) = 60731
futex(0x3c3a2c8, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344386, tv_nsec=945532881}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344386, tv_nsec=946887828}) = 0
futex(0x3c3a2c8, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a2cc, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344387, tv_nsec=887496276}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344387, tv_nsec=887683293}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344387, tv_nsec=888087325}) = 0
epoll_wait(14, [{EPOLLIN, {u32=17, u64=17}}], 1024, 0) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344388, tv_nsec=961701301}) = 0
read(17, "\1\0\0\0\0\0\0\0", 1024)      = 8
write(23, "A\220\353(Wc\342E\206\256V`i\250Y[\7\7y\227\6\362Ww+\344\316\312\2149\206\30"..., 65536) = 65536
futex(0x3c3a2cc, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
clock_gettime64(CLOCK_REALTIME, {tv_sec=1654496192, tv_nsec=430263116}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344388, tv_nsec=964822710}) = 0
futex(0x3c3a2cc, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a2cc, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344390, tv_nsec=41465560}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344390, tv_nsec=41675132}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344390, tv_nsec=41856574}) = 0
epoll_wait(14, [{EPOLLIN, {u32=17, u64=17}}], 1024, 0) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344391, tv_nsec=115584606}) = 0
read(17, "\1\0\0\0\0\0\0\0", 1024)      = 8
write(23, "S]8\374b\230\216\\)\37\n\2178:f\240\\\301\7\200\355\262\250\323\311\177\2!0{\240\251"..., 65536) = 65536
futex(0x3c3a2c8, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344391, tv_nsec=118296409}) = 0
futex(0x3c3a2c8, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a2c8, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344391, tv_nsec=867224778}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344391, tv_nsec=867414609}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344391, tv_nsec=867592940}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344391, tv_nsec=867824512}) = 0
epoll_wait(14, [{EPOLLIN, {u32=17, u64=17}}], 1024, 0) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344392, tv_nsec=938511618}) = 0
read(17, "\1\0\0\0\0\0\0\0", 1024)      = 8
write(23, "k\207\247\25\244\321\257\331-g\6\256\3451\345^sgY\274\223\21zw\214b\36\347}\352\210["..., 65536) = 65536
futex(0x3c3a2c8, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344392, tv_nsec=941149773}) = 0
futex(0x3c3a2cc, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a2cc, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x3c3a284, FUTEX_WAKE_PRIVATE, 1) = 1
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=981545725}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=981754019}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=981968738}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=982304827}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=982507973}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=982739841}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=982909265}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=983102207}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=983285890}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=983501888}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=983807643}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=983982197}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=984238842}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=984425803}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=984645652}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=984824373}) = 0
epoll_ctl(14, EPOLL_CTL_DEL, 24, 0xbe9e0750) = 0
close(24)                               = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=985811824}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=986120247}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=986329763}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=986564316}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=986748036}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=986939774}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=344393, tv_nsec=987118587}) = 0
```

### Zigbee2MQTT version

 1.25.2

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

Koenkk, 2022-06-06:
Could you provide the herdsman debug log and indicate the moment the CPU goes to 100%?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Tsury, 2022-06-06:
I started it with `DEBUG=zigbee-herdsman* npm start` and will wait for it to reach 100% CPU again. Not sure how to ensure I catch it on time though.

closed 2022-07-15 by github-actions[bot]

segdy, 2023-06-02:
Can this be opened again?
I have exactly the same after upgrading Home Assistant and mosquito Add-on yesterday! Nothing else changed.

Same thing ... starts working and after some time 100% CPU and clock_gettime64 when I connect via strace. My whole home automation is down now :-(

I have tried rebooting the zigbee2mqtt machine (RPi Zero) and updating to latest z2m ... no changes.

Nothing problematic in the log files for now. I'm running `DEBUG='zigbee-herdsman*' npm start | tee fu.log` now...

@Tsury  how did this disappear for you?