Koenkk/zigbee2mqtt #12736: Zemismart wall socket ZIGBEE-B09-UK can't control L1/L2 separately

tjxism, 2022-06-06:
### What happened?

when I turn L1 on， then L2 will on, too.
when I turn L1 off， then L2 will off, too.

It seems the same issue as. described in #12481 and probably both used the same tuya solution.  However as this is a wall switch I can't control it outside of HA.  

### What did you expect to happen?

L1 and L2 should be controlled independently. The log kind of shows both payloads are sent while change state in Z2M. 

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

NEWEST

### Adapter

sonoff plus

### Debug log

Info 2022-06-06 20:55:08MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:55:08+08:00","linkquality":171,"power_outage_memory":"restore","state_l1":"ON","state_l2":"ON"}'
Info 2022-06-06 20:55:08MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:55:08+08:00","linkquality":171,"power_outage_memory":"restore","state_l1":"OFF","state_l2":"ON"}'
Info 2022-06-06 20:55:08MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:55:08+08:00","linkquality":171,"power_outage_memory":"restore","state_l1":"OFF","state_l2":"ON"}'
Info 2022-06-06 20:55:08MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:55:08+08:00","linkquality":171,"power_outage_memory":"restore","state_l1":"OFF","state_l2":"OFF"}'
Info 2022-06-06 20:55:10MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:55:10+08:00","linkquality":156,"power_outage_memory":"restore","state_l1":"OFF","state_l2":"OFF"}'
Info 2022-06-06 20:55:10MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:55:10+08:00","linkquality":171,"power_outage_memory":"restore","state_l1":"ON","state_l2":"OFF"}'
Info 2022-06-06 20:55:10MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:55:10+08:00","linkquality":171,"power_outage_memory":"restore","state_l1":"ON","state_l2":"OFF"}'
Info 2022-06-06 20:55:10MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:55:10+08:00","linkquality":174,"power_outage_memory":"restore","state_l1":"ON","state_l2":"ON"}'

tjxism, 2022-06-06:
The log below shows what happens when turning on L1 in HA. A payload to turn on L2 is automatically sent a split second after "turn on L1" is sent. 

Info 2022-06-06 20:58:03MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:58:03+08:00","linkquality":171,"power_outage_memory":"restore","state_l1":"OFF","state_l2":"OFF"}'
Info 2022-06-06 20:58:03MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:58:03+08:00","linkquality":171,"power_outage_memory":"restore","state_l1":"ON","state_l2":"OFF"}'
Info 2022-06-06 20:58:03MQTT publish: topic 'zigbee2mqtt/HAL Tuya Wall Socket', payload '{"last_seen":"2022-06-06T20:58:03+08:00","linkquality":171,"power_outage_memory":"restore","state_l1":"ON","state_l2":"ON"}'

Koenkk, 2022-06-07:
Fixed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)