Koenkk/zigbee2mqtt #12741: After performing an OTA on GL-C-008P the device is not functioning correctly.

Laver, 2022-06-07:
### What happened?

Before the OTA, GL-C-008P was working fine, full color control and CCT as expected. Firmware was originally "V_1_0", after update it's "11076802".

Following the update the device turns on / off, RGB controls somewhat work (only Red and Blue LEDs light, no green), but CCT control is completely broken (reported as Exposes color_temp, but setting a color_temp value results in no LEDs lit).

I assume that the firmware version being used for OTA is incompatible with my hardware version. 

Is there anyway to get back to the old firmware? I have a second GL-C-008P which hasn't been updated and is still reporting the old firmware, can the firmware be retrieved through zigbee, perhaps?

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2

Koenkk, 2022-06-07:
You should be able to force an update on it via: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files. You can use the following OTA file: https://github.com/Koenkk/zigbee-OTA/raw/master/images/Gledopto/GL-C-008P_V17A1_OTAV7_20210303--V1-4.ota

malaakso, 2022-06-08:
If you go to Dev Console and query for the hwVersion attribute from basic cluster, what do you get? I have three with hwVersion 2, out of which one has firmware V_1_4 and two have 11076802, and one with hwVersion 3 and firmware V_1_6_H2. I wonder if the 11076802 is only good for hwVersion 2.

Laver, 2022-07-01:
Thanks for the speedy replies, sorry for taking so long to get back.

Read result of 'genBasic': {"hwVersion":2,"dateCode":"20211020","swBuildId":"11076802"}

Running the downgrade now.

iculookn, 2022-07-22:
just received this firmware from gledopto for the GL-C-008P. Looks to be newer than V_1_7_A1, but one version down from "11076802" so maybe this does not have the same issue?

I HAVE NOT TRIED IT YET.
 
[GL-C-008P_V11076801_OTAV12_211020%.zip](https://github.com/Koenkk/zigbee2mqtt/files/9164099/GL-C-008P_V11076801_OTAV12_211020.zip)

closed 2022-08-30 by github-actions[bot]

ndoo, 2024-08-20:
I force updated my GL-C-008P to the available OTA firmwares that resulted in the same symptoms as OP (Red and Blue LEDs only).

| V_1_8_T1 (original shipping firmware) | V11376802 Build 20230306 | V11076801 Build 20211020 | V_1_7_A1 Build 20210303 |
|---|---|---|---|
| Red | Red | Red | Red |
| Green | Green | Blue | Blue |
| Blue | Blue | NC | NC |
| Warm White | Warm White | NC | NC |
| Cool White | Cool White | NC | NC |

I received a firmware from Gledopto, V11376802 build date 20230306, that fixed the issue, here it is should anyone have this hardware revision:

[GL-C-008P_V11376802_20230306.zip](https://github.com/user-attachments/files/16676787/GL-C-008P_V11376802_20230306.zip)

Unhelpfully, `Read result of 'genBasic': {"hwVersion":2}` so we can't differentiate between which PCB maps to which OTA files.