Koenkk/zigbee2mqtt #12752: Ikea starkvind, can't set speed

realjax, 2022-06-08:
### What happened?

Trying to set the speed of the fan results in an error, here's the entry from the logs:

```
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'fan_mode' to 'Starkvind' failed: 'Error: Write 0xb4e3f9fffe2b2595/1 manuSpecificIkeaAirPurifier({\"fanMode\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4476,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"Starkvind"},"type":"zigbee_publish_error"}'
```

### What did you expect to happen?

I expected it to work :)

### How to reproduce it (minimal and precise)

Just try to set the mode through the 'exposes' tab of the web interface.

### Zigbee2MQTT version

 1.24.0 

### Adapter firmware version

20210120

### Adapter

electralama zig a zag

sjorge, 2022-06-08:
Can you set anything else on the device? Re error `MAC no ack` would indicated the device is not responding at all.

realjax, 2022-06-08:
Yes, II can set the child lock and the LED just fine.

sjorge, 2022-06-08:
What firmware does it have? Just tested on mine and it's working fine

```
debug 2022-06-08 14:35:00Received MQTT message on 'zigbee2mqtt/airpurifier/bedroom/set' with data '{"fan_state":"ON"}'
debug 2022-06-08 14:35:00Publishing 'set' 'fan_state' to 'airpurifier/bedroom'
debug 2022-06-08 14:35:00Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":1,"fanSpeed":30}' from endpoint 1 with groupID 0
info 2022-06-08 14:35:00MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"unknown","child_lock":"UNLOCK","fan_mode":"auto","fan_speed":0,"fan_state":"ON","led_enable":true,"linkquality":94,"pm25":-1,"replace_filter":false}'
info 2022-06-08 14:35:01MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"unknown","child_lock":"UNLOCK","fan_mode":"auto","fan_speed":5,"fan_state":"ON","led_enable":true,"linkquality":94,"pm25":-1,"replace_filter":false}'
debug 2022-06-08 14:35:13Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"particulateMatter25Measurement":3}' from endpoint 1 with groupID 0
info 2022-06-08 14:35:14MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"excellent","child_lock":"UNLOCK","fan_mode":"auto","fan_speed":5,"fan_state":"ON","led_enable":true,"linkquality":94,"pm25":3,"replace_filter":false}'
debug 2022-06-08 14:35:30Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanSpeed":25}' from endpoint 1 with groupID 0
info 2022-06-08 14:35:31MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"excellent","child_lock":"UNLOCK","fan_mode":"auto","fan_speed":4,"fan_state":"ON","led_enable":true,"linkquality":91,"pm25":3,"replace_filter":false}'
debug 2022-06-08 14:35:50Received MQTT message on 'zigbee2mqtt/airpurifier/bedroom/set' with data '{"fan_mode":"5"}'
debug 2022-06-08 14:35:50Publishing 'set' 'fan_mode' to 'airpurifier/bedroom'
debug 2022-06-08 14:35:50Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":30,"fanSpeed":30}' from endpoint 1 with groupID 0
info 2022-06-08 14:35:50MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"excellent","child_lock":"UNLOCK","fan_mode":"5","fan_speed":4,"fan_state":"ON","led_enable":true,"linkquality":94,"pm25":3,"replace_filter":false}'
info 2022-06-08 14:35:50MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"excellent","child_lock":"UNLOCK","fan_mode":"5","fan_speed":5,"fan_state":"ON","led_enable":true,"linkquality":94,"pm25":3,"replace_filter":false}'
debug 2022-06-08 14:36:05Received MQTT message on 'zigbee2mqtt/airpurifier/bedroom/set' with data '{"fan_mode":"off"}'
debug 2022-06-08 14:36:05Publishing 'set' 'fan_mode' to 'airpurifier/bedroom'
debug 2022-06-08 14:36:05Received Zigbee message from 'airpurifier/bedroom', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanMode":0,"fanSpeed":0}' from endpoint 1 with groupID 0
info 2022-06-08 14:36:05MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"excellent","child_lock":"UNLOCK","fan_mode":"off","fan_speed":5,"fan_state":"OFF","led_enable":true,"linkquality":91,"pm25":3,"replace_filter":false}'
info 2022-06-08 14:36:06MQTT publish: topic 'zigbee2mqtt/airpurifier/bedroom', payload '{"air_quality":"excellent","child_lock":"UNLOCK","fan_mode":"off","fan_speed":0,"fan_state":"OFF","led_enable":true,"linkquality":91,"pm25":3,"replace_filter":false}'
```

Mine firmware is:
```
Firmware build date: 20210428
Firmware version: 1.0.033
```

realjax, 2022-06-08:
Firmware build date: 20210428
Firmware version: 1.0.033

exact same...  weird.

realjax, 2022-06-08:
Now it is working ! 
hmm...

I wonder if my son had turned the thing off earlier...  in that case, my apologies!   lets quickly close this

closed 2022-06-08 by realjax