Koenkk/zigbee2mqtt #12753: [New device support]: Tuya-like temperature and humidity sensor with LCD screen

Roofcatbusy, 2022-06-08:
### Link

https://aliexpress.com/item/1005004064535901.html

### Database entry

{"id":20,"type":"EndDevice","ieeeAddr":"0xa4c13858619c1a5f","nwkAddr":61110,"manufId":4417,"manufName":"_TZE200_qoy0ekbd","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[1,1026,1029,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_qoy0ekbd","powerSource":3,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1654671804259,"defaultSendRequestWhen":"immediate"}

### Comments

Cheap thermometer-hygrometer, purchased on Ali. Instead of developing external converter, I've just added its "Zigbee Manufacturer" (_TZE200_qoy0ekbd) to tuya.js TS0601 fingerprint field. please let me know if this was wrong way.

Koenkk, 2022-06-08:
> Instead of developing external converter, I've just added its "Zigbee Manufacturer" (_TZE200_qoy0ekbd) to tuya.js TS0601 fingerprint field. please let me know if this was wrong way.

Does it work after this?

Roofcatbusy, 2022-06-09:
> > Instead of developing external converter, I've just added its "Zigbee Manufacturer" (_TZE200_qoy0ekbd) to tuya.js TS0601 fingerprint field. please let me know if this was wrong way.
> 
> Does it work after this?

yes, it works, temperature and humidity data are consistent  with LCD display and Xiaomi sensor nearby.

Roofcatbusy, 2022-06-10:
> 

TS0601_temperature_humidity_sensor. Just in case `diff -c ` output follows:

```
roofcat@jethub-zavety:/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices$ diff -c tuya.js.orig tuya.js
*** tuya.js.orig        2022-06-08 10:28:02.287272018 +0300
--- tuya.js     2022-06-08 10:31:22.721965261 +0300
***************
*** 142,148 ****
      },
      {
          fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_bq5c8xfe'},
!             {modelID: 'TS0601', manufacturerName: '_TZE200_bjawzodf'}],
          model: 'TS0601_temperature_humidity_sensor',
          vendor: 'TuYa',
          description: 'Temperature & humidity sensor',
--- 142,149 ----
      },
      {
          fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_bq5c8xfe'},
!             {modelID: 'TS0601', manufacturerName: '_TZE200_bjawzodf'},
!             {modelID: 'TS0601', manufacturerName: '_TZE200_qoy0ekbd'}],
          model: 'TS0601_temperature_humidity_sensor',
          vendor: 'TuYa',
          description: 'Temperature & humidity sensor',

```

Koenkk, 2022-06-11:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-06-11 by Koenkk