Koenkk/zigbee2mqtt #12755: New devices support Niko 552-72301 + 552-72201

bjorndeboodt, 2022-06-08:
### Link

https://www.niko.eu/nl-be/artikel/552-72301 +++ https://www.niko.eu/nl-be/artikel/552-72201

### Database entry

{"id":22,"type":"Router","ieeeAddr":"0xcc86ecfffe25d228","nwkAddr":14321,"manufId":4703,"manufName":"NIKO NV","powerSource":"Mains (single phase)","modelId":"Connectable motor control,3A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,4,5,258,2821,64512,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Connectable motor control,3A","manufacturerName":"NIKO NV","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":6,"hwVersion":0,"dateCode":"20200930","swBuildId":"313-690-00"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":0,"dateCode":"20200930","swBuildId":"313-690-00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648236904583,"useImplicitCheckin":true}

### Comments

I tried to follow the supporting new device page but got stuck at point "2. Adding Your Device".
I try to connect 2 new devices from Niko: 552-72301 + 552-72201. I mannaged to pair these but when I started with creating an external converter file, the devices do not respond. I suppose these devices will need additional configuration.

Below I mentionned the links to the devices + the entry of each device from the database.

If someone is interessted in adding these devices to the zigbee2MQTT supported devices, I am willing to send these 2 devices by post so the more experienced programmers can use these devices for the programmation.

Below you can find the "external converter" file I have tried so far.

### External converter

```shell
External converter for device 552-72201:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Connectable dimmer, 3-200W, 2-wire'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '552-7221', // Vendor model number, look on the device for a model number
    vendor: 'Niko', // Vendor of the device (only used for documentation and startup logging)
    description: 'Connectable dimmer', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch(), e.battery(), e.temperature(), e.humidity(), e.action(['on', 'off', 'brightness_move_to_level'])], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

sjorge, 2022-06-08:
The dinner should already be supported in the dev branch
https://github.com/Koenkk/zigbee-herdsman-converters/blob/72ab25394432e30d21bc6207fcaba5ebfdb22326/devices/niko.js#L222-L236

sjorge, 2022-06-08:
If you're willing to test I can probably take a 'best guess' for the motor controller, but busy today so it will be tomorrow at the earliest.

bjorndeboodt, 2022-06-09:
Hi,

The dimmer 552-72201 in the dev brach works!

bjorndeboodt, 2022-06-09:
I also tried the converter file from @sjorge and get following response in the logging:

Device '0xcc86ecfffe25d228' is supported, identified as: Niko Connectable motor control (552-72301)
Jun 09 21:32:38 raspberrypi npm[1947]: Zigbee2MQTT:info  2022-06-09 21:32:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Connectable motor control","exposes":[{"features":[{"access":3,"name":"state","property":"state","type":"enum","values":["OPEN","CLOSE","STOP"]},{"access":7,"description":"Position of this cover","name":"position","property":"position","type":"numeric","value_max":100,"value_min":0}],"type":"cover"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"552-72301","options":[{"access":2,"description":"Inverts the cover position, false: open=100,close=0, true: open=0,close=100 (default false).","name":"invert_cover","property":"invert_cover","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Niko"},"friendly_name":"0xcc86ecfffe25d228","ieee_address":"0xcc86ecfffe25d228","status":"successful","supported":true},"type":"device_interview"}'

I see values, etc. when pressing the buttons:

Zigbee2MQTT:debug 2022-06-09 21:38:29: Received Zigbee message from '0xcc86ecfffe25d228', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":98}' from endpoint 1 with groupID 0

until now, the device doe not list up in the zigbee2mqtt bridge in nodered

sjorge, 2022-06-09:
Aside from not showing node-red does it respond to the following on mqtt ?

```bash
# NOTE: should open/close the blind to the ~50% position
mosquitto_pub  -t zigbee2mqtt/<dev>/set/position -m 50
# NOTE: should fully close the blind
mosquitto_pub  -t zigbee2mqtt/<dev>/set/state -m 'CLOSE"
# NOTE: should fully open the blind
mosquitto_pub  -t zigbee2mqtt/<dev>/set/state -m 'OPEN"
```

bjorndeboodt, 2022-06-13:
Hello @sjorge, it does repond to your lines and does exactly what it is expected to do (fully open/close + go to 50%)

sjorge, 2022-06-13:
Excellent, I’ll get a PR open to add support tomorrow.

Not sure why it’s not showing up in node-red though.

bjorndeboodt, 2022-06-20:
Hello @sjorge , can I help you with something else on this issue?

sjorge, 2022-06-20:
Basic support should now be in the dev branch, I think the node-red stuff might need a release to work. (Not sure)

kjjdejong, 2022-06-22:
Hi George and Koen,

How do you get the zigbee-code from those Niko devices? I want to build (could give it a try) or let some one else build an app in Homey Pro.

sjorge, 2022-06-22:
For basic stuff they mostly follow the ZCL, so you can use that to help you https://github.com/Koenkk/zigbee-herdsman/blob/master/docs/07-5123-08-Zigbee-Cluster-Library.pdf

Some other stuff uses custom attributes in one of the manufacturer reserved cluster ids, for those it’s mostly trial and error and guesswork.

Unfortunately NIKO does not seem to respond af all when asking them for the technical documentation on there clusters.

kjjdejong, 2022-06-22:
It seems Homey runs on an older ZCL, is that possible? https://etc.athom.com/zigbee_cluster_specification.pdf

sjorge, 2022-06-22:
It's an older version of the standard document yes, but that generally should not be an issue. There might be some things in ZCL v7 that are not in ZLC v6 but it should be backwards compatible, all Niko devices I own report they are using v3.

kjjdejong, 2022-06-22:
Ok, you also own 552-72301?

sjorge, 2022-06-22:
No, just 552-80699, 552-721X1, and 170-33505.

As you can see in this thread I just did a guess based on my previous experience with NIKO devices and the info provided here.

closed 2022-07-31 by github-actions[bot]

Sennevds, 2024-06-29:
Hi everyone,

I just found this issue in the ZHA repo where it looks like they found out how to set the traveling up and down time:
https://github.com/zigpy/zha-device-handlers/issues/3200
Maybe we need to integrate this also in z2m. If I find some time I will check how I can include it or if someone with some knowledge in writing this code can help

Sennevds, 2024-06-29:
Okay so just as test I've added the extra attributes to the cluster.js file in the herdsman package and I can read and write the state:
[image]
So I think that is the only thing needed to have it working!

Sennevds, 2024-06-29:
Koen already merged it. I'm not going to include the attributes as default on the UI (by adding them to converter) because you only need it once Soni don't see the need now