Koenkk/zigbee2mqtt #12760: TS011F _TZ3000_r6buo8ba not fully working

testos77, 2022-06-09:
### What happened?

Installed latest dev version. 
Device TS011F _TZ3000_r6buo8ba with sw_version":"20B-TZSKT11BS3.4" is successfully connected.
converter tuya.js treats device as: tuya-ts011f_plug_1

Device can be switched on and off.
**Power Monitoring is not functional (power, current, voltage)** no data recieved
child lock not functional (see debug error)

### What did you expect to happen?

Functional device - working converter version which I found in another thread (only missing child lock, indicator mode atm) 

```

module.exports = [
{
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_r6buo8ba'}],
        model: 'TS011F_plug',
        description: 'Smart plug (with power monitoring)',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1, acVoltageDivisor: 1, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1,
                acPowerDivisor: 1,
            });
        },
           exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage'),
        onEvent: (type, data, device, options) => {
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(globalStore.getValue(device, 'interval'));
                globalStore.clearValue(device, 'interval');
            } else if (!globalStore.hasValue(device, 'interval')) {
                const seconds = options && options.measurement_poll_interval ? options.measurement_poll_interval : 60;
                if (seconds === -1) return;
                const interval = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                        await endpoint.read('seMetering', ['currentSummDelivered']);
                    } catch (error) {/* Do nothing*/}
                }, seconds*1000);
                globalStore.putValue(device, 'interval', interval);
            }
        },
    }
];
 
```

### How to reproduce it (minimal and precise)

install dev version as above, link plug with firmware version as above, see output on frontend.

### Zigbee2MQTT version

1.25.2-dev (commit #389ff84)

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Debug log

info  2022-06-09 08:31:34: Configuring '0x70b3d52b60009054'
info  2022-06-09 08:31:34: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b60009054', payload '{"child_lock":"UNLOCK","current":null,"energy":null,"indicator_mode":null,"last_seen":"2022-06-09T08:31:34+02:00","linkquality":102,"power":null,"power_outage_memory":"off","state":"ON","update":{"state":null},"update_available":null,"voltage":null}'
debug 2022-06-09 08:31:34: Received Zigbee message from '0x70b3d52b60009054', type 'readResponse', cluster 'genBasic', data '{"appVersion":160," manufacturerName":"_TZ3000_r6buo8ba","modelId":"TS011F","powerSource":1,"zclVersion":2}' from endpoint 1 with groupID 0
info  2022-06-09 08:31:34: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b60009054', payload {"child_lock":"UNLOCK","current":null,"energy":null,"indicator_mode":null,"last_seen":"2022-06-09T08:31:34+02:00","linkquality":105,"power":null,"power_outage_memory":"off","state":"ON","update":{"state":null},"update_available":null,"voltage":null}'
error 2022-06-09 08:31:34: Failed to configure '0x70b3d52b60009054', attempt 2 (Error: Read 0x70b3d52b60009054/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,
"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/tuya.js:1571:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))

fernando155, 2022-06-09:
same device but I can not make it work. Any help for a newby please?

StawAr, 2022-06-10:
this plug does not work for me too :(  i am waiting for official support

testos77, 2022-06-11:
For reference:
 #12683
[with commit ](https://github.com/Koenkk/zigbee-herdsman-converters/commit/9b140b25944c02103b32368a9c26bb6fd45b9584) 
in upcoming dev version didn't work.

stanionascu, 2022-06-13:
I think that I have the same plug:

```
Zigbee Model
TS011F
Zigbee Manufacturer
_TZ3000_r6buo8ba
Firmware build date
20210720
Firmware version
20B-TZSKT11BS3.4
```

I also got a Tuya Gateway + Core IoT Platform for getting the Data Points. And the `indicator_mode` is missing on mine.

vbakker, 2022-06-16:
Same issue here.

unintended-consequences, 2022-06-24:
I tried the fernando155 solution. I have the on/off and child lock features functional. But the energy reporting metrics remain Null.

`Failed to configure '0xxxxxxxxxxxxxx', attempt 2 (Error: Read 0x70b3d52b6000921d/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22) at Object.configure (/opt/zigbee2mqtt/dist/util/externally-loaded.js:24:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21))`

fernando155, 2022-06-24:
> I tried the fernando155 solution. I have the on/off and child lock features functional. But the energy reporting metrics remain Null.
> 
> `Failed to configure '0xxxxxxxxxxxxxx', attempt 2 (Error: Read 0x70b3d52b6000921d/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22) at Object.configure (/opt/zigbee2mqtt/dist/util/externally-loaded.js:24:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:55:21))`
> 
> `MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0xxxxxxxxxxxx"},"error":"Failed to configure (Read 0xxxxxxxxxxx/1 genBasic([\"manufacturerName\",\"zclVersion\",\"appVersion\",\"modelId\",\"powerSource\",65534], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))","status":"error","transaction":"jl1vv-2"}'`

I tryed a few bit and things, and nothing worked, just what you said.

vbakker, 2022-06-24:
i tried it also but not on the edge version. so i skipped the tuya.js part.
i also replaced the name of the model switch in model: 'TS011F_plug_2',

now i have voltage, ampere but no kwh.

stanionascu, 2022-06-26:
My plug is BSD48 and according to the seller, the indicator mode was incorrectly advertised as supported on the marketplace. The Tuya DPIDs from the Tuya IoT portal are as [defined in Tuya docs](https://developer.tuya.com/en/docs/iot/product-standard-function-introduction?id=K9tp15ceh63gr).

Z2M db entry: https://pastebin.com/KPwq2UsP
Device link: https://www.aliexpress.com/item/1005002355897054.html

When trying to subscribe to `seMetering:currentSummDelivered`, the device replies back `UNREPORTABLE_ATTRIBUTE`, same with any of the `haElectricalMeasurement` attributes:
```
Zigbee2MQTT:error 2022-06-26 16:51:52: Failed to configure 'Test Plug', attempt 3 (Error: ConfigureReporting 0x.../1 seMetering([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Object.currentSummDelivered (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:119:9)
    at Object.configure (/app/dist/util/externally-loaded.js:34:4)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
``` 

The following external converter as bsd48.js seems to work fine.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');	
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
	{
		fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_r6buo8ba'}],
		model: 'TS011F_plug_3',
		description: 'Smart plug (with power monitoring)',
		vendor: 'TuYa',
		ota: ota.zigbeeOTA,
		fromZigbee: [
			fz.on_off,
			fz.electrical_measurement,
			fz.metering,
			fz.tuya_switch_power_outage_memory,
			fz.ts011f_plug_child_mode],
		toZigbee: [
			tz.on_off,
			tz.tuya_switch_power_outage_memory,
			tz.ts011f_plug_child_mode],
		configure: async (device, coordinatorEndpoint, logger) => {
			const endpoint = device.getEndpoint(1);
			await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
			endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
			endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
			device.save();
		},
		options: [exposes.options.measurement_poll_interval()],
		exposes: [
			e.switch(),
			e.power(),
			e.current(),
			e.voltage().withAccess(ea.STATE),
			e.energy(),
			exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
				.withDescription('Recover state after power outage'),
			e.child_lock()],
		onEvent: async (type, data, device, options) => {
			const endpoint = device.getEndpoint(1);
			if (type === 'stop') {
				clearInterval(globalStore.getValue(device, 'interval'));
				globalStore.clearValue(device, 'interval');
			} else if (!globalStore.hasValue(device, 'interval')) {
				const seconds = options && options.measurement_poll_interval ? options.measurement_poll_interval : 60;
				if (seconds === -1) return;
				const interval = setInterval(async () => {
					try {
						await endpoint.read('seMetering', ['currentSummDelivered']);
						await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
					} catch (error) {/* Do nothing */}
				}, seconds*1000);
				globalStore.putValue(device, 'interval', interval);
			}
		}
	}
];
```

[image]

iculookn, 2022-06-29:
> My plug is BSD48 and according to the seller, the indicator mode was incorrectly advertised as supported on the marketplace. The Tuya DPIDs from the Tuya IoT portal are as [defined in Tuya [...]

Thanks, this worked for my BSD48 plug. Mine was from AliExpress branded as ELIVCO
https://www.aliexpress.com/item/1005002675027731.html?spm=a2g0o.order_list.0.0.18751802Qa7StN

[image]

dsolva, 2022-07-09:
> @stanionascu 's external converter works perfectly for my _TZ3000_r6buo8ba too. Thanks! So it seems that this model is in fact the polled version? Then I guess the _TZ3000_r6buo8ba in the dev branch is placed incorrectly in the TS011F_plug_1?

I worked for me too, i also have the _TZ3000_r6buo8ba. I think it was released that way as well in the official branch as Im not part of dev and got that as TS011F_plug_1 instead of _3

vbakker, 2022-07-12:
> Does someone have the original TuYa gateway to make a sniff when pairing
> this device to it?
> https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

I have but how can I do that?

alex-bristol, 2022-07-17:
Hi Koenkk, I have also purchased three Elivco BSD48s on 26th May 2022 and just tried the .js configuration file listed above, this afternoon, controls work well, the consumption figures look about right, voltage reading spot on, the watts and amps a little low by about 20%.  Someone has done a good job, Well Done, whoever you are.  And, Well Done Keonkk, for all your time and effort supporting Zigbee2Mqtt, really appreciated.

alex-bristol, 2022-07-17:
Here is the item I purchased, for anyone interested: [Elivco Zigbee plug I use with Home Assistant, different part to item listed above](https://www.aliexpress.com/item/1005002688846037.html)

testos77, 2022-08-06:
I can confirm that the commit fixed the issue. Thx!

closed 2022-08-06 by testos77

fdkgenie, 2023-01-28:
Hi, I tried @stanionascu 's bsd48.js (https://github.com/Koenkk/zigbee2mqtt/issues/12760#issuecomment-1166575842) for my TuYa _TZ3000_okaz9tjs but was unsuccessful, after resetting Z2M it shows error with options of "measurement_poll_interval" that does not exist in my device. I still can poll current / voltage / power value in Dev console. So please help me to fix it. Many thanks !

[image]

[image]

[image]

stanionascu, 2023-01-28:
> Hi, I tried @stanionascu 's bsd48.js ([#12760 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/12760#issuecomment-1166575842)) for my TuYa _TZ3000_okaz9tjs but was unsuccessful, after resetting Z2M it shows error with options of "measurement_poll_interval" that does not exist in my device. I still can poll current / voltage / power value in Dev console. So please help me to fix it. Many thanks !
> 

It's likely a [TS011F_plug_3 (with power monitoring by polling)](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html), but it was detected as [TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html). So looks like the converter did not match the `manufacturerName`.

You might need to adapt the converter to match your device, so that it is identified as a plug_3.

Also a z2m db entry for the plug would be really helpful to compare the feature set to other plug versions.

Hope that helps!

geronimo76, 2023-02-10:
> It's likely a [TS011F_plug_3 (with power monitoring by polling)](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html), but it was detected as [...]

Hi, I have the same issue. Already had some smart plugs working (labeled BSD33 somewhere on the plug and reporting as _TZ3000_r6buo8ba on Z2M. I ordered some extra plugs, also labeled BSD33 somewhere on the plug, but these are reporting as _TZ3000_okaz9tjs on Z2M.
I tried the converter yesterday, but somehow Z2M failed to boot with the converter in place (probably my mistake...).
Anyway, here are the db entries: [smart-plugs.zip](https://github.com/Koenkk/zigbee2mqtt/files/10708773/smart-plugs.zip)
Two db entries, id7 is the functional _TZ3000_r6buo8ba, id8 is the non-functional _TZ3000_okaz9tjs.

Just to be clear, the _TZ3000_okaz9tjs can be used as switch, but no metering data is received in Z2M.