Koenkk/zigbee2mqtt #12761: [Feature request]: LiXee ZLinky_TIC adding STGE interprétation

navycrow, 2022-06-09:
### Is your feature request related to a problem? Please describe

No

### Describe the solution you'd like

STGE indicates status register as bytes string corresponding to a binary string
bits are described here in the enedis referential [Enedis-NOI-CPT_54E](https://www.enedis.fr/media/2035/download)

Interpretations are missing

### Describe alternatives you've considered

Here is my template
```
      - name: compteur_linky_status_register_interpretation
        state: >
          {% set STGE = states("sensor.compteur_linky_status_register") %}
          {# Convert byte string into binary string #}
          {% set STGE_binary = "{:b}".format(STGE[0]|int(base=16)).zfill(4) +
                              "{:b}".format(STGE[1]|int(base=16)).zfill(4) +
                              "{:b}".format(STGE[2]|int(base=16)).zfill(4) +
                              "{:b}".format(STGE[3]|int(base=16)).zfill(4) +
                              "{:b}".format(STGE[4]|int(base=16)).zfill(4) +
                              "{:b}".format(STGE[5]|int(base=16)).zfill(4) +
                              "{:b}".format(STGE[6]|int(base=16)).zfill(4) +
                              "{:b}".format(STGE[7]|int(base=16)).zfill(4) %}
          {{ STGE_binary }}
        attributes:
          Contact sec: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[31]|int(base=2) }}
          Organe de coupure: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[28:31]|int(base=2) }}
          État du cache-bornes distributeur: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[27]|int(base=2) }}
          Non utilisé (1): >
            {{ states("sensor.compteur_linky_status_register_interpretation")[26]|int(base=2) }}
          Surtension sur une des phases: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[25]|int(base=2) }}
          Dépassement de la puissance de référence: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[24]|int(base=2) }}
          Fonctionnement producteur/consommateur: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[23]|int(base=2) }}
          Sens de l’énergie active: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[22]|int(base=2) }}
          Tarif en cours sur le contrat fourniture: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[18:22]|int(base=2) }}
          Tarif en cours sur le contrat distributeur: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[16:18]|int(base=2) }}
          Mode dégradée de l’horloge (perte de l’horodate de l’horloge interne): >
            {{ states("sensor.compteur_linky_status_register_interpretation")[15]|int(base=2) }}
          État de la sortie télé-information: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[14]|int(base=2) }}
          Non utilisé (2): >
            {{ states("sensor.compteur_linky_status_register_interpretation")[13]|int(base=2) }}
          État de la sortie communication Euridis: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[11:13]|int(base=2) }}
          Statut du CPL: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[9:11]|int(base=2) }}
          Synchronisation CPL: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[8]|int(base=2) }}
          Couleur du jour pour le contrat historique tempo: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[6:8]|int(base=2) }}
          Couleur du lendemain pour le contrat historique tempo: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[4:6]|int(base=2) }}
          Préavis pointes mobiles: >
            {{ states("sensor.compteur_linky_status_register_interpretation")[2:4]|int(base=2) }}
          Pointe mobile (PM): >
            {{ states("sensor.compteur_linky_status_register_interpretation")[0:2]|int(base=2) }}
```

### Additional context

Could you add these attributes ?

Koenkk, 2022-06-09:
@vk496 @SilentT-FR could you check this?

vk496, 2022-06-09:
I didn't wanted implement this because there is a lot of attributes only when mode is standard (notice that in legacy mode there is no register attribute).

There is already problems with too many attributes and heavy size of HA DB because Zlinky floods a lot. So sending just the raw bytes of the register is enough in my opinion. As you showed, you could extract information you need directly in HA if needed.

Feel free to send PR if you think it is really useful and should be provided natively by Z2M. Or at least, people showing interest in it would be enough to consider me implement it.

BR,
Valentin

SilentT-FR, 2022-06-11:
It's simply conversion of a hexadecimal value to understandable text,.

Whether the process is done on Z2M does not shock me @Koenkk  an opinion ? or an example on another integration

closed 2022-07-20 by github-actions[bot]

afaucogney, 2022-08-08:
@navycrow hi, how do you get STGE from Z2M ? I do not see anything about that !

navycrow, 2022-08-27:
STGE apparait uniquement en mode standard.