Koenkk/zigbee2mqtt #12766: [Feature request]: Don't debounce first message

klada, 2022-06-09:
### Is your feature request related to a problem? Please describe

The Busch-Jaeger RB01 and RM01 devices send duplicte events on button presses:

```
info  2022-06-09 20:10:26: MQTT publish: topic 'zigbee2mqtt/sw_bj', payload '{"action":"on_row_1","linkquality":10}'
info  2022-06-09 20:10:26: MQTT publish: topic 'zigbee2mqtt/sw_bj', payload '{"action":"on_row_1","linkquality":3}'
info  2022-06-09 20:10:29: MQTT publish: topic 'zigbee2mqtt/sw_bj', payload '{"action":"off_row_1","linkquality":61}'
info  2022-06-09 20:10:29: MQTT publish: topic 'zigbee2mqtt/sw_bj', payload '{"action":"off_row_1","linkquality":61}'
info  2022-06-09 20:10:31: MQTT publish: topic 'zigbee2mqtt/sw_bj', payload '{"action":"on_row_1","linkquality":58}'
info  2022-06-09 20:10:31: MQTT publish: topic 'zigbee2mqtt/sw_bj', payload '{"action":"on_row_1","linkquality":58}'
```

This can be worked around by `debounce` and `debounce_ignore`. However this introduces a little lag, because right now `debounce` seems to queue messages for the entire debounce interval until it publishes a message.

### Describe the solution you'd like

It would be nice if we could get a better control on when to debounce. In this context an additional attribute `debounce_leading_edge=true|false` would be very helpful.

`debounce_leading_edge=false` would be the default and current behavior (debounce first and then publish message)

`debounce_leading_edge=true` would be the new behavior where a message would be published instantly and only further incoming messages of the same type are held back.

#10253 contains the required code for this, although I have not tested it.

### Describe alternatives you've considered

Right now I am using a fairly small `debounce` interval to work around the duplicate events, but this (obviously) introduces a lag.

### Additional context

There have been several requests for this feature and issues which could possibly be solved by this: #3677, #8139

McGiverGim, 2022-06-09:
Yes, I asked some time ago for the same. Very interested.

london89, 2022-09-21:
yes, this idea interesting for me, too

klada, 2022-11-23:
unstale

vandalon, 2023-03-24:
I am also missing this feature. May a debounce function switch that makes it switch between "wait for x time after first message" and "ignore for x time after first message" would be a good compromise?

klada, 2023-05-29:
bump

@Koenkk  Is it possible to disable the stale bot for some issues so we can stop playing this bump game?

copystring, 2023-11-20:
Is this the same problems as I have in this issue? https://github.com/Koenkk/zigbee2mqtt/issues/19454
@Koenkk do you have time to take a look at this, so debounce on leading edge would be available in z2m?

Koenkk, 2023-11-21:
Could you try to locally change https://github.com/Koenkk/zigbee2mqtt/blob/311ea070b167ddc6d8de1f8922aed0d8b38025d7/lib/extension/receive.ts#L41

to 

```
 }, time * 1000, {immediate: true}),
```

Run `npm run build` and see if that fixes the issue?

copystring, 2023-11-21:
No luck:

```
npm run build

> zigbee2mqtt@1.33.1-dev build
> tsc && node index.js writehash

lib/extension/receive.ts:41:33 - error TS2345: Argument of type '{ immediate: boolean; }' is not assignable to parameter of type 'boolean'.

41                 }, time * 1000, {immediate: true}),
                                   ~~~~~~~~~~~~~~~~~

Found 1 error in lib/extension/receive.ts:41
```

copystring, 2023-11-21:
`}, time * 1000, true),` is the correct code and fixed the issue for me.

copystring, 2023-11-22:
@Koenkk looking closer, this has not solved my issue, but I got more details.
For testing, I turned off and on my light switch via the frontend:

OFF via frontend:
```
Info 2023-11-22 08:55:32MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T08:55:31+01:00","linkquality":153,"power_on_behavior":"previous","state":"ON","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Info 2023-11-22 08:55:32MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T08:55:31+01:00","linkquality":153,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Info 2023-11-22 08:55:32MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T08:55:32+01:00","linkquality":153,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
```

ON via frontend:
```
Info 2023-11-22 08:55:34MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T08:55:34+01:00","linkquality":149,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Info 2023-11-22 08:55:34MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T08:55:34+01:00","linkquality":149,"power_on_behavior":"previous","state":"ON","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Info 2023-11-22 08:55:36MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T08:55:36+01:00","linkquality":149,"power_on_behavior":"previous","state":"ON","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
```

OFF without frontend directly on the physical light switch:
```
Info 2023-11-22 08:56:07MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T08:56:07+01:00","linkquality":149,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
```

ON without frontend directly on the physical light switch:
```
Info 2023-11-22 08:56:11MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T08:56:10+01:00","linkquality":149,"power_on_behavior":"previous","state":"ON","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
```

Koenkk, 2023-11-22:
Could you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

copystring, 2023-11-22:
```
Debug 2023-11-22 19:41:12Received MQTT message on 'zigbee2mqtt/LichtschalterHobbyraum/set' with data '{"state":"OFF"}'
Debug 2023-11-22 19:41:12Publishing 'set' 'state' to 'LichtschalterHobbyraum'
Info  2023-11-22 19:41:13MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T19:41:13+01:00","linkquality":76,"power_on_behavior":"previous","state":"ON","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Info  2023-11-22 19:41:13MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T19:41:13+01:00","linkquality":76,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Debug 2023-11-22 19:41:13Received Zigbee message from 'LichtschalterHobbyraum', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info  2023-11-22 19:41:13MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T19:41:13+01:00","linkquality":80,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Debug 2023-11-22 19:41:13Received MQTT message on 'zigbee2mqtt/LichtschalterHobbyraum/set' with data '{"state":"ON"}'
Debug 2023-11-22 19:41:13Publishing 'set' 'state' to 'LichtschalterHobbyraum'
Info  2023-11-22 19:41:13MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T19:41:13+01:00","linkquality":76,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Info  2023-11-22 19:41:13MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T19:41:13+01:00","linkquality":76,"power_on_behavior":"previous","state":"ON","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Debug 2023-11-22 19:41:14Received Zigbee message from 'LichtschalterHobbyraum', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0

Debug 2023-11-22 19:41:26Received Zigbee message from 'LichtschalterHobbyraum', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info  2023-11-22 19:41:26MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T19:41:26+01:00","linkquality":83,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Debug 2023-11-22 19:41:29Received Zigbee message from 'LichtschalterHobbyraum', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info  2023-11-22 19:41:29MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T19:41:29+01:00","linkquality":87,"power_on_behavior":"previous","state":"ON","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'

```

Koenkk, 2023-11-22:
These is published because you have the last seen option enabled in your `configuration.yaml`

copystring, 2023-11-22:
Enabling last seen makes z2m publish a false state? Seems odd.

copystring, 2023-11-22:
Check this one for example. Sent OFF via the frontend. z2m publishes ON and then OFF twice. Why publish ON when OFF is sent? Seems like a bug to me. Having last seen enabled should not make a device publish ON after OFF has been sent. This is exactly what messes up my triggers.
```
Debug 2023-11-22 22:03:00Received MQTT message on 'zigbee2mqtt/LichtschalterHobbyraum/set' with data '{"state":"OFF"}'
Debug 2023-11-22 22:03:00Publishing 'set' 'state' to 'LichtschalterHobbyraum'
Info 2023-11-22 22:03:01MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T22:03:01+01:00","linkquality":43,"power_on_behavior":"previous","state":"ON","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Info 2023-11-22 22:03:01MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T22:03:01+01:00","linkquality":43,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
Debug 2023-11-22 22:03:01Received Zigbee message from 'LichtschalterHobbyraum', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-11-22 22:03:01MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T22:03:01+01:00","linkquality":21,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'
```

Koenkk, 2023-11-23:
To confirm, does it work after disabling last seen?

copystring, 2023-11-23:
Yes, I can confirm this does not happen with last seen disabled.

[image]

Koenkk, 2023-11-24:
Then the behaviour is correct.

```
// Here you send a request to turn the light OFF, the last known state of z2m is ON
Debug 2023-11-22 22:03:00Received MQTT message on 'zigbee2mqtt/LichtschalterHobbyraum/set' with data '{"state":"OFF"}'
Debug 2023-11-22 22:03:00Publishing 'set' 'state' to 'LichtschalterHobbyraum'

// The command is successful, meaning that the device is online, therefore last seen is published with the last known state.
// As you debounce the state, it's is still ON at this point
Info 2023-11-22 22:03:01MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T22:03:01+01:00","linkquality":43,"power_on_behavior":"previous","state":"ON","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'

// Debounce expires, state OFF is published, last known state by z2m is OFF now
Info 2023-11-22 22:03:01MQTT publish: topic 'zigbee2mqtt/LichtschalterHobbyraum', payload '{"last_seen":"2023-11-22T22:03:01+01:00","linkquality":43,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4110,"latest_version":4110,"state":"idle"}}'

```

copystring, 2023-11-24:
Thank you. I'll adjust my scripts accordingly.