Koenkk/zigbee2mqtt #12768: Aqara flood sensor, temperatur is now device temperatur and value wrong

umtauscher, 2022-06-09:
### What happened?

Updating homeassistant zigbee2mqtt
Flood sensor SJCGQ11LM xiaomi temperature change to device temperatur but the value of this sensor is now at least 7 degrees higher than before the update.
I don‘t mind the change of sensor name but the measure temperature should not change.

### What did you expect to happen?

I expected the temperature to be correct

### How to reproduce it (minimal and precise)

1.25.1-2 installed, updated to 1.25.2-1

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

20220219

### Adapter

Sonoff zigbee 3.0 stick

Koenkk, 2022-06-11:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/12700, `temperature` is now `device_temperature`, you have to remove the old `temperature` from `data/state.json` yourself.

closed 2022-06-11 by Koenkk

umtauscher, 2022-06-11:
Thanks for you answer.
But I think it doesn‘t reflect the problem.
The temperatur- despite form the name change - jumps from 7 to 13 degrees C after the update. Since this is a device that doesn‘t relay consume any energy, I cannot believe the chip temperature differs that much from the  climate temperature.
Sincerely

Wilhelm Speck 

Wilhelm Speck

commedia, Wilhelm Speck
Strandläuferweg