Koenkk/zigbee2mqtt #12769: [New device support]: TuYa Smoke detector TS0601  _TZE200_ntcy3xu1

madspoulsen, 2022-06-09:
### Link

https://www.aliexpress.com/item/1005004069930404.html?spm=a2g0o.order_list.0.0.52381802E42VDj

### Database entry

{"id":45,"type":"EndDevice","ieeeAddr":"0x00124b0025131f6c","nwkAddr":59271,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"DS01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"DS01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":5,"hwVersion":1,"dateCode":"20211103"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0026b7409d","zoneState":1}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b7409d","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":5,"hwVersion":1,"dateCode":"20211103","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1654805059615,"defaultSendRequestWhen":"immediate"}

### Comments

So I just adjusted one of the other examples, and now it works fine with Smoke detection, Tamper, and Battery Low.
Except from battery status, on the Z2M "devices" page, it is listed as 2% with blinking red. That annoys me a bit..
The smoke detector sends "2" when battery is OK and "0" when battery is low. Nothing in between AFAIK I can see.

FYI: In the converter below, I didn't adjust the "zigbeemodel". Don't know where it plays a role..

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    tuya_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.state:
                    result.smoke = value === 0;
                    break;
                case 4:
                    result.tamper = value; 
                    break
                case 14:
                    result.battery_low = value === 0;
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${ dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    }
}

const definition = {
    zigbeeModel: ['5p1vj8r'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ntcy3xu1'}],
    model: 'TS0205',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [fzLocal.tuya_smoke],
    toZigbee: [],
    exposes: [e.smoke(), e.battery_low(), e.tamper()],
};

module.exports = definition;
```

callacomp, 2022-06-10:
can't believe this, just done this myself, my first one. As you've said not sure if battery low indicates anything between 2 and 0. I've included 1, but this could be removed so that you at least get a high and low battery level, so automations for battery still work, you could add this? I've done the tamper slightly different as it's not binary. 

The "battery: {0: "10", 1: "50", 2: "75"}[value]," could just be "battery: {0: "10", 2: "75"}[value]," at least then you can have automations fire with batteries less than 10%, if this is allowed?

I don't know how to add this mind to the GitHub mind!
```
    tuya_smoke_2: {                                                                                                                   
        cluster: 'manuSpecificTuya',                                                                                         
        type: ['commandDataResponse', 'commandDataReport'],                                                                          
        convert: (model, msg, publish, options, meta) => {                                                                           
            const result = {};                                                                                                           
            for (const dpValue of msg.data.dpValues) {                                                                
                const dp = dpValue.dp;                                                                     
                const value = tuya.getDataValue(dpValue);                                                                               
                switch (dp) {                                                                                             
                case tuya.dataPoints.state:                                                                                      
                    result.smoke = value === 0;                                                                           
                    break;                                                                               
                case 4:                                                                                                           
                    return {tamper: value > 0 ? true : false};                                                                 
                    break;                                                                                                              
                case 14:                                                                                                               
                    return {                                                                                                                
                        battery: {0: "10", 1: "50", 2: "75"}[value],                                                                        
                        battery_low: value === 0,                                                                                           
                    };                                                                                                                      
                    break;                                                                                                                  
                default:                                                                                                       
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${ dp} with data ${JSON.stringify(dpValue)}`);
                }                                                                                                                           
            }                                                                                                                               
            return result;                                                                                                                  
        },                                                                                                                                  
    },

```
along with this
```
const definition = {
    zigbeeModel: ['ntcy3xu1'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ntcy3xu1'}],
    model: 'TS0601_smoke',
    vendor: 'TuYa',
    description: 'Smoke sensor', 
    fromZigbee: [fz.tuya_smoke_2], 
    toZigbee: [], 
    onEvent: tuya.onEventSetTime, // not sure if this is needed?
    exposes: [e.smoke(), e.battery(), e.battery_low(), e.tamper()],
};

module.exports = definition;
```

callacomp, 2022-06-10:
Can you put all of the file you've shown into the external js file? I put the fzlocal stuff into fromzigbee file?

callacomp, 2022-06-11:
I've just tested the power with a variable power supply and the battery value goes from 2 to 0 when it's roughly under 2.35 Volts. There is no number 1. So for me I've set it as 
```
                case 14:
                    return {
                        battery: {0: "10", 2: "75"}[value],
                        battery_low: value === 0,                                                                                           
                    }; 
```

FHannes, 2022-06-14:
Nice, though I already provided an implementation in an earlier issue: https://github.com/Koenkk/zigbee2mqtt/issues/12622

SamMousa, 2022-07-02:
Can we close this as a duplicate of the issue mentioned above? We now have 2 base converter implementations (they are practically the same) as well as one variation of each.

We should put all information in one thread so we can move towards a clean implementation that can lead to native device support.

Koenkk, 2022-07-02:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/12622

closed 2022-07-02 by Koenkk