Koenkk/zigbee2mqtt #12772: Lidl remote (FB20-002) not working

fplanjer, 2022-06-10:
### What happened?

I paired the remote (Lidl FB20-002) and it is visible in the frontend. When binding it to a ligt or a group pressing the buttons (on/off, brightness) is not reflecting on the lights.
[image]

### What did you expect to happen?

I expected the lights to turn on or off or the brightness to be changed

### Zigbee2MQTT version

1.25.2 commit: 3c5854f

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

Database.db: 
`{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024c8dcae","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"}
{"id":2,"type":"Router","ieeeAddr":"0x804b50fffea3cb58","nwkAddr":40531,"manufId":4098,"manufName":"_TZ3000_kdi2o9m6","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,11,242],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_kdi2o9m6","zclVersion":3}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8dcae","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1654851072048,"defaultSendRequestWhen":"immediate"}
{"id":3,"type":"Router","ieeeAddr":"0x842e14fffe02cba9","nwkAddr":55913,"manufId":4098,"manufName":"_TZ3000_el5kt5im","powerSource":"Mains (single phase)","modelId":"TS0502A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502A","manufacturerName":"_TZ3000_el5kt5im","stackVersion":0,"dateCode":"","zclVersion":3}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorMode":2,"colorTemperature":351}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":29}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8dcae","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8dcae","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1654851473948,"defaultSendRequestWhen":"immediate"}
{"id":4,"type":"Router","ieeeAddr":"0x842e14fffe25b8fc","nwkAddr":35607,"manufId":4098,"manufName":"_TZ3000_el5kt5im","powerSource":"Mains (single phase)","modelId":"TS0502A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502A","manufacturerName":"_TZ3000_el5kt5im","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorMode":2,"colorTemperature":428}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":43}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1654851455142,"defaultSendRequestWhen":"immediate"}
{"id":5,"type":"Router","ieeeAddr":"0x588e81fffed12ba8","nwkAddr":38237,"manufId":4098,"manufName":"_TZ3000_el5kt5im","powerSource":"Mains (single phase)","modelId":"TS0502A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502A","manufacturerName":"_TZ3000_el5kt5im","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorMode":2,"colorTemperature":428}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":43}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1654851455901,"defaultSendRequestWhen":"immediate"}
{"id":6,"type":"EndDevice","ieeeAddr":"0xbc33acfffe4bf359","nwkAddr":10580,"manufId":4098,"manufName":"_TYZB01_bngwdjsr","powerSource":"Battery","modelId":"TS1001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS1001","manufacturerName":"_TYZB01_bngwdjsr","stackVersion":0,"dateCode":""}}},"binds":[{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x842e14fffe02cba9","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x842e14fffe02cba9","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x842e14fffe02cba9","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":81,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1654851520079,"defaultSendRequestWhen":"immediate"}
[... cut ...]
info  2022-06-10 11:11:42: MQTT publish: topic 'zigbee2mqtt/ZolderRemote/action', payload 'brightness_step_up'
info  2022-06-10 11:11:42: MQTT publish: topic 'zigbee2mqtt/ZolderRemote', payload '{"action":"brightness_step_down","action_step_size":51,"action_transition_time":0.1,"brightness":153,"linkquality":76}'
info  2022-06-10 11:11:42: MQTT publish: topic 'zigbee2mqtt/ZolderRemote', payload '{"action":"","brightness":153,"linkquality":76}'
info  2022-06-10 11:11:42: MQTT publish: topic 'zigbee2mqtt/ZolderRemote/action', payload 'brightness_step_down'
`

esatemre, 2022-07-16:
Same setup, same issue here as well.

Koenkk, 2022-07-17:
My guess is that this device does not support group binding, to confirm someone should sniff the traffic after binding it (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531). Chance that this can be fixed is low.

closed 2022-08-24 by github-actions[bot]

toastking42, 2024-06-16:
Hello, the issue still persists. The LIDL remote can successfully be bound to a LIDL ceiling lamp, but pressing the buttons on the remote has no effect. It used to work just fine with the old LIDL Zigbee Base (I switched to HA recently). Has anyone figured this out, or are the LIDL remotes now useless?

Schlunze, 2024-07-14:
i got "action:" "on" (on button) working using 1.39.0-dev commit: [bd78f56]
and i can see it in homeassistant logs:)

brightness up / down and off button don´t work

Log Webinterface (on button):
```
info 2024-07-14 11:53:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Fernbedienung', payload '{"action":"on","linkquality":112}'
info 2024-07-14 11:53:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Fernbedienung', payload '{"action":"","linkquality":112}'
info 2024-07-14 11:53:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Fernbedienung/action', payload 'on'
```
i think the other buttons cause action in debug protocols but not show up in the webinterface logs as action.
it is exactly the device [FB20-002](https://www.zigbee2mqtt.io/devices/FB20-002.html) shown in this picture.

oxodao, 2024-07-21:
> i got "action:" "on" (on button) working using 1.39.0-dev commit: [[bd78f56](https://github.com/Koenkk/zigbee2mqtt/commit/bd78f56d5ff3b56c9f2c583237909a673e43183d)] and i can see it in homeassistant logs:)
> 
> brightness up / down and off button don´t work
> 
> Log Webinterface (on button):
> 
> ```
> info 2024-07-14 11:53:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Fernbedienung', payload '{"action":"on","linkquality":112}'
> info 2024-07-14 11:53:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Fernbedienung', payload '{"action":"","linkquality":112}'
> info 2024-07-14 11:53:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Fernbedienung/action', payload 'on'
> ```
> 
> i think the other buttons cause action in debug protocols but not show up in the webinterface logs. They have no action payload in z2m. it is exactly the device [FB20-002](https://www.zigbee2mqtt.io/devices/FB20-002.html).

I have the same issue on this device: https://www.zigbee2mqtt.io/devices/FB21-001.html

Its the HG08376 for me but it looks like exactly the same.

Button off / brightness up / brightness down do not send anything to MQTT

Button on send this:
```json
{
    "action": "switch_scene",
    "action_scene": 1,
    "linkquality": 188
}
```

Note that the action_scene starts at 0, and goes up at every press. It then rolls back to 0 when reaching 3 (4 stage)