Koenkk/zigbee2mqtt #12773: [New device support]: KlikAanKlikUit (Trust) ZSDR-850

Step-han, 2022-06-10:
### Link

https://klikaanklikuit.nl/product/slimme-draadloze-rookmelder/

### Database entry

{"id":59,"type":"EndDevice","ieeeAddr":"0x000d6f001698e4c6","nwkAddr":28028,"manufId":4619,"manufName":"Trust","powerSource":"Battery","modelId":"SmokeSensor-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SmokeSensor-EM","manufacturerName":"Trust","powerSource":3,"zclVersion":1,"appVersion":21,"stackVersion":2,"hwVersion":16,"dateCode":"2020.10.14","swBuildId":"1.1.1"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00215ffec5","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00215ffec5","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":21,"stackVersion":2,"hwVersion":16,"dateCode":"2020.10.14","swBuildId":"1.1.1","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1654848967653,"defaultSendRequestWhen":"immediate"}

### Comments

Its paired as a HEIMAN HS3SA:

[image]

[image]

[image]

In HomeAssistant:
[image]

Its working but i dunno if you want to support it as a ZSDR-850.

If you need something let me know.

Koenkk, 2022-06-11:
Done!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-06-11 by Koenkk

asychev, 2022-06-30:
@Koenkk I think it is actually [HEIMAN HS1SA](https://www.zigbee2mqtt.io/devices/HS1SA.html), not [HS3SA](https://www.zigbee2mqtt.io/devices/HS3SA.html), at least visually

Koenkk, 2022-07-01:
@asychev I don't understand what you mean, it will be recognised as a Trust ZSDR-850 now, not HEIMAN HS1SA

asychev, 2022-07-01:
@Koenkk Yep, after update to [1.26.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.26.0) it is now properly detected, thanks!

osker2k, 2022-09-20:
Hi,

Sorry for my bad English.  I bought 5 of these devices and connected them successfully too.  Unfortunately I don't see a way if one triggers the other ones too.  Is it possible that the alarm can also be triggered as an action?

lukasmedia, 2023-09-01:
I have the same issue as @osker2k
Also the battery level is not showing on the dashboard; only at zigbee devices. How to fix this?

[image]

Note: the slider above is not working. Smoke detection is.

TheoSl93, 2023-09-10:
I'mj having the same issue @osker2k, I would like to trigger all the sirens if one of the sensors detects smoke. Is it possible to do it?

lukasmedia, 2023-11-16:
> I'mj having the same issue @osker2k, I would like to trigger all the sirens if one of the sensors detects smoke. Is it possible to do it?

Hi Theo, no this is not possible with these kind of smoke detectors. These are "1 way" and you (we) need smoke detectors which are bi directional.

piwy84, 2024-07-17:
Is it worth conneting these kaku smoke detectors via zigbee2mqtt?
Mine are currently  connected with zigbee (sonoff3.0dongle)
Will i have more options if i move them to zigbee2mqtt? (Like triggering the alarms with an automation)