Koenkk/zigbee2mqtt #12774: QS-Zigbee-CP03: intermittent error: 'No network route' (205)

cunode, 2022-06-10:
### What happened?

When communicating with a QS-Zigbee-CP03 curtain module I frequently get the following error message:
`error 2022-06-10 07:01:34: Publish 'set' 'state' to 'Store SZ Sued' failed: 'Error: Command 0xa4c1385b260f0939/1 closuresWindowCovering.upOpen({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`

Usually, when resubmitting the same command multiple times, the device finally executes it as expected. But there is a 50% chance to fail and hence my automation is absolutely unreliable.

Based on the finding in #11539 (QS-Zigbee-CP03) I switched on May 13, 2022 to the Zigbee2MQTT dev branch for Linux. I also use the latest firmware for my SLAESH coordinator (CC2652RB_coordinator_20220219.hex) together with a 3 dB antenna connected to a Raspberry Pi 4 running Debian GNU/Linux 11 (bullseye).

### What did you expect to happen?

Communication should be reliable as there are short distances to the coordinator (~6 meter) or the next router (~3 meter). The link quality in the Zigbee2MQTT Map never shows zero and the device's Availability is shown as "Online".
Any help is highly appreciated.

### How to reproduce it (minimal and precise)

Problem is intermittent and hence cannot reproduce with clear steps.

### Zigbee2MQTT version

Zigbee2MQTT version 1.25.1-dev (commit #bc5fd1a5)

### Adapter firmware version

CC2652RB_coordinator_20220219.hex

### Adapter

CC2652RB development stick

torwanbukaj, 2022-06-11:
Similar story here. It behaves strange. In my case, it is possible that the device "hangs" too far away from the coordinator (linkquality ~7 when it sends movement updates) but I am not sure. I brought a Zigbee socket outlet (linkquality ~56) 3m away from it (to work as an mesh repeater) but it does not help.

`Zigbee2MQTT:error 2022-06-11 19:37:09: Publish 'set' 'state' to '0xa4c138f39ace9565' failed: 'Error: Command 0xa4c138f39ace9565/1 closuresWindowCovering.stop({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

Zigbee2MQTT:error 2022-06-11 19:37:21: Publish 'set' 'state' to '0xa4c138f39ace9565' failed: 'Error: Command 0xa4c138f39ace9565/1 closuresWindowCovering.upOpen({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

Zigbee2MQTT:error 2022-06-11 19:37:33: Publish 'set' 'state' to '0xa4c138f39ace9565' failed: 'Error: Command 0xa4c138f39ace9565/1 closuresWindowCovering.stop({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`

torwanbukaj, 2022-06-12:
I flashed my old CC2531 with router firmware and put it 2m away from my Lonsonho QS-Zigbee-C03. I also flashed my SLAESH coordinator with CC2652RB_coordinator_20220219.hex. No improvements. For a moment the linkquality jumped from 7-14 to 130-140 and the curtain could be operated remotely but my happiness did not last longer than 5 mins. It went back to linkquality 5-15 even if a router (CC2135) is hanging 2m away from the curtain module. Let me stress that the router is reporting linkquality 130-140 so quite good. Lonsonho QS-Zigbee-C03 located 2m away from it reports linkquality which is 10 times smaller.

galambert75, 2022-06-19:
Same issue here...

System:

- Z2M: 1.25.2-1 (Home Assistant Add-on)
- Sonoff ZigBee 3.0 USB Dongle (CC1352P2_CC2652P_launchpad_coordinator_20220219)
- RaspberryPI 4 Model B 8GB
- Home Assitant: core-2022.6.6, supervisor-2022.05.3, OS 8.2
- 7 QS-Zigbee-CP03, all connected directly to the coordinator (I think... When I try to load the Map, it's stuck searching). LQI on the devices ranges from 91 to 10. 

I get the _**'No network route' (205))'**_ on all seven of the devices, but it happens more often on the ones with lower LQI. 

What's interesting is that Z2M receives state events, even when it can't send commands. The following is a combination of the Z2M Log and MQTT Log. You can see MQTT is aware that I manually closed and opened the cover, even though it's not working from HA:

`Zigbee2MQTT:error 2022-06-19 18:18:56: Publish 'set' 'state' to 'P2 Habitacion Matrimonio Persiana' failed: 'Error: Command 0xa4c138d5c1078887/1 closuresWindowCovering.downClose({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:error 2022-06-19 18:19:08: Publish 'set' 'state' to 'P2 Habitacion Matrimonio Persiana' failed: 'Error: Command 0xa4c138d5c1078887/1 closuresWindowCovering.stop({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:error 2022-06-19 18:19:20: Publish 'set' 'state' to 'P2 Habitacion Matrimonio Persiana' failed: 'Error: Command 0xa4c138d5c1078887/1 closuresWindowCovering.upOpen({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
18:19:27 - P2 Habitacion Matrimonio Persiana moving changed to DOWN
18:19:56 - P2 Habitacion Matrimonio Persiana moving changed to STOP
18:19:57 - P2 Habitacion Matrimonio Persiana was closed
18:38:00 - P2 Habitacion Matrimonio Persiana moving changed to UP
18:38:01 - P2 Habitacion Matrimonio Persiana was opened
18:38:29 - P2 Habitacion Matrimonio Persiana moving changed to STOP
Zigbee2MQTT:error 2022-06-19 18:41:23: Publish 'set' 'state' to 'P2 Habitacion Matrimonio Persiana' failed: 'Error: Command 0xa4c138d5c1078887/1 closuresWindowCovering.upOpen({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:error 2022-06-19 18:41:34: Publish 'set' 'state' to 'P2 Habitacion Matrimonio Persiana' failed: 'Error: Command 0xa4c138d5c1078887/1 closuresWindowCovering.upOpen({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
[image]
`

There have been a few issues posted with 'No network route' (205). There's even a report (#12468) of the issue being fixed by switching from a Sonoff to a Slaesh stick. However, we can rule that out based on @[torwanbukaj](https://github.com/torwanbukaj) experience.

torwanbukaj, 2022-06-19:
Just a little update from my side. I dismantled my module and switched to a WiFi-driven which I tasmotized. So far so good. In the meantime, out of curiosity I "hanged" my QS-Zigbee-CP03 a few meters from the coordinator and everything was working fine for a few hours (linkquality good>140, the device fully communicative, no lags etc). Then I disconnected it and it's waiting for better times. So for me, it looks like the device struggle in a scenario with routers, hanging far away from the coordinator. I do not know if you noticed, it sends (when it sends something) some "backlight" attributes. This initially integrated even in my Home Assistant but disappeared after HA image update. It suggests that the firmware may be coming from some other device (most likely Tuya in this case). Maybe something was wrongly ported by the manufacturer (firmware)?

cunode, 2022-06-21:
I still have hope in my QS-Zigbee-CP03. And as you can see above, the problem is intermittent which could indicate poor signal quality despite the reasonably high LQI. I would love to debug the communication between the coordinator or any router and the CP03, I suspect the problem could also be a timeout which kicks in too early. Any help on how to debug would be appreciated.

galambert75, 2022-06-21:
> I still have hope in my QS-Zigbee-CP03. And as you can see above, the problem is intermittent which could indicate poor signal quality despite the reasonably high LQI. I would love to debug the [...]

Yes, I'd also prefer not to replace the 10 devices I bought. Unfortunately I'm not much of a programmer, so I can't help with the debugging.
I was thinking about testing the devices with ZHA instead of Z2M to check if the problem persists. At least that would help narrow down the issue. I'll see if I can make some time over the weekend.

cunode, 2022-06-23:
I turned on debugging in my configuration.yaml
`  log_level: debug`
and amended some device parameters (retain, qos, and retention):
`'0xa4c1385b260f0939':
    friendly_name: Store SZ Sued
    description: '0xa4c1385b260f0939'
    retain: true
    qos: 2
    retention: 60
    homeassistant: {}
    legacy: false`

Unfortunately, no significant improvement. When I do a network scan, the log file shows:

`warn  2022-06-23 20:45:03: Failed to ping 'Store SZ Sued' (attempt 1/1, Read 0xa4c1385b260f0939/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))`
`debug 2022-06-23 20:45:06: Active device 'Store SZ Sued' was last seen '67.50' minutes ago.`
`warn  2022-06-23 20:45:09: Failed to ping 'Store SZ Sued' (attempt 1/1, Read 0xa4c1385b260f0939/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))`

cunode, 2022-06-23:
Without any change in the configuration or the setup, an hour later, the device is seen again:
`debug 2022-06-23 21:41:50: Received Zigbee message from 'Store SZ Sued', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0`
`debug 2022-06-23 21:41:50: Device 'Store SZ Sued' reconnected`
`info  2022-06-23 21:41:50: MQTT publish: topic 'zigbee2mqtt/Store SZ Sued/availability', payload 'online'`
`info  2022-06-23 21:41:50: MQTT publish: topic 'zigbee2mqtt/Storen/availability', payload 'online'`

magostinelli, 2022-06-28:
I have the same problem.

magostinelli, 2022-06-30:
I found a trick, start a map creation, the device start to work again. If you attach a physical button it also works.

lukeIam, 2022-07-03:
For me I can't control my IKEA DRIVER (kitchen light) for about a week anymore...
Restarted the PI, started a map creation (here it was shown directly connected to coordinator with a good link) -> but still can't control it.
Also my TRADFRI on/off switch does not longer work. Other devices (Hue, Osram, Aqara, TRADFRI bulbs) work fine.

```
Debug 2022-07-03 20:58:20Received MQTT message on 'zigbee2mqtt/ikea_licht_kuechenschrank/set' with data '{"state":"ON"}'
Debug 2022-07-03 20:58:20Publishing 'set' 'state' to 'ikea_licht_kuechenschrank'
Error 2022-07-03 20:58:43Publish 'set' 'state' to 'ikea_licht_kuechenschrank' failed: 'Error: Command 0x804b50fffe4b627d/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Debug 2022-07-03 20:58:43Error: Command 0x804b50fffe4b627d/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Info 2022-07-03 20:58:43MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'ikea_licht_kuechenschrank' failed: 'Error: Command 0x804b50fffe4b627d/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ikea_licht_kuechenschrank"},"type":"zigbee_publish_error"}'
```

nickles-lee, 2022-07-03:
I'm observing similar issues on my smlight-slzb-05, if that is at all helpful. 

`Publish 'set' 'position' to 'Living Room Main Curtain' failed: 'Error: Write 0x04cf8cdf3c7396f2/1 genAnalogOutput({"85":{"value":75,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`

galambert75, 2022-07-04:
> I found a trick, start a map creation, the device start to work again. If you attach a physical button it also works.

What do you mean with "attach a physical button"?

magostinelli, 2022-07-06:
> > I found a trick, start a map creation, the device start to work again. If you attach a physical button it also works.
> 
> What do you mean with "attach a physical button"?

I connected a switch to the device, so if you press the button (open or close) it works like a charm.

cunode, 2022-07-06:
Connecting a switch to the device is like a passive open/close button and yes, this of course works. But this has nothing to do with automation. I don't want to activate a manual switch whenever my automation triggers the device.

fvillena, 2022-07-07:
Same issue here with TS0011_switch_module. The problem appears when it is connected through TS110E_1gang router. I receive status updates but I cannot set states ('No network route' (205) error). When it is connected directly to the coordinator it works fine.

cunode, 2022-07-07:
Made the same observation: when connected directly to the coordinator and having an acceptable link quality (LQI) all works fine. But if only a router device with good link quality connects to the PC-03 I cannot send shutter commands, it then always gives the 'No network route' (205) error.

Koenkk, 2022-07-07:
Could someone make a sniff of this starting from the point where it works (control it a few times) until it stops working? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

galambert75, 2022-07-07:
> Could someone make a sniff of this starting from the point where it works (control it a few times) until it stops working? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

I don't have a CC2531. Is it doable with a SONOFF ZigBee 3.0 USB Dongle? If so, I could give it a shot. I have a couple of devices that seem to fail very consistently...

Koenkk, 2022-07-07:
@galambert75 currently there is no sniffer fw available for the sonoff stick.

fvillena, 2022-07-07:
> Same issue here with TS0011_switch_module. The problem appears when it is connected through TS110E_1gang router. I receive status updates but I cannot set states ('No network route' (205) error). When it is connected directly to the coordinator it works fine.

Another clue: If I connect TS0011_switch_module through TS0505B router it works without issues. I think TS110E_1gang is not working properly as a router.

galambert75, 2022-07-12:
> @galambert75 currently there is no sniffer fw available for the sonoff stick.

So... is there anyone out there who owns a CC2531 and could provide the requested "sniff"?

cunode, 2022-07-12:
I' afraid my CC2652RB cannot be used instead of a CC2531 as sniffer.

Want to summarize my observations so far:
1. in the _Devices_ list, the C03 is always reported as "Online" with a reasonable _LQI_ (e.g. "20" currently). The _LQI_ is changing frequently even though the physical configuration is not changing.
2. in the _Map_, the C03 is as always reported as connected to the _Coordinator_ as well as to 2 _Routers_ in my network. When getting the 'No network route' (205) error I usually have only 1 Link quality number (e.g. "16"), when it is working I have 2 Link quality numbers (e.g. "20/24")
3. When using the manual up/down buttons I have connected to the device, the _Position_ information is always updated
4. a second C03 in the same room as the _Coordinator_ works 100% reliable. To exclude a hardware issue I swapped this device with the one working unreliable in the room above (separated by a concrete floor). Result was the same: device in same room as _Coordinator_ works now perfectly, the one in the room above is unreliable.

mpuff, 2022-07-17:
i have now the same problem, I had problems to pair a ikea fyrtur, than I connected and disconnected a few times a ikea signal repeater, than this routing problem was starting at my side :/

yavinmaster, 2022-07-24:
Started to get this error also a couple of days ago.  Tried updating things and still doesn't work.  My temperature and humidity sensors are reporting data but I can't control lights and switched any more.

Brkie, 2022-07-29:
Having same issue. Updated firmware on zzh!, updated HA, updated Zigbee2mqtt, moved router away from my rasp. Pi, restart, reboot everything - not working. Unplugged my IKEA socket, plugged in again. Everything works.... Wierd.. If you still havin problem, try to unplugg something and plug in again and see if it helps.

cunode, 2022-08-01:
Again, since days have no connection to the QS-Zigbee-CP03 even it is in the same room as my CC2652RB coordinator. Persistently getting the 'No network route' (205) error. Have some additional information in the log file:
`debug 2022-08-01 13:25:10: Error: Command 0xa4c1385b260f0939/1 closuresWindowCovering.downClose({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)`

Also noticed that even all my QS-Zigbee-CP03 are shown as Routers, none has a network connection to its neighbor  QS-Zigbee-CP03 even they are separated a few centimeters only.

wrichter, 2022-08-08:
I‘m seeing the same problem. Adding QS-Zigbee-CP03 to the network works as long as they are connected directly to the coordinator (SONOFF ZigBee 3.0 USB Dongle), once they are only connected to other routers I get 205 errors.

ACJUK, 2022-08-15:
I started getting this a couple of weeks ago.  CC2652R.  Devices won't respond under HA but work fine on Node-RED automations.  Walk into the bedroom, one light comes on instead of 2.  'No network route' (205)) on the one that doesn't work.  Walk out.  Both lights go out as they should.  Walk back in.  The other light works this time.  This is on a system that's been working fine for 18 months. Checked wifi last night and no overlaps.

macgyverek, 2022-08-29:
I have the same problem with  Tuya TS0001 and TS0002. I spend many hours on looking the problem. 

Publish 'set' 'state' to 'Tuya Hall Relay 1' failed: 'Error: Command 0xa4c138522e49c56c/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

But I can read a state of light, when I am used physical button connected to the device.  Its working like one direction. Problem is only when are two devices on network. With one devices working without problem. I thinking that problem is when few devices make a connection between two routers. Maybe its a Tuya problem or Zigbee2Mqtt. 

I will change this zigbee devices to WiFi module - Tasmota.

sk8er000, 2022-09-04:
I'm having the same random problem from Moes MS-104BZ and Lonsonho QS-Zigbee-S05-LN. What I noticed is that when this error occurs the device is not directly connected with the coordinator (they don't have any end device connected since they are connected with IKEA and Aqara routers). If I repair the module again it start working again for a random amount of time, it can be hours or days. Unfortunately it's not easy to find any other 2 gang switch with the same dimesions (or smaller) that are know to always work

ACJUK, 2022-09-04:
After much messing about removing devices and adding them back, I bit the bullet and changed ZigBee channels from 11 to 25. Everything is 100% stable now. 

Would it be more useful if it was NOT ch11 by default?

macgyverek, 2022-09-04:
I changed channel too from default 11 to 25. Problem still exists, but it's little better. On 11 channel I had communication lost after 1 day. On channel 25 working about 2 days.

sk8er000, 2022-09-04:
O used ch25 since day one because before setting the ZigBee network I've scanned multiple times the WiFi areas around me and the channel 25 seemed the best.
I had a pretty stable network for about a month than the error started to occur once every 1 or 2 days with different devices mostly tuya switch modules (Moes and similar) and osram/lightify RGB bulbs, very occasionally with one Ikea socket.
I've route the network triyng to keep as much good router as possible connected with che coordinator and to link every end device with the nearest and most suitable router available (for example aqara sensors with aqara smart plug and Ikea button with Ikea bulbs)

galambert75, 2022-09-08:
After some further testing, I wanted to report some findings to the round. Yesterday I stopped the Z2M service and added all 7 of my QS-Zigbee-CP03 devices via ZHA. They have now been running for > 24 hours with no interruptions, which is longer than I had observed with Z2M. 

Since all other components are unchanged (coordinator device & firmware, router device & firmware, ZigBee devices, location, etc.) it would seem the *'No network route' (205))'* issue is specific to either Z2M or MQTT.

OTOH, I am experiencing an issue with ZHA which I will report there. On some of the devices (apparently the ones connected via router and not directly to the coordinator), the position of the shutters seems to be reset to "closed" after a few minutes (~10).

rosarioscavo, 2022-09-09:
Has someone tried to switch to conbee and see if things gets better?

wrichter, 2022-09-10:
I switched to QS-Zigbee-C01 devices and still see the issue. At the moment, I have 3 devices connected - one works , the other two show the 205 error when I try to control them. All three devices still publish status updates, i.e. if I raise/lower the blinders manually I can see their state in the Z2M web UI.

I noticed that contrary to @sk8er000's comment, the two devices which DO work are connected directly to the coordinator, albeit with a horrible link quality: 
[image]

rosarioscavo, 2022-09-10:
> 

So you're able to use AZi Rolladen rechts but not Azi Rolladen links, right? It's definitely weird.

wrichter, 2022-09-11:
I've since rebuilt my network (forgot to change the network key when I originally set it up). The problem still exists in the "Bad Rolladen" actor. BUT: I've tried the workaround described in https://github.com/Koenkk/zigbee2mqtt/issues/9249#issuecomment-1004037749 and it seems to apply here as well. I've added the actor to a group (while it was responsive) and I can now control (open/close/stop/go to percentage, etc...) the group INCLUDING the "Bad Rolladen" actor even when accessing it directly yields 205 errors.

galambert75, 2022-09-12:
4 days since the change to ZHA and no connectivity issues at all. There was even a blackout and once power was restored all seven devices were immediately visible.

> After some further testing, I wanted to report some findings to the round. Yesterday I stopped the Z2M service and added all 7 of my QS-Zigbee-CP03 devices via ZHA. They have now been running for > [...]

Ucsus, 2022-09-14:
Same problem. ZHA integration, BSEED switch - turned off from time to time and had to be reconfigured. Changed the switch and problem persists. In other places, these switches work properly. I moved to Z2M - a couple of days of happiness and again a problem with that switch. In another room, there is a router socket above it, perhaps sometimes the switch is connected to the socket and after that it can no longer connect to the coordinator.

AlexMKX, 2022-09-16:
Same problem here with Lonsonho devices (_TZ3210_wdexaypg)  on CC2652P

The wifi (2.4) is moved to the upper channels, so no frequency overlap.
The dimmers losing connectivity very often (several times a day). Tried different availability settings - with no effect.  After repair - they work for some time, but the problem returns in couple hours.

Today I have noticed the following :  if you turn on/off 3-4 times, the dimmer wont turn on after couple retries, however, the zigbee2mqqt says it turned on. I managed it to solve by send turn on command different different brightnes values  in a loop.

However, the connectivity loss is ruining this trick :)

lucianojss, 2022-09-17:
Got same problem with some TS130F and Coordinator sonoff zigbee 3.0 dongle plus (with latest firmware)

I noticed, that sending command from `zigbee2mqtt` it doesn't work I get the "No network route" error, but if I click physically in the button it immediately updates the value in the `zigbee2mqtt` dashboard.. 

So it means that in fact there is connection between the `TS130F` => to Coordinator but from Coordinator => `TS130F` not.

wrichter, 2022-09-17:
I've been running the workaround I described above (single device groups) for a week now and it has been stable. I therefore strongly believe this is the same issue as #9249

KidA001, 2022-10-02:
Having the same exact issues here. Using a Ssonoff zigbee 3.0 (latest firmware) - My devices are a bit far from the coordinator with multiple routers in-between. I have ensured no other 2.4Ghz WiFi is running on channel 11. It's pretty frustrating, happy to provide logs or more data if that's useful.

KidA001, 2022-10-03:
Some more context from my end:

- I am using a Sonoff 3.0 with latest firmware
- The Sonoff is plugged into a USB Extension
- Z2M is running on 2.4Ghz channel 11, my other wifi devices are on ch 1 and 6
- Devices randomly go offline
- These devices are either my Hue Bulbs or Dresden Elektronik Mega23M12 (which is all I really have on my network)
- Most of my devices are linked through routers as the coordinator is a bit away (map below)
- "offline" devices respond if I have used a bind to a remote
- Unplugging and plugging the device brings it back "online"

Any help appreciated - I'm going a bit crazy having to physically unplug my devices at night to go to bed.

[image]

PlanetDyna, 2022-10-22:
My CP2102/CP2109 also stopped working. Error: "No network route". I see LQI status on some devices other showing nothing anymore. Seems to be a general problem.

Koenkk, 2022-10-22:
Without a sniff I cannot do anything: https://github.com/Koenkk/zigbee2mqtt/issues/12774#issuecomment-1177633964

awhitwam, 2022-10-23:
I have the exact same problem.  The devices always show in the Map with multiple routes but goes offline intermittantly.   I would love to provide a sniff, but my units are at a remote site overseas.  Not sure i can do it remotely but will try.

awhitwam, 2022-10-28:
I've just noticed that all my CP03 devices (even the ones that work) only ever show one number on the network map for connections other than to the coordinator.  For example in the screenshot above;

Blind_utility has a 88/87 connection to the coordinator, but just 67 connection to the Plug_kitchen

Blind_utility has a very poor connection to the coordinator, reading 2 (which i would expect) but again  showing just a 112 to the Plug_bedroom.

This pattern is repeated for all CP03 devices.   
Also I noticed there is a never a connection shown from CP03 device to CP03 device (blind to blind), which is odd given I have some a few feet away from each other.

It strikes me these units only work when there is a strong connection to the coordinator.

Koenkk, i am more than happy to purchase a couple of these for you if it would help?

Koenkk, 2022-10-29:
> Koenkk, i am more than happy to purchase a couple of these for you if it would help?

It would be better if you buy a sniffer yourself, it is not guaranteed that the issue also occurs in my network.

Logima, 2022-11-07:
I might be affected by this issue. Currently I have one lamp which doesn't respond to commands (Paulmann 371000002, 0xe29c). Reconfiguring seems to succeed, but doesn't affect anything. Network map shows it has connection to three other routers (0x8256, 0x173e and 0xaa46) with decent LQI, but no direct connection to coordinator. This is a bit odd, as this lamp is the nearest device to coordinator.

Earlier I had similar case with another lamp, which solved by itself when the lamp got direct connection to coordinator again.

Here is a quick sniff of sending a command to 0xe29c until Z2M shows:
```
Publish 'set' 'state' to 'tyohuoneen_valo' failed: 'Error: Command 0x00158d000581bad3/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
```
[no-network-route.zip](https://github.com/Koenkk/zigbee2mqtt/files/9947180/no-network-route.zip)
Network key: `01030307010406020008090100030501`
Coordinator is Sonoff ZBDongle-P, running CC1352P2_CC2652P_launchpad_coordinator_20221102.
Z2M version 1.28.1.

slashbv, 2022-11-07:
I have the same issue with QS-Zigbee-CP03. I didn't manage to find a solution so far.
[image]

sk8er000, 2022-11-07:
After a quite long period I got this error again and this time on a TRÅDFRI outlet, this is the first time I notice this error on a IKEA device

tvdbroeck, 2022-11-11:
I had the same problem after connecting my 1 gang dimmer from Lonsonho/Girier (TS110E) to my zigbee network. Z2M received updates FROM the dimmer, but I could not send commands TO the dimmer.
After creating a dummy group with 1 device, my dimmer, I can send commands to the group like I would to the dimmer. This seems to work fine now.

cunode, 2022-11-14:
The workaround with the dummy group seems to work. I set them up 2 months ago and since then my QS-Zigbee-CP03 curtain module works without problems.
This proves that my setup does not have a hardware nor a device firmware problem nor a weak network. Instead, the Zigbee2MQTT (version 1.25.1-dev) seems sending commands to individual devices differently than grouped devices.

toxic0berliner, 2022-11-17:
I have the same issue using a TS130F QS-Zigbee-C03 using Z2mqtt v1.28.2 and a sonoff USB stick.
The group workaround also works for me. So I'm also suspecting it's something with Z2mqtt itself more than the USB dongle or the device itself.
Whenever it's unresponsive with the no route errors it is still responsive to the cabled switch and Hass is still recieving the current status of motor moving and actual curtains position. Sending commands to the group does not "revive" the device itself for me, the device responds properly to the command sent to the group but the only way for me to get the device control back without the group workaround is to switch off then on the breaker for the curtains, actually rebooting the modules. But after a few hours the same behavior comes up.

drknexus666, 2022-11-21:
Possibly same issue, using zigbee2mqtt-git 1.28.2, CC2652P (Zigstar Stick v4) running CC1352P2_CC2652P_launchpad_coordinator_20220219. Have lots of repeaters on network. 
Devices seemingly randomly start going offline or stop accepting commands with 'No network route' (205) errors. Sometimes repeatedly sending a command will bring the device back to functioning. Usually group commands will still work and they will sometimes bring the device back to normal function but not always. Resetting power to all devices will cause the network to work without problem for ~1day.
Wireshark capture with everything working at first then things start failing. Some examples, 0xB595, 0xA4AD, 0x0421C. 
[WorkingThenEventuallyNoRoute.zip](https://github.com/Koenkk/zigbee2mqtt/files/10051074/WorkingThenEventuallyNoRoute.zip)

Koenkk, 2022-11-21:
@drknexus666 can you provide me your network key so that I can decrypt the sniff?

toxic0berliner, 2022-11-22:
Seems drknexus666 provided it. I'm not savvy enough to produce such a sniff yet but with a bit of help I can most probably learn by myself. 
Reading on z2m I believe that the group workaround works because the message gets broadcasted and any group member is responsible to act on it while the message to the device itself needs to be "routed". 
I haven't tested with another software than z2m, did someone confirm it is not happening with ZHA for instance or any other bridge?

[96 more replies]

closed 2023-07-12 by github-actions[bot]