Koenkk/zigbee2mqtt #12779: Manual ota update error smart plug TuYa

AiramsRT, 2022-06-10:
### What happened?

When trying to manually update a TuYa smart plug, it tells me that there is no image available even though the update skips

The error it indicates is:

Failed to check if update available for 'Enchufe_Termo' (No image available for imageType '5634')

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

2022.05.3

### Adapter

TS011F_plug_3

Koenkk, 2022-06-11:
It means we don't have a OTA available for this specific version of the TS011F, to add one follow https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

AiramsRT, 2022-06-11:
> It means we don't have a OTA available for this specific version of the TS011F, to add one follow https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

Oops... I don't use the TuYa bridge, I use Conbee II as a bridge since I use z2mqtt in HA supervisor

closed 2022-07-20 by github-actions[bot]