Koenkk/zigbee2mqtt #12788: Neo NAS-AB02B0 Interview Failed

TankedThomas, 2022-06-11:
### What happened?

I am trying to pair a brand new Neo NAS-AB02B0 to Zigbee2MQTT. I was going to avoid posting an issue because this device has been covered a lot, but it seems my scenario is at least a little unique.

I am running Zigbee2MQTT and the Mosquitto broker both under Docker on a Raspberry Pi CM4, with an Electrolama zzh! (CC2652R) coordinator on the latest 3.x.0 coordinator firmware. 
Docker images used were koenkk/zigbee2mqtt:latest and :latest-dev (same behaviour on both).

The inside of the siren says it's a model NAS-AB02B0 V3 20190910, exactly the same as shown in [#7051#issuecomment-818181554](#7051#issuecomment-818181554) et al.

This is what the frontend shows after pairing:
[image]

Pressing the Reconfigure button simply displays an error saying that the device cannot be configured.

### What did you expect to happen?

Judging by #7051 and #8331, I figured this device should be ready to go out of the box now, but instead of returning even partial information like other people (such as manufacturer name), I'm getting almost nothing of value, the interview fails, and the device therefore doesn't work.

I tried manually adding the converters as suggested in [https://github.com/Koenkk/zigbee2mqtt/issues/10348#issuecomment-1001136578](#10348#issuecomment-1001136578) but that didn't work (and yes, I added the tuya const as well).

I've looked through all the issues relating to this device and can't find anything that quite matches my exact problem, nor a working solution. I have not tried pairing with a Tuya gateway, however, as I do not own one (nor do I particularly want to buy one).

### How to reproduce it (minimal and precise)

After enabling pairing via the Zigbee2MQTT frontend, connecting the siren by long-pressing the button allows it to pair, and then it proceeds to produce this behaviour every time.

This occurs whether I use battery or USB power.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220219

### Adapter

Electrolama zig-a-zig-ah! (zzh!) (CC2652R)

### Debug log

Please note that I've removed information from my log on unrelated devices for privacy (it also makes it a lot shorter).
None of the other devices are currently active so they shouldn't be interfering. They do, however, all work flawlessly.

[z2mlog.txt](https://github.com/Koenkk/zigbee2mqtt/files/8884010/z2mlog.txt)

I've also followed Docker's log in real-time whilst trying different methods to get this working, but the results are always identical.

Any help would be greatly appreciated.

Koenkk, 2022-06-12:
Can you provide the herdsman debug logging when trying to pair the device? First force remove it, restart z2m and then try to pair.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

TankedThomas, 2022-06-12:
That's the exact log I provided (z2mlog.txt, at the bottom) using the Docker commands on that page. The only information I removed from the log was information related to other sensors, and my personal IDs such as the network key (though in hindsight, it would have been fine to leave the panID intact).
It doesn't really add much over the regular logs, however.

I uploaded the file like the template asked but here the log is in a code block:

`Using '/app/data' as data directory
Zigbee2MQTT:debug 2022-06-12 01:08:32: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2022-06-12 01:08:32: Logging to console and directory: '/app/data/log/2022-06-12.01-08-32' filename: log.txt
Zigbee2MQTT:debug 2022-06-12 01:08:32: Removing old log directory '/app/data/log/2022-06-11.12-26-24'
Zigbee2MQTT:info  2022-06-12 01:08:32: Starting Zigbee2MQTT version 1.25.2 (commit #3c5854fa)
Zigbee2MQTT:info  2022-06-12 01:08:32: Starting zigbee-herdsman (0.14.34)
Zigbee2MQTT:debug 2022-06-12 01:08:32: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":true},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[26],"extendedPanID":[REDACTED],"networkKey":"HIDDEN","panID":REDACTED},"serialPort":{"adapter":"auto","path":"/dev/ttyACM0"}}'
2022-06-11T13:08:32.876Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[REDACTED],"panID":REDACTED,"extendedPanID":[REDACTED],"channelList":[26]},"serialPort":{"path":"/dev/ttyACM0","adapter":"auto"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":true,"concurrent":null,"delay":null}}'
2022-06-11T13:08:32.880Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-06-11T13:08:32.892Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2022-06-11T13:08:32.896Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-06-11T13:08:32.899Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-06-11T13:08:32.910Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2022-06-11T13:08:32.911Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2022-06-11T13:08:32.912Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2022-06-11T13:08:32.914Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2022-06-11T13:08:32.916Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-06-11T13:08:32.916Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-06-11T13:08:32.917Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-06-11T13:08:32.922Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2022-06-11T13:08:32.922Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2022-06-11T13:08:32.922Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2022-06-11T13:08:32.922Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2022-06-11T13:08:32.922Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-06-11T13:08:32.922Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2022-06-11T13:08:32.923Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2022-06-11T13:08:32.928Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,10,97,2,2,1,2,7,1,59,137,52,1,0,233]
2022-06-11T13:08:32.929Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,10,97,2,2,1,2,7,1,59,137,52,1,0,233]
2022-06-11T13:08:32.929Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 10 - 3 - 1 - 2 - [2,1,2,7,1,59,137,52,1,0] - 233
2022-06-11T13:08:32.929Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20220219}
2022-06-11T13:08:32.930Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-06-11T13:08:32.930Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 16
2022-06-11T13:08:32.930Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20220219})
2022-06-11T13:08:32.931Z zigbee-herdsman:adapter:zStack:startup beginning znp startup
2022-06-11T13:08:32.933Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":130}
2022-06-11T13:08:32.934Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,130,0,178]
2022-06-11T13:08:32.939Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,24,0,104]
2022-06-11T13:08:32.940Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,24,0,104]
2022-06-11T13:08:32.940Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [24,0] - 104
2022-06-11T13:08:32.940Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":24}
2022-06-11T13:08:32.941Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-06-11T13:08:32.941Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvReadExt - {"id":130,"offset":0}
[... cut ...] rk":{"channel":26,"extendedPanID":"REDACTED","panID":REDACTED},"permit_join":false,"version":"1.25.2"}'
[REDACTED]
Zigbee2MQTT:info  2022-06-12 01:08:33: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-06-12 01:08:33: MQTT publish: topic 'zigbee2mqtt/0x804b50fffe04740e', payload '{"device":{"friendlyName":"0x804b50fffe04740e","ieeeAddr":"0x804b50fffe04740e","model":"unknown","networkAddress":6481,"type":"Unknown"},"last_seen":"2022-06-12T00:36:50+12:00"}'`

Koenkk, 2022-06-12:
Interviewing can fail for many reason, maybe you tried to pair the device in a different place now? Without the logs of the failed attempt it is impossible to say what went wrong.

> One line that stood out to me was Device '0x804b50fffe04740e' is already in database with different networkAddress, updating networkAddress. I'm not sure if this has anything to do with the problem I had though.

This is expected when you repair it (since it will get a new network address)

Feel free to close this.

TankedThomas, 2022-07-17:
Sorry, forgot all about this because I've been busy!

I paired it in the same place. Everything was the same except the time and date, best I can really say.

Oh well, since I don't have a proper log of the failed attempt, I guess it's a moot point. Sorry to be a pain, but I appreciate your help. If I come across this problem again in the future, I'll make sure to get a proper log so the issue (if there is one and it's not some anomaly or user error) can be resolved.

I'll close this since there's nothing more to be done for now.

closed 2022-07-17 by TankedThomas