Koenkk/zigbee2mqtt #12793: MS-105Z (and other similar models, TS0601 / _TZE200_la2c2uo9) cannot set a minimum brightness limit

joaquinvacas, 2022-06-11:
### What happened?

Using latest firmware on Sonoff/Itead Zigbee 3.0 USB coordinator.

Using latest dev branch.

MS-105Z dimmer tested with original MOES/Tuya bridge can set minimum value, even the WiFi model of this dimmer.

I mean, for example, locking the 1% exposed value to being a real 30% (so lights will dim but never be turned off)

Those settings aren't available from the Zigbee2Mqtt interface.

### What did you expect to happen?

Being able to set a minimum value.

### How to reproduce it (minimal and precise)

Just join this device.

### Zigbee2MQTT version

 1.25.2-dev commit: 1390bc1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus by ITead CC2652P

### Debug log

There's no need for debug in this case.

joaquinvacas, 2022-06-16:
@Koenkk 

Sorry, didn't mind a screenshot would help.

Here it is from the offcial Tuya (MOES) LAN Zigbee gateway.

[image]

Also, I've noticed that the Tuya model SMD-01Z is supported by Zigbee2Mqtt and exposes a "min_brightness" setting. Maybe it's the same dimmer after all (Tuya/generic ones).

Any thoughts? Anything I could change for testing the SMD-01Z profile with my device to check if it works with "min_brightness"?

joaquinvacas, 2022-06-16:
Missed this part too, not sure if it changes a thing

[image]

Koenkk, 2022-06-16:
What are the data points of the min brightness?

https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

joaquinvacas, 2022-06-16:
Here they go:

Datapoint for min brightness = 3

Also, taken in account another those:

Datapoint for type of light source = 4
(LED, halogen, incandescent)

Also "countdown" = 6
(in seconds, for example, 3600s)

joaquinvacas, 2022-06-16:
Oh! And brightness levels are shown as 1000 = 100%
And in my case (36% equals to 360 in "Event Details")

joaquinvacas, 2022-06-16:
[image]

Maybe this can help too @Koenkk

joaquinvacas, 2022-06-17:
External converter isn't working.

When I try to load it, startup keeps spamming those logs

[image]

Koenkk, 2022-06-18:
Made a mistake, updated https://github.com/Koenkk/zigbee2mqtt/issues/12793#issuecomment-1158947022 can you try again?

joaquinvacas, 2022-06-18:
Keep getting this errors

[image]

Koenkk, 2022-06-18:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/12793#issuecomment-1158947022 again

joaquinvacas, 2022-06-18:
[image]

Now it looks different 😅

Koenkk, 2022-06-18:
Are you sure the contents of `ext_converter.js` are exactly like https://github.com/Koenkk/zigbee2mqtt/issues/12793#issuecomment-1158947022 . The converter loads fine for me.

joaquinvacas, 2022-06-18:
Okay. Seems I've had issues with the "copy" button on GitHub, copied manually and now it's working.

When I try to config the dimmer, it shows me this error:

_Zigbee2MQTT:error 2022-06-18 23:43:29: Publish 'set' 'min_brightness' to 'Guirnalda exterior' failed: 'ReferenceError: utils is not defined'_

[image]

joaquinvacas, 2022-06-18:
Okay, added 

const utils = require('zigbee-herdsman-converters/lib/utils');

for the imports and now it's working. 

The only "issue" it's that it doesn't show the current state of min_brightness, it does adjust it, but as soon as you come back to the device's settings, those are showing 0 (even if it's correct, set at 92 right now)

[image]

Also, can I use this converter for enabling countdown? (datapoint 6)

Not too much into programming, but I may try.

Koenkk, 2022-06-20:
This adds the countdown, but what exactly does it do?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    tuya_min_max_brightness: {
        key: ['min_brightness', 'max_brightness', 'countdown'],
        convertSet: async (entity, key, value, meta) => {
            let newValue, dp;
            if (key === 'min_brightness') {
                if (value >= 1 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 1, 255, 0, 1000);
                    dp = tuya.dataPoints.dimmerMinLevel;
                } else {
                    //throw new Error('Dimmer min_brightness is out of range 1..255');
                }
            } else if (key === 'max_brightness') {
                if (value >= 1 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 1, 255, 0, 1000);
                    dp = tuya.dataPoints.dimmerMaxLevel;
                } else {
                    //throw new Error('Dimmer min_brightness is out of range 1..255');
                }
            } else if (key === 'countdown') {
                newValue = value;
                dp = 6;
            }
            // Always use same transid as tuya_dimmer_state (https://github.com/Koenkk/zigbee2mqtt/issues/6366)
            await tuya.sendDataPointValue(entity, dp, newValue, 'dataRequest', 1);
        },
    },
}

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_la2c2uo9'}],
    model: 'MS-105Z',
    vendor: 'Moes',
    description: '1 gang 2 way Zigbee dimmer switch',
    fromZigbee: [fz.moes_105_dimmer, fz.ignore_basic_report],
    toZigbee: [tz.moes_105_dimmer, tzLocal.tuya_min_max_brightness],
    meta: {turnsOffAtBrightness1: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
    exposes: [e.light_brightness().setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).withMinBrightness().withMaxBrightness().setAccess(
        'min_brightness', ea.STATE_SET).setAccess('max_brightness', ea.STATE_SET), exposes.numeric('countdown', ea.STATE_SET).withValueMin(0)],
};

module.exports = definition;

```

joaquinvacas, 2022-06-20:
Countdown establishes a timer (hardware wise, if you establish a countdown you can even unplug your coordinator) to turn off the device.

In a dimmer isn't as useful as on a switch, for example: Letting the switch turn off after a period like 10min. (Actually using a small water pump for water recirculation at home. It makes me rely on an automation for turning it off at 10min of usage)

joaquinvacas, 2022-07-21:
Hey!

I'm not nearby home to test the dimmer countdown but can assure min_brightness is working.

How can I make permanent min_brightness setting without ext_converter?

vascozorrinho, 2022-08-03:
Hey!

did this work? i would like to have this as well

joaquinvacas, 2022-08-03:
> Hey!
> 
> did this work? i would like to have this as well

Min_brightness does work, haven't tested the countdown.

vascozorrinho, 2022-08-03:
Thanks!

timer isn't very important to me anyway.

another question. my setup only starts to light when the value is grater 50. I think because of that, the lights take a second to  light up when i press the button. does this min brightness value fixes that delay?

joaquinvacas, 2022-08-03:
> Thanks!
> 
> timer isn't very important to me anyway.
> 
> another question. my setup only starts to light when the value is grater 50. I think because of that, the lights take a second to light up when i press the button. does this min brightness value fixes that delay?

For that issue, MOES/Tuya should update their firmware, this is something that can't be controlled (actually, you can control the transition but we cannot configure it on a hardware standalone level.

joaquinvacas, 2022-09-03:
@Koenkk

> I see, can you check if it works with the above converter and also check that the state is updated correctly after the countdown?

The min_brightness only issue right now is that even with the value established, it doesn't show the value, but you can change it and makes the change available to the dimmer.

I mean, you can change it and it works, but if you check exposed settings, it does show "0" and not the real value.

Koenkk, 2022-09-06:
I've updated https://github.com/Koenkk/zigbee2mqtt/issues/12793#issuecomment-1160626108 , can you check again?

soleyj, 2022-09-12:
Hello, 

One doubt, I have searching if there is any possibility to update the setting of a tuya device without make a change in file configuration and I didn't find anything. 

So we can just update the setting with a mqtt post, like : zigbee2mqtt/sw1/set {data_point:4, data:"led"}

Thanks!

joaquinvacas, 2022-09-16:
> I've updated [#12793 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/12793#issuecomment-1160626108) , can you check again?

I think you've forgotten declaring utils again :eyes:

joaquinvacas, 2022-09-16:
[image]

Also, showing 0 as before. :shrug:

Koenkk, 2022-09-19:
@joaquinvacas added the utils (https://github.com/Koenkk/zigbee2mqtt/issues/12793#issuecomment-1160626108). If it doesn't work, please provide me the debug log when changing the `min_brightness`.

joaquinvacas, 2022-10-02:
2022-10-02 18:03:00Received Zigbee message from 'Medidor consumo DIN', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":264}' from endpoint 1 with groupID 0
2022-10-02 18:03:04Received MQTT message on 'zigbee2mqtt/Patio/set' with data '{"min_brightness":67}'
2022-10-02 18:03:04Publishing 'set' 'min_brightness' to 'Patio'
2022-10-02 18:03:04Received Zigbee message from 'Patio', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,4],"type":"Buffer"},"datatype":2,"dp":3}],"seq":35594}' from endpoint 1 with groupID 0
2022-10-02 18:03:04from moes_105_dimmer, dp=[3], datatype=[2], value=[260]
2022-10-02 18:03:04zigbee-herdsman-converters:moes_105_dimmer:NOT RECOGNIZED DP #3 with data {"dp":3,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,4]}}
2022-10-02 18:03:05Received MQTT message on 'zigbee2mqtt/Patio/set' with data '{"min_brightness":67}'
2022-10-02 18:03:05Publishing 'set' 'min_brightness' to 'Patio'
2022-10-02 18:03:05Received Zigbee message from 'Patio', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,4],"type":"Buffer"},"datatype":2,"dp":3}],"seq":35850}' from endpoint 1 with groupID 0
2022-10-02 18:03:05from moes_105_dimmer, dp=[3], datatype=[2], value=[260]
2022-10-02 18:03:05zigbee-herdsman-converters:moes_105_dimmer:NOT RECOGNIZED DP #3 with data {"dp":3,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,4]}}
2022-10-02 18:03:05Received Zigbee message from 'Medidor consumo DIN', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":272}' from endpoint 1 with groupID 0
2022-10-02 18:03:07Received MQTT message on 'zigbee2mqtt/Patio/set' with data '{"min_brightness":69}'
2022-10-02 18:03:07Publishing 'set' 'min_brightness' to 'Patio'
2022-10-02 18:03:07Received Zigbee message from 'Patio', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,12],"type":"Buffer"},"datatype":2,"dp":3}],"seq":36106}' from endpoint 1 with groupID 0
2022-10-02 18:03:07from moes_105_dimmer, dp=[3], datatype=[2], value=[268]
2022-10-02 18:03:07zigbee-herdsman-converters:moes_105_dimmer:NOT RECOGNIZED DP #3 with data {"dp":3,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,12]}}

It does change the min_brightness. The issue here is that the min_brightness doesn't show the actual status (just at the moment of changing the value)

Koenkk, 2022-10-04:
Can you update to z2m 1.28.0 and try the following external converter: https://gist.github.com/Koenkk/affcce1eaac01960384067dc568d569e

This should give you all the supported features.

joaquinvacas, 2022-10-12:
Actually it's working, it reminds the last status of min_brightness :heart: 

Also, got the "power_on_behavior" too.

Koenkk, 2022-10-12:
great, integrated the changes.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-12 by Koenkk