Koenkk/zigbee2mqtt #12794: Stuck OTA after Z2M crash

h3x4d3c1m4l, 2022-06-11:
### What happened?

After Z2M crashed due of some kind of timeout error, any devices that were OTA updating seem to become stuck in a state of updating, but they are not actually being updated anymore (as Z2M crashed). Due to this I can't restart the OTA. This is how it looks in the web interface:

https://pasteboard.co/dFeZ5sARi9Tl.png

### What did you expect to happen?

After the restart of Z2M, I expected it to cancel the OTA so I can retry.

### How to reproduce it (minimal and precise)

It happens randomly, so hard to tell... :(

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220219

### Adapter

zzh!

### Debug log

```
Zigbee2MQTT:info  2022-06-11 23:32:19: MQTT publish: topic 'zigbee2mqtt/badkamer_schakelaar', payload '{"action":"","battery":60,"click":null,"last_seen":1654983139688,"linkquality":109,"update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2022-06-11 23:32:19: MQTT publish: topic 'zigbee2mqtt/badkamer_schakelaar', payload '{"action":null,"battery":60,"click":"","last_seen":1654983139688,"linkquality":109,"update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2022-06-11 23:32:19: MQTT publish: topic 'zigbee2mqtt/badkamer_schakelaar/action', payload 'on'
Zigbee2MQTT:info  2022-06-11 23:32:19: MQTT publish: topic 'zigbee2mqtt/badkamer_schakelaar/click', payload 'on'
Zigbee2MQTT:info  2022-06-11 23:32:19: MQTT publish: topic 'zigbee2mqtt/badkamer_verlichting', payload '{"brightness":104,"state":"ON"}'
Zigbee2MQTT:info  2022-06-11 23:32:19: MQTT publish: topic 'zigbee2mqtt/badkamer_lamp', payload '{"brightness":104,"last_seen":1654983139731,"linkquality":120,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-06-11 23:32:19: MQTT publish: topic 'zigbee2mqtt/gang_schakelaar_deur', payload '{"action":null,"battery":21,"click":null,"last_seen":1654983139924,"linkquality":109,"update":{"progress":3.28,"remaining":6878,"state":"updating"},"update_available":false}'
Error: Read 0x00158d000327a7d2/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 18062 - 1 - 50 - 6 - 1 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
```

Koenkk, 2022-06-12:
Can you provide the debug log when starting z2m? There is already some logic in place which fixes the OTA state when z2m crashes/restarts during an OTA: https://github.com/Koenkk/zigbee2mqtt/blob/a252914ecd1325cc1099f03af61872eb5cf0a405/lib/extension/otaUpdate.ts#L63

h3x4d3c1m4l, 2022-06-17:
At a certain moment, the OTA states were cleared and I could restart the OTA updates.

I haven't been able te reproduce it anymore: it didn't crash while having log level on debug and all of my device are up to date on OTA now...

I'd say we close this issue for now. If there's new OTA updates I once again check if I can reproduce it!

closed 2022-06-17 by h3x4d3c1m4l

Bascht74, 2022-12-30:
I have the same problem now (even with the latest dev version). 
z2m crashed during OTA and this is the result after restarting: 

[image]

This was the error causing a restart: 
```
error 2022-12-30 01:10:40: Failed to check if update available for '0x54ef44100053e5ee' (Device didn't respond to OTA request)
error 2022-12-30 01:18:13: Adapter disconnected, stopping
error 2022-12-30 01:18:13: Failed to stop Zigbee2MQTT

```

This is the log after a restart: 
```
debug 2022-12-30 01:25:09: Loaded state from file /config/zigbee2mqtt/state.json
info  2022-12-30 01:25:09: Logging to console and directory: '/config/zigbee2mqtt/log/2022-12-30.01-25-09' filename: log.txt
debug 2022-12-30 01:25:09: Removing old log directory '/config/zigbee2mqtt/log/2022-12-29.11-19-11'
info  2022-12-30 01:25:09: Starting Zigbee2MQTT version 1.28.4-dev (commit #d8f1f3e)
info  2022-12-30 01:25:09: Starting zigbee-herdsman (0.14.83)
debug 2022-12-30 01:25:09: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[20],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":14772},"serialPort":{"path":"tcp://192.168.175.195:6638"}}'
info  2022-12-30 01:25:11: zigbee-herdsman started (resumed)
info  2022-12-30 01:25:11: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20221220,"transportrev":2},"type":"zStack3x0"}'
debug 2022-12-30 01:25:11: Zigbee network parameters: {"channel":20,"extendedPanID":"0x00124b00257920d9","panID":14772}
info  2022-12-30 01:25:11: Set transmit power to '10'
info  2022-12-30 01:25:11: Currently 119 devices are joined:
info  2022-12-30 01:25:11: Lampe_EG_Flur_Komode (0x0017880103483fad): 8718696449691 - Philips Hue White A60 Single bulb E27/B22 (Router)
info  2022-12-30 01:25:11: Schalter_EG_Flur_Komode (0x0017880103a54379): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2022-12-30 01:25:11: Lampe_OG_Flur_Komode (0x0017880102fcfb11): 8718696449691 - Philips Hue White A60 Single bulb E27/B22 (Router)
info  2022-12-30 01:25:11: Schalter_OG_Flur_Komode (0x0017880108749646): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2022-12-30 01:25:11: Schalter_OG_Kara_Eingang (0x001788010874ffa4): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2022-12-30 01:25:11: Lampe_OG_Kara_Schreibtisch (0x00178801036113ff): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
info  2022-12-30 01:25:11: Schalter_OG_Kara_Schreibtisch (0x001788010808ab6d): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2022-12-30 01:25:11: Schalter_OG_Noah_Eingang (0x00178801087511aa): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2022-12-30 01:25:11: Schalter_OG_Noah_Bett (0x001788010b72f6ee): 929002398602 - Philips Hue dimmer switch (EndDevice)
info  2022-12-30 01:25:11: Schalter_OG_Schlafzimmer_Eingang (0x0017880108f370b2): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2022-12-30 01:25:11: Schalter_OG_Leona_Eingang (0x0017880108742814): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2022-12-30 01:25:11: Schalter_OG_Leona_Bett (0x001788010b730dd2): 929002398602 - Philips Hue dimmer switch (EndDevice)
info  2022-12-30 01:25:11: Lampe_OG_Leona_Stehlampe (0x0017880104a6e705): 8718696449691 - Philips Hue White A60 Single bulb E27/B22 (Router)
info  2022-12-30 01:25:11: Lampe_OG_Leona_Bett (0x00178801011cad4d): 915005106701 - Philips Hue white and color ambiance LightStrip plus (Router)
info  2022-12-30 01:25:11: Lampe_OG_Schlafzimmer_Schreibtisch (0x00178801090c5ade): 9290022944 - Philips Hue white ambiance E14 (with Bluetooth) (Router)
info  2022-12-30 01:25:11: Router_OG_Abstellraum (0x54ef4410002c7ad2): SP-EUC01 - Xiaomi Aqara EU smart plug (Router)
info  2022-12-30 01:25:11: Router_OG_Schlafzimmer (0x54ef441000313f5e): SP-EUC01 - Xiaomi Aqara EU smart plug (Router)
info  2022-12-30 01:25:11: Schalter_EG_Wohnzimmer_Markise (0x04cf8cdf3c7c3c9f): WXCJKG11LM - Xiaomi Aqara Opple switch 1 band (EndDevice)
info  2022-12-30 01:25:11: Lampe_EG_Wohnzimmer_Sideboard (0x001788010124504c): 7299760PH - Philips Hue Bloom (Router)
info  2022-12-30 01:25:11: Router_EG_Kueche (0x54ef441000139f6b): SP-EUC01 - Xiaomi Aqara EU smart plug (Router)
info  2022-12-30 01:25:11: Vibrationssensor_Aussen_Terrasse_Markise_Rechts (0x00158d00080513a7): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
info  2022-12-30 01:25:11: Vibrationssensor_Aussen_Terrasse_Markise_Links (0x00158d00080513a8): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
info  2022-12-30 01:25:11: Thermometer_EG_Wohnzimmer (0x00158d00044609e1): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-12-30 01:25:11: Thermometer_OG_Bad (0x00158d00051e3961): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-12-30 01:25:11: Thermometer_OG_Schlafzimmer (0x00158d0005bd8dea): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-12-30 01:25:11: Thermometer_OG_Noah (0x00158d0006bead11): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
[... cut ...]
info  2022-12-30 01:25:12: MQTT publish: topic 'homeassistant/sensor/0x00158d0006bfa14b/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0006bfa14b"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"Thermometer_OG_Kara","sw_version":"3000-0001"},"device_class":"temperature","enabled_by_default":true,"name":"Thermometer_OG_Kara_temperature","state_class":"measurement","state_topic":"zigbee2mqtt/Thermometer_OG_Kara","unique_id":"0x00158d0006bfa14b_

Bascht74, 2022-12-30:
complete log:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10322665/log.txt)

state.json
[state.txt](https://github.com/Koenkk/zigbee2mqtt/files/10322674/state.txt)

Bascht74, 2022-12-30:
After changing updating to idle in the state.json and a restart, I can restart OTA again and it will continue where it left (e.g. 80% and 55%). So there might be something missing to automatically restart the running OTAs...

mattbruman, 2023-01-09:
> After changing updating to idle in the state.json and a restart, I can restart OTA again and it will continue where it left (e.g. 80% and 55%). So there might be something missing to automatically restart the running OTAs...

Thanks, I had some stuck Hue updates too and this let me try again.

Write, 2023-03-05:
> After changing updating to idle in the state.json and a restart, I can restart OTA again and it will continue where it left (e.g. 80% and 55%). So there might be something missing to automatically restart the running OTAs...

Thanks a lot !
This is indeed, the way

LeoS3, 2023-03-31:
> After changing updating to idle in the state.json and a restart, I can restart OTA again and it will continue where it left (e.g. 80% and 55%). So there might be something missing to automatically restart the running OTAs...

Hi, can you share a bit more detailed step by state on how to edit the state.json of zigbee2mqtt please?

I'm having the same issue, and I looked at the config/raw menu of zigbee2mqtt, and couldn't find any status=updating variables there. Though if I go to the detail of the stuck device, under 'state', I can see the status are 'updating'.

Thanks in advance.

SHxKM, 2023-05-25:
@Koenkk this still happens if Z2M is restarted during an OTA. Is it possible to have some kind of health check on the OTA updates? If you can point to where to get started I can maybe work on a PR.

Koenkk, 2023-05-26:
@SHxKM easiest way would be to clear all OTA states when z2m is started, this can be done here: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/otaUpdate.ts#L44

SHxKM, 2023-05-26:
> @SHxKM easiest way would be to clear all OTA states when z2m is started, this can be done here: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/otaUpdate.ts#L44

Thank you. Do you think the OTA states should be reset differently for devices that are stuck? In my manual fix for this I’d reset it to `idle` and change the updateAvailable flag back to `true`? The other option is to blindly reset all states.

Koenkk, 2023-05-26:
I would say to clear the state completely (as the state is unknown after a restart). So idle and update available to false.

dasb00ter, 2023-11-16:
Was the PR ever incorporated. I have several xiaomi blind motors that I dont want updated stuck in an updating animation. I would like to ignore the update but cant because its stuck in the process.

In state.json I have tried setting it something like this but it reverts on HA restart and doesnt fix the problem

```
    "0x54eblah1000blah": {
        "position": 0,
        "battery": 62,
        "device_temperature": 20,
        "power_outage_count": 0,
        "state": "OFF",
        "motor_state": "pause",
        "running": false,
        "linkquality": 54,
        "charging_status": false,
        "update": {
            "state": "idle",
            "installed_version": 3609,
            "latest_version": 3870
        },
        "update_available": false
```

tokaz, 2023-11-29:
> @SHxKM easiest way would be to clear all OTA states when z2m is started, this can be done here: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/otaUpdate.ts#L44

I'm still kinda new to this. 

Can you explain how to change this?

Bascht74, 2024-02-01:
I have this problem again after some time: 
[image]

Bascht74, 2024-02-01:
The old trick, changing state.json helped again...

@Koenkk could you re-open that issue, or should I create a new one?
I think it would be nice to reset these states to idle after a restart...
OTA is not picking up the progress anyway (at least for me).

shouser, 2024-02-29:
This still an issue.  Had a Phillips Hue LCT001 hang because power went out mid-OTA update.  I tried numerous cycles and combinations of restarting Z2M, restarting HAOS, and power cycling the bulb to try to get it to stop the update.  Nothing worked.  Only solution that did work was the state.json update to idle from updating.  I'd second the need to set all the OTAs to idle on fresh boot of Z2M.  Either that or give a documented way to reset OTA for a device via MQTT message.