Koenkk/zigbee2mqtt #12800: Nue / 3A NUE-ZBFLB appears not to be supported as per documents

JoshualFitzgerald, 2022-06-12:
### What happened?

Raspi zero 2w with  Zbeeduo adapter
Connected and appeared in config file, renamed friendly and that was it. Have numerous other devices running ok.  Was a log message about router.  I attempted to follow the support instructions but failed miserably, only got the first channel operating. 

### What did you expect to happen?

To connect seemlessly like my other devices.

### How to reproduce it (minimal and precise)

power up and reset in range

### Zigbee2MQTT version

not sure

### Adapter firmware version

not sure

### Adapter

zbeeduo

JoshualFitzgerald, 2022-06-13:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LXX60-FN27LX1.0'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'NUE-ZBFLB', // Vendor model number, look on the device for a model number
    vendor: 'Nue3A', // Vendor of the device (only used for documentation and startup logging)
    description: 'Light and 3 Speed Fan Switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('button_light'), e.switch().withEndpoint('button_fan_high'),
        e.switch().withEndpoint('button_fan_med'), e.switch().withEndpoint('button_fan_low')],
    endpoint: (device) => {
        return {'button_light': 1, 'button_fan_high': 2, 'button_fan_med': 3, 'button_fan_low': 4};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(4), coordinatorEndpoint, ['genOnOff']);
    },
};
module.exports = definition;

//this as external converter seems to work after blind luck and adapting another example.

shannonpasto, 2022-09-12:
I've just had one of these installed. Pairs fine however I can't seem to figure out the extend file. Using this still results in an unsupported device. Any tips?

JoshualFitzgerald, 2022-09-12:
I followed the instructions on 

https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions

I created the external converter file I named NUEZBFLB.js in the same folder as the configuration.yaml file which is /opt/zigbee2mqtt/data .  Then copy the above into the external converter file.

I didn't progress any further once it was operational due my limited understanding

shannonpasto, 2022-09-12:
Thanks. I tried to follow but I figure been able to figure them out...first time with zigbee2mqtt. Any chance you could share the file?

shannonpasto, 2022-09-12:
Actually, I just found NUE-ZBFLB referenced in `node_modules/zigbee-herdsman-converters/devices/smart_home_pty.js`. I added my model "LXX60-FN27LX1.0" to zigbeeModel and it works

JoshualFitzgerald, 2022-09-12:
Just create a new file as I mentioned in the previous post and copy and paste  the content from the 13 june post into that file