Koenkk/zigbee2mqtt #12801: HA plugin can't open config file at null/configuration.yaml

jesseas, 2022-06-12:
### What happened?

I was running this as a plugin to Home Assistant. After restoring my HA environment from backup the plugin will no longer start. In the plugin logs I see the following message over and over:

```
INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.25.2 start
> node index.js
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:310:20)
    at getInternalSettings (/app/lib/util/settings.ts:406:21)
    at Object.reRead (/app/lib/util/settings.ts:684:5)
    at start (/app/index.js:93:14)
```

I have confirmed that permissions on config files seem reasonable and that they are accessible inside the Zigbee2MQTT container. I have also tried restarting and uninstalling/reinstalling the plugin.

### What did you expect to happen?

I expected the plugin to start but instead it seems to not have actually started. To be clear the HA plugin page says it's running but the UI isn't up and I don't see sensors showing up in HA.

### How to reproduce it (minimal and precise)

I don't know, sorry.

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

I don't know, sorry

### Adapter

Sonoff Zigbee 3.0 dongle

### Debug log

I don't seem to be getting to this point.

jesseas, 2022-06-12:
Ok, so not sure what I did... I restarted things a few times and edited the YAML in the HA UI in a way that got stripped out during save (added quotes around data_path) saved and restarted and things seem like they are working... I am going to keep this open in case there is something I can provide to track this down - I am happy to help if I can. But feel free to close this if there is nothing that makes sense to do.

Stooovie, 2022-07-12:
@jesseas your fix helped me as well. The weird thing is the quotes get stripped after saving the YAML anyway, but it did fix the issue. No idea how this can randomly happen.

martin131, 2022-08-17:
It just happened to me too. I restarted the host server and after starting zigbee2mqtt didn't work. I had to restart the addon manually.

martin131, 2022-08-17:
It seems like a bug. Happened to me again after reboot of host system.

max255, 2022-08-18:
I confirm. Already 2 times the electricity was turned off - 2 times in a row zigbee2mqtt does not start itself.

Koenkk, 2022-08-18:
Can someone provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

max255, 2022-08-19:
> Can someone provide the herdsman debug log of this?

According to the instructions, I added the zigbee_herdsman_debug parameter to the configuration (I use the Z2M 1.27.0-1 add-on, HA supervised 2022.8.6). After a manual reboot, the herdsman logs appeared on the Log tab. After a hard power-down, the Log tab is just this:
```
> zigbee2mqtt@1.27.0 start
> node index.js
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
[12:36:45] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
[12:36:49] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
[12:36:54] INFO: Handing over control to Zigbee2MQTT Core ...
```
It feels like he just doesn't get there.
PS: Stick V4 from Egony https://github.com/egony/cc2652p_cc1352p_RF-STAR/wiki/Home-EN , firmware 20220302

savenko-egor, 2022-08-19:
I have same error after power loss.
```
[19:19:06] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
```

Koenkk, 2022-08-20:
What is shown for the `data_path`?

[image]

sweetpalma, 2022-08-20:
@Koenkk for me it's the same.
[image]

savenko-egor, 2022-08-20:
Yep. I have same config.[image]

Koenkk, 2022-08-21:
Can you provide the herdsman debug log when z2m is started? 

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Calimerorulez, 2022-08-21:
(Following this one, I've got the same problem, running HAOS)

max255, 2022-08-21:
> Can you provide the herdsman debug log when z2m is started?

I think that's not the problem. Normal startup just doesn't happen. If you look at the logs, you can see that instead of the path to the config folder, NULL is displayed for some reason. 
```
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
```
I looked at the sources and found the following in the run file:

```
export ZIGBEE2MQTT_DATA="$(bashio::config 'data_path')"
```

I powered off the server in order to reproduce the error and connected to the docker console. Next, I ran the test script to test and got this:

```
#!/usr/bin/with-contenv bashio

ZIGBEE2MQTT_DATA="$(bashio::config 'data_path')"
bashio::log.info "${ZIGBEE2MQTT_DATA}"
```
[image]

Next, I modified the script based on the old sources:
```
#!/usr/bin/with-contenv bashio

CONFIG_PATH=/data/options.json
DATA_PATH=$(jq --raw-output ".data_path" $CONFIG_PATH)
bashio::log.info "${DATA_PATH}"
```
[image]

Next, I reloaded the addon using home assistant. Changed the test script to the original version... and it worked. It seems to me that the problem is in this place. Perhaps bashio is incorrectly initialized or something else ... Further, I already lack the knowledge to parse it.

```
#!/usr/bin/with-contenv bashio

ZIGBEE2MQTT_DATA="$(bashio::config 'data_path')"
bashio::log.info "${ZIGBEE2MQTT_DATA}"
```
[image]

Calimerorulez, 2022-08-22:
When I reboot my Proxmox server, where I run HAOS in a VM, and mapped my CC2531 stick, I'm seeing this error too:

```
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
```

Then, when I restart the addon, it first can't connect to the adapter, but after two retries, it connects and then works.

Sometimes, the error on the config file happens by itself. Only remedy is to repeat the above...

Koenkk, 2022-08-22:
@Calimerorulez thanks for debugging, this looks like a HA problem to me. I've notified one of the HA developers.

mdegat01, 2022-08-22:
@max255 Can you run your test again but this time add the following to the very top of that test script:
```sh
bashio::log_level 'debug'
```
That will cause bashio to log out the details of everything. In particular every API call it makes with supervisor.

`bashio::config` doesn't actually look at any files in the container, it makes an API call to supervisor to ask it for config. So it seems like something is going wrong in that communication, setting log level to debug should hopefully make it more clear.

max255, 2022-08-22:
> Can you run your test again

Oh sure. Suggestion didn't work... fixed syntax according to bashio sources. When specifying the DEBUG logging level, nothing has changed. But with the ALL parameter, this happened:
```
#!/usr/bin/with-contenv bashio

bashio::log.level "all"

ZIGBEE2MQTT_DATA="$(bashio::config 'data_path')"
bashio::log.info "${ZIGBEE2MQTT_DATA}"
```
[image]

frenck, 2022-08-22:
> As a result, when zigbee2mqtt starts, it receives an empty configuration file and writes it to the cache. And then uses it.

Do you have debug logs of the original request made for the config? 
The logs above are already taking it from cache, it would be interesting to see the original request for the configuration (which is logged fully)

<https://github.com/hassio-addons/bashio/blob/5a1278193ed6cc3aafe3edf04619a40092c28086/lib/api.sh#L56-L60>

mdegat01, 2022-08-23:
@max255 do you have watchdog enabled for the addon? If so can you do one more simple test for me? On the addon turn watchdog off and leave "start at boot" enabled.

I think what may be happening is watchdog is starting the addon before it is supposed to and that's creating all sorts of issues. If that's the case then disabling watchdog before restarting the host should fix it as it should start at the time it is supposed to.

max255, 2022-08-23:
> watchdog enabled for the addon?

Yes, watchdog is enabled. After disabling the watchdog addon starts normally. I checked it twice.

> Do you have debug logs of the original request made for the config?

Unfortunately no. As I understand it, I need to change the container itself or the files inside it. How to do this I don't know yet. Perhaps I need to rebuild the container completely... or is there some other way?

mdegat01, 2022-08-23:
Thanks for confirming. https://github.com/home-assistant/supervisor/pull/3815 should fix it in that case. I just cut a new beta of supervisor ([2022.8.4](https://github.com/home-assistant/supervisor/releases/tag/2022.08.4)) with that fix. If you anyone on here wants to give it a shot and let me know if that resolves the issue for you I would appreciate it.

For those unaware if you switch to the beta channel in [Settings -> System -> Updates](https://my.home-assistant.io/redirect/updates/) it will install the beta version of supervisor.

max255, 2022-08-23:
> If you anyone on here wants to give it a shot and let me know if that resolves the issue for you I would appreciate it.

Did I understand you correctly? Need to switch to the beta channel and test the work with the watchdog enabled? Or do I need to wait for an update and answer if it helped or not?

mdegat01, 2022-08-23:
If you switch to the beta channel right now and update to the latest supervisor beta Z2M should work after a reboot even with watchdog enabled. I would appreciate the feedback if you are willing to but obviously you don't have to do that.

Calimerorulez, 2022-08-24:
@mdegat01 

I tested it with the beta you mentioned (I joined the beta channel), rebooted the server, and Z2M starts fine with no error.

Calimerorulez, 2022-08-24:
@mdegat01 

Well, I celebrated too quickly, just after another reboot, I'm getting the error again:

```
[20:56:30] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
```

mdegat01, 2022-08-25:
@Calimerorulez so just to clarify you are on supervisor version `2022.8.4` right? Also how are you rebooting the host? Are you doing it through the HA UI, entering `ha host reboot` in the CLI or some other way?

Calimerorulez, 2022-08-25:
@mdegat01, yes I'm running supervisor `2022.8.4`. My physical host is a Proxmox server. HAOS is running as a VM on the host. When I reboot, I choose (sorry, maybe it's different in English, but I'm using the Dutch language settings in HA) Settings -> Hardware -> Virtual Machine -> threedotmenu -> Reboot host, or something like that.

I just rebooted again, and now it does not error (it only takes a while for Z2M to find and connect the CC2531 adapter, but that's something else).

Doing the same over and over again gives sometimes the error, and sometimes not. It's frustrating for debugging, I know... :(

mdegat01, 2022-08-26:
@Calimerorulez did another beta bump (2022.08.5) just now. I was seeing some strangeness around yaml processing on our ci recently and two fixes cleaned it up so hoping that fixes this as well. Specifically https://github.com/home-assistant/supervisor/pull/3821 and https://github.com/home-assistant/supervisor/pull/3823 . If you wouldn't mind testing again I'd appreciate the feedback.

Calimerorulez, 2022-08-26:
@mdegat01 
Updating now, and I will report back.

Calimerorulez, 2022-08-26:
@mdegat01 
First two restart of the VM and the physical host, the Z2M addon starts ok. I will monitor/play around tonight and tomorrow and let you know.

Calimerorulez, 2022-08-29:
@mdegat01 

Seems to be running stable now :)

Koenkk, 2022-08-29:
Thanks all, assuming this can be closed now.

closed 2022-08-29 by Koenkk

mdegat01, 2022-08-29:
Thanks for all the help on here! That was a tough one to crack. Supervisor stable is now `2022.08.5` so everyone should see this issue resolved soon 👍

fliespl, 2022-09-02:
I have homeassistsnt with supervisor 8.5 and just experienced it 3 times today.

Home Assistant 2022.8.7
Supervisor 2022.08.5

marcomsousa, 2022-09-05:
I'm fully updated and have this problem.

rwjack, 2022-09-05:
yeah, happened to me as well

Home Assistant 2022.8.7
Supervisor 2022.08.5

HAOS Full VM Reboot, with Z2M watchdog enabled, does not start Z2M.
Restarting the addon manually starts it. Disabled watchdog for now, but haven't tested if the addon starts on boot when watchdog is disabled.

Calimerorulez, 2022-09-05:
@mdegat01 Seems it isn't over yet...

oNSXo, 2022-09-05:
Agreed, seeing the same issue still w/ versions:

Home Assistant 2022.8.7
Supervisor 2022.08.5
Operating System 8.5

oNSXo, 2022-09-05:
@wifispray That does look like a separate issue since it's getting to the config, but failing to start zigbee.

wifispray, 2022-09-05:
@Koenkk shall I raise another issue?

Koenkk, 2022-09-06:
@wifispray can you proivde the herdsman debug logging? It looks something goes wrong with the adapter communication.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

wifispray, 2022-09-06:
Zigbee2MQTT:debug 2022-09-06 13:50:57: Loaded state from file
/config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-09-06 13:50:57: Logging to console and
directory: '/config/zigbee2mqtt/log/2022-09-06.13-50-57' filename:
log.txt
Zigbee2MQTT:debug 2022-09-06 13:50:57: Removing old log directory
'/config/zigbee2mqtt/log/2022-09-06.13-49-31'
Zigbee2MQTT:info  2022-09-06 13:50:57: Starting Zigbee2MQTT version
1.27.2 (commit #unknown)
Zigbee2MQTT:info  2022-09-06 13:50:57: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:debug 2022-09-06 13:50:57: Using zigbee-herdsman with
settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":64290},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2022-09-06 13:50:57: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-09-06 13:50:57: Coordinator firmware version:
'{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:debug 2022-09-06 13:50:57: Zigbee network parameters:
{"channel":25,"extendedPanID":"0xdddddddddddddddd","panID":64290}
Zigbee2MQTT:error 2022-09-06 13:50:57: Failed to start zigbee
Zigbee2MQTT:error 2022-09-06 13:50:57: Check
https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html
for possible solutions
Zigbee2MQTT:error 2022-09-06 13:50:57: Exiting...
Zigbee2MQTT:error 2022-09-06 13:50:58: Error: not supported
    at DeconzAdapter.setTransmitPower
(/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1033:15)
    at Controller.setTransmitPower
(/app/node_modules/zigbee-herdsman/src/controller/controller.ts:395:29)
    at Zigbee.start (/app/lib/zigbee.ts:138:33)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)

Koenkk, 2022-09-07:
@wifispray does removing `transmit_power` from your z2m configuration fix the issue?

bigboban, 2022-09-11:
A have this problem few weeks. I am on last 2022.9.

DrMikeyS, 2022-09-14:
I am on 2022.9.2 and also getting the same. Mosquitto seems to be running fine but same error. I see 2022.9.3 has been released overnight, will try that and report back but feeling un-optimisitic.

Error:
`[07:59:46] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)`

socat:
`enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: "-d -d"
log: false`

/config/zigbee2mqtt/configuration.yaml:
`external_converters: []
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: addons
  password: #password
serial:
  port: /dev/ttyACM0
advanced:
  log_level: debug
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
availability: false
`

BAJ810, 2022-09-16:
I am having this issue and I am fully updated. I'm moving from ZHA to Z2M.

cautje, 2022-09-17:
Yes, here to.
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'

Home Assistant 2022.9.4
Supervisor 2022.08.6
Operating System 9.0
Frontend 20220907.2 - latest

Zigbee2MQTT
Current version: 1.27.2-1 (Changelog)

Please let me know if i can help.

THX!

ricargr, 2022-09-17:
Home Assistant 2022.9.4
Supervisor 2022.08.6
Operating System 9.0
Frontend 20220907.2 - latest

Zigbee2MQTT
Current version: 1.27.2-1

I reinstall and reconfig the zigbee2mqtt docker but no progress

manunited10, 2022-09-20:
Got the same error on z2m `1.27.2-1` and HA `2022.8.6` after a power loss.
Stopped the addon and started again (in Settings/Addons/Zigbee2Mqtt) and it started working without any issue.
```Proxmox > Debian 11 > HA Supervised + Z2M addon from official store.```

I should say that I have two other separate zigbee networks (two more coordinators) both running on the same machine. These two are in separate Z2M docker containers. They had no problem starting up after the power loss.

Calimerorulez, 2022-09-20:
Sadly all GitHub issues about this problem are closed as so it seems. I don't know if anyone of the devs is actually is debugging it.

djp3k05, 2022-09-20:
same issue here. but I get this out of the sudden, every day. not just after power loss. A z2m service restart fixes the issue, but just for few hours.

mariokil, 2022-09-20:
same issue for me. After z2m service restart everything is ok. In my case it looks that service "hanging" in the night between 1-3 AM (stop reporting to influxdb). At the beginning I supposed that maybe scheduled backup block some how but after change schedule for backup situation is the same :(

ricargr, 2022-09-20:
It looks like the problem is about hass, not zigbee2mqtt, someone should open a case.

mdegat01, 2022-09-20:
Ok let's break this down a bit. The current stable version of supervisor at this time is `2022.09.1`. If someone running on `2022.09.1` sees this specific error in their logs:
```
INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.25.2 start
> node index.js
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:310:20)
    at getInternalSettings (/app/lib/util/settings.ts:406:21)
    at Object.reRead (/app/lib/util/settings.ts:684:5)
    at start (/app/index.js:93:14)
```
with the `null/configuration.yaml` in there, please open an issue [here](https://github.com/home-assistant/supervisor/issues). Include your Z2M config and the supervisor logs around the same time. Also please summarize roughly what happened leading into it if anything specific (power loss? host reboot? nothing at all?)

> same issue for me. After z2m service restart everything is ok. In my case it looks that service "hanging" in the night between 1-3 AM (stop reporting to influxdb). At the beginning I supposed that maybe scheduled backup block some how but after change schedule for backup situation is the same :(

This doesn't sound like the same issue at all unless you also have the `null/configuration.yaml` in your log. And even then it sounds like Z2M was running fine and then just hung up and stopped running without warning. If so I would suggest opening a new issue here in this repo with your details

Calimerorulez, 2022-09-21:
> Ok let's break this down a bit. The current stable version of supervisor at this time is `2022.09.1`. If someone running on `2022.09.1` sees this specific error in their logs:
> 
> ```
> > node index.js
> Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
>     at Object.openSync (node:fs:585:3)
>     at Object.readFileSync (node:fs:453:35)
>     at Object.read (/app/lib/util/yaml.ts:7:29)
>     at read (/app/lib/util/settings.ts:310:20)
>     at getInternalSettings (/app/lib/util/settings.ts:406:21)
>     at Object.reRead (/app/lib/util/settings.ts:684:5)
>     at start (/app/index.js:93:14)
> ```
> 
> with the `null/configuration.yaml` in there, please open an issue [here](https://github.com/home-assistant/supervisor/issues). Include your Z2M config and the supervisor logs around the same time. Also please summarize roughly what happened leading into it if anything specific (power loss? host reboot? nothing at all?)
> 
> > same issue for me. After z2m service restart everything is ok. In my case it looks that service "hanging" in the night between 1-3 AM (stop reporting to influxdb). At the beginning I supposed that maybe scheduled backup block some how but after change schedule for backup situation is the same :(
> 
> This doesn't sound like the same issue at all unless you also have the `null/configuration.yaml` in your log. And even then it sounds like Z2M was running fine and then just hung up and stopped running without warning. If so I would suggest opening a new issue here in this repo with your details

Hi @mdegat01 

I'm running supervisor 2022.9.1, and I'm still seeing this issue. Sadly I'm not able to reproduce it constantly. Sometimes it happens after rebooting the host, sometimes it happens after restarting the addon. After the error, and restarting the addon, it works again...

oNSXo, 2022-09-21:
Submitted here: https://github.com/home-assistant/supervisor/issues/3884

Happens even VM reboot for me, fully updated.

mariokil, 2022-09-24:
My configuration:
Proxmox - 6.4.4
ZtM - 1.27.2-1
Home Assistant 2022.9.6
Supervisor 2022.09.1
Operating System 9.0
User interface - 20220907.2 - latest

Log (part)
[14:52:06] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)

Interestingly, the z2m hang off occurs fairly regularly at every night between 2:50-3:10 A.M. (I have no schedule tasks at this time of day). I'm not a Linux guy but it looks like there is some internal process set up that runs regularly and causes these problems.

Koenkk, 2022-09-25:
Let's continue in https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/387

Koenkk, 2022-09-25:
@ all; please check if this issue is fixed in the latest edge addon.

If not please comment here: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/387