Koenkk/zigbee2mqtt #12803: [New device support]: LDSENK09 Vendor ADEO similar to ZS130000178 Vendor Linkind

mickadam29, 2022-06-12:
### Link

https://www.leroymerlin.fr/produits/electricite-domotique/domotique-et-objets-connectes/domotique/solutions-de-commande/badge-porte-cles-alarme-lexman-82428869.html

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x5c0272fffee8ad03","nwkAddr":7605,"manufId":4727,"manufName":"ADEO","powerSource":"Battery","modelId":"LDSENK09","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1025,"inClusterList":[0,3,1280,2821],"outClusterList":[3,25,1281],"clusters":{"genBasic":{"attributes":{"modelId":"LDSENK09","manufacturerName":"ADEO","powerSource":3,"zclVersion":3,"appVersion":0,"stackVersion":6,"hwVersion":0,"dateCode":"20200929-18","swBuildId":"3.15"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00258dd909","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":6,"hwVersion":0,"dateCode":"20200929-18","swBuildId":"3.15","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1655036849013,"defaultSendRequestWhen":"immediate"}

### Comments

Hi

Could you add this model ADEO LDSENK09 as Linkind ZS130000178 ?
Both models look similar

I could test with the dev version

Thks

[image]
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LDSENK09'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LDSENK09', // Vendor model number, look on the device for a model number
    vendor: 'ADEO', // Vendor of the device (only used for documentation and startup logging)
    description: 'Alarm Remote Control', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.command_arm,fz.command_panic], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.action('disarm', 'arm_all_zones', 'arm_day_zones', 'panic')], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

Koenkk, 2022-06-12:
Does it work with this external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LDSENK09'],
    model: 'LDSENK09',
    vendor: 'ADEO',
    description: 'Security system key fob',
    fromZigbee: [fz.command_arm, fz.command_panic],
    toZigbee: [],
    exposes: [e.action(['panic', 'disarm', 'arm_partial_zones', 'arm_all_zones'])],
    onEvent: async (type, data, device) => {
        // Since arm command has a response zigbee-herdsman doesn't send a default response.
        // This causes the remote to repeat the arm command, so send a default response here.
        if (data.type === 'commandArm' && data.cluster === 'ssIasAce') {
            await data.endpoint.defaultResponse(0, 0, 1281, data.meta.zclTransactionSequenceNumber);
        }
    },
};

module.exports = definition;

```

mickadam29, 2022-06-13:
It works partially. The acknowledgment is missing. Normally the remote control flashes with a green LED. With this converter, a red LED lights up permanently. So outside the home we don't know if the order has been taken into account

mickadam29, 2022-06-13:
I haven't seen your code. I will test it and give you feedback

mickadam29, 2022-06-21:
@Koenkk 
That works today. Probably with the latest version of Jeedom 4.2.19. When i press a button, action is ok ans then remote blink twice (green led, no more red after about 10 seconds). Could you merge this converter ? Thk

mickadam29, 2022-06-22:
Thank’s