Koenkk/zigbee2mqtt #12809: Bug with lighting Set Brightness 0

Cossey, 2022-06-13:
### What happened?

Version 1.25.2 had what I would consider a breaking change for my environment. 

When I want to toggle off the lighting in my home (assortment of Phillips Hue bulbs) my scripts and environment set `{ "brightness": 0 }`. This would then get back the brightness and state in the state topic. Now in version 1.25.2, this only returns the state in the state topic, which causes problems for my scripts and openHAB.

### What did you expect to happen?

I expected that setting `{ "brightness": 0 }` to actually return 
```
{ "brightness": 0, "state": "OFF" }
```
on the state topic, not just 
```
{ "state": "OFF" }
```

### How to reproduce it (minimal and precise)

Set `{ "brightness": 0 }` on a command topic for a Phillips hue light. I would assume that this would also occur on other bulbs, but I don't have any others to test.

Also just pulled and tested latest-dev, and it is still a problem! For now I am sitting on 1.25.1 until the problem can be resolved.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220219

### Adapter

SONOFF Z3DP

Koenkk, 2022-06-14:
I think this is because of https://github.com/Koenkk/zigbee-herdsman-converters/pull/4211 , @[pyos](https://github.com/pyos) could you check this?

Cossey, 2022-06-27:
Thanks! I just tested `1.25.2-dev commit: 8adb2f1` and now it works as it did in `1.25.1`!

closed 2022-06-27 by Cossey