Koenkk/zigbee2mqtt #12810: Paulmann 50064 stopped working on 1.25.2

pplu19, 2022-06-13:
### What happened?

I have several Paulmann 50064 that have stop working after i have upgraded to version 1.25.2. The log shows the following warings

Warning 2022-06-13 15:10:26Received message from unsupported device with Zigbee model 'CCT light ' and manufacturer name 'Paulmann Licht GmbH '
Warning 2022-06-13 15:10:26Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

On previous versions they were working as expected

Thanks

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

0x26720700

### Adapter

ConBee2

oberschwabe, 2022-06-13:
same here with the Slaesh's CC2652RB stick.
After upgrading from 1.25.0 to 1.25.2 all Paulmann 50064 got status "not supported" in the GUI and couldn´t be used any more. 
The Paulman color strips have no issue.

I went now back to version 1.25.0 where everything is OK.

pplu19, 2022-06-13:
Same problem with Astuta/ZB-CCT that it was also working on previous versions.

2022-06-13 17:59:17Received message from unsupported device with Zigbee model 'CCT Light' and manufacturer name 'Astuta/ZB-CCT'
2022-06-13 17:59:17Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

I have also updated the conbee to the latest firmware and the issue persits.

Koenkk, 2022-06-14:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/12701

closed 2022-06-14 by Koenkk