Koenkk/zigbee2mqtt #12817: RTCGQ14LM - Timeout changing exposed config

novoa9mad, 2022-06-13:
### What happened?

When trying to change any options exposed it timeouts. The specific options does update correctly.
It fails when changing any of this options:

- motion_sensitivity
- detection_interval
- trigger_indicator

### What did you expect to happen?

I expected to be able to change config options between the available options.
_motion_sensitivity_ between **low**, **medium** and **high**
_detection_interval_ between **2s** and **65535s**
_trigger_indicator_ to **true** or **false**

### How to reproduce it (minimal and precise)

Trying to change any of the exposed options from the frontend or directly publishing a 'set' MQTT message.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220219

### Adapter

Electrolama zzh!

### Debug log

Zigbee2MQTT:debug 2022-06-13 22:29:08: Received MQTT message on 'zigbee2mqtt/motion.kitchen/set' with data '{"motion_sensitivity":"high"}'
Zigbee2MQTT:debug 2022-06-13 22:29:08: Publishing 'set' 'motion_sensitivity' to 'motion.kitchen'
Zigbee2MQTT:debug 2022-06-13 22:29:32: Error: Write 0x54ef441000491dae/1 aqaraOpple({"268":{"value":3,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 20887 - 1 - 211 - 64704 - 4 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)

novoa9mad, 2022-06-13:
@Otnow I think you've been working on this implementation and maybe you could help me out here (I am a newbie here, sorry). Am I doing something wrong?? Maybe I can help debugging more of this if you need it.

Otnow, 2022-06-13:
@novoa9mad, immediately after changing the option, you need to press the pairing button once so that the sensor wakes up and accepts the command.

novoa9mad, 2022-06-13:
@Otnow ok, doing it like that now and it works flawlessly. I weren't doing it quick enough before.

I think that we should add this kind of workarounds explanations on devices' doc pages that need this to be able to work. Maybe like a note on the bottom of the page of the device docs.
What do you think @Koenkk ? Could be helpful for users that bump into this problem not knowing what to do

closed 2022-06-14 by Koenkk