Koenkk/zigbee2mqtt #12818: [New device support]: Outdoor Siren by Lexman (LDSENK07)

Sonyoyo, 2022-06-13:
### Link

https://enki-home.com/produit/sirene-exterieure-connectee-sur-batterie-lexman/

### Database entry

{"id":20,"type":"EndDevice","ieeeAddr":"0x04cd15fffe6390ed","nwkAddr":20687,"manufId":4727,"manufName":"ADEO","powerSource":"Battery","modelId":"LDSENK07","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,3,32,1280,1282,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LDSENK07","manufacturerName":"ADEO","powerSource":3,"zclVersion":3,"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"20200521-15","swBuildId":"2.3"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024c315d8","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":6480}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c315d8","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"20200521-15","swBuildId":"2.3","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1655132684738,"defaultSendRequestWhen":"active"}

### Comments

I started using Home Assistant and Zigbee2MQTT a couple of month ago and this is my first 'unsupported' device.

I tried to add a dedicated converter following the instructions and now the siren looks like it is supported...but actually nothing is really working with the siren. Neither the battery status is showing up correctly.

To fill the converter I took a look to other sirens already supported. However, honestly, I do not know how I can check what really correspond to the functionalities of my device.  

As far as I understood the process (and the Zigbee protocol), to discover features of devices such as remote controller, one can press the buttons and look at the log to see the messages which are sent. But in the case of a siren, no button is available to press and the functionalities are essentially services that can be called or states that can be read...if only I knew their names ! 

Thanks in advance for any help.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LDSENK07'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LDSENK07', // Vendor model number, look on the device for a model number
    vendor: 'ADEO', // Vendor of the device (only used for documentation and startup logging)
    description: 'Lexman Wireless Smart Outdoor Siren', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.battery, fz.ias_siren],
    toZigbee: [tz.warning, tz.squawk],
    exposes: [e.warning(), e.squawk(), e.battery(), e.battery_low(), e.tamper()]
};

module.exports = definition;
```

Sonyoyo, 2022-06-24:
Can I have some help on how to test what I should put in `fromZigbee`, toZigbee` and èxposes`? Is there a place (in the zigbee2mqtt gui or in Home Assistant) where I should try some commands (to call services for example, or to retrieve entity's states or whatever) to define what I should put in the external converter? What would be the syntax and pertinent keywords of such commands?
Thanks for any help...

Koenkk, 2022-06-26:
You can try to execute the squawk and warnings commands from the z2m gui (click on your device -> exposes tab)

Sonyoyo, 2022-06-27:
Thanks for this suggestion.
I tried to manually execute the squawk command, which resulted in the following sequence of messages in the log:

```
Debug 2022-06-27 23:06:40Received MQTT message on 'zigbee2mqtt/Sirena esterna/set' with data '{"warning":{"duration":5,"level":"medium","mode":"burglar","strobe":true,"strobe_duty_cycle":1,"strobe_level":"medium"}}'
Debug 2022-06-27 23:06:40Publishing 'set' 'warning' to 'Sirena esterna'
Debug 2022-06-27 23:09:22Saving state to file /app/data/state.json
Error 2022-06-27 23:10:03Publish 'set' 'squawk' to 'Sirena esterna' failed: 'Error: Command 0x04cd15fffe6390ed/1 ssIasWd.squawk({"squawkinfo":144}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13753 - 1 - 35 - 1282 - 11 after 10000ms)'
Debug 2022-06-27 23:10:03Error: Command 0x04cd15fffe6390ed/1 ssIasWd.squawk({"squawkinfo":144}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13753 - 1 - 35 - 1282 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)
Info 2022-06-27 23:10:04MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'squawk' to 'Sirena esterna' failed: 'Error: Command 0x04cd15fffe6390ed/1 ssIasWd.squawk({\"squawkinfo\":144}, {\"sendWhen\":\"active\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 13753 - 1 - 35 - 1282 - 11 after 10000ms)'","meta":{"friendly_name":"Sirena esterna"},"type":"zigbee_publish_error"}' 
```

Sonyoyo, 2022-06-29:
The siren Is already less than a meter away from the coordinator and I think it was already there when I last paired it.
However, I removed the device and paired again. Then I tried several 'squawk' commands with no effect, even waiting some minutes after them. The log shows the commands sent and the associated (but delayed) error messages that I posted above.

Koenkk, 2022-07-01:
Can you try with the following external converter? Make sure to reapair the device after loading the new converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LDSENK07'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LDSENK07', // Vendor model number, look on the device for a model number
    vendor: 'ADEO', // Vendor of the device (only used for documentation and startup logging)
    description: 'Lexman Wireless Smart Outdoor Siren', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.battery, fz.ias_siren],
    toZigbee: [tz.warning, tz.squawk],
    exposes: [e.warning(), e.squawk(), e.battery(), e.battery_low(), e.tamper()],
    configure: async (device, coordinatorEndpoint, logger) => {
        device.defaultSendRequestWhen = 'immediate';
        device.save();
        await device.getEndpoint(1).unbind('genPollCtrl', coordinatorEndpoint);
    },
};

module.exports = definition;
```

Sonyoyo, 2022-07-02:
Thanks for your support !!!
Now I can call the "warning" and it works with nearly no delay ! Wonderful ! I can also stop it setting `mode=stop`.

`squawk` does not seems to work...but that is not really an issue for me since I can use the "warning" call...am I missing something ?

I can now call the siren from Home Assistant but only passing through the `mqtt.publish` service : I have no action available directly associated to the siren...

What should I then do next ?

Koenkk, 2022-07-02:
Great, added it now for support out-of-the-box.

HA doesn't provide a nice way to integrate a siren device, so you should use the `mqtt.publish` service.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-09 by github-actions[bot]

maxill1, 2023-07-11:
> Thanks for your support !!! Now I can call the "warning" and it works with nearly no delay ! Wonderful ! I can also stop it setting `mode=stop`.
>
> `squawk` does not seems to work...but that is not [...]

Hi, @Sonyoyo,

I own the same device and it seems that any "warning" "mode" I select in z2m UI, _the siren produces the same sound_. 
Is it the same for you?

Also tested publishing via MQTT, same results:
- {"warning": {"mode": "burglar", "duration": 10 }}
- {"warning": {"mode": "fire", "duration": 10 }}
- {"warning": {"mode": "emergency", "duration": 10 }}
- {"warning": {"mode": "police_panic ", "duration": 10 }}
- {"warning": {"mode": "fire_panic", "duration": 10 }}
- {"warning": {"mode": "emergency_panic", "duration": 10 }}

Sonyoyo, 2025-02-10:
Hello @maxill1, I think I did not perform many tests but I have the same feeling: the siren only sounds in one way. 

Hello @mechrekt, here follow an example of script I use to make the siren to sound (you have to change "deviceName" accordingly with your device):

```
alias: Test Siren
sequence:
  - data:
      topic: zigbee2mqtt/deviceName/set
      payload: >-
        {"warning":{"duration":1,"level":"high","mode":"emergency","strobe":true,"strobe_level":"high"}}
    action: mqtt.publish
mode: single
```

mechrekt, 2025-02-10:
Ok, thank you very much for your replay! I'll test it soon!

Is there a way to make it just light without sound?

Sonyoyo, 2025-02-10:
I do not know, but I am interested if you find the way. 😉

mechrekt, 2025-02-10:
Hi,
I used {"warning": {"mode":"burglar","level":"low","strobe_level":"very_high","strobe":true,"strobe_duty_cycle":1,"duration":5}}

Setting level = low my siren doesn't sound, if I set a value from medium upwards, the siren sounds with a volume that is actually getting louder

I did not find a match in the strobe_duty_cycle parameter which seems to have no effect

mechrekt, 2025-02-10:
I noticed also that level "very high" doesn't works.
Only works:
low = silent
medium = medium
high = max (?)