Koenkk/zigbee2mqtt #12822: z2m fallsback to /dev/ttyUSB0 even when other is specified

lukepighetti, 2022-06-14:
### What happened?

Went to bed, everything was working, woke up, none of my zigbee devices were functioning.

Checked logs, z2m container was restarting after error that it couldn't connect to `/dev/ttyUSB0` because the path didn't exist.

I am explicitly providing an alternative in my configuration

```yaml
serial:
  port: tcp://192.168.10.10:6638
```

### What did you expect to happen?

I expected it to use the configuration setting in all failure modes

### How to reproduce it (minimal and precise)

Unfortunately I do not have a MCVE for this issue

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Debug log

Unfortunately I don't have a copy of my logs

lukepighetti, 2022-06-20:
Just happened again. No known stimulus on my part.

lukepighetti, 2022-06-28:
Just happened again, after adding an Aqara door/window sensor. Not sure if it was that or random like the others.

```
Using '/app/data' as data directory

Zigbee2MQTT:info  2022-06-28 19:51:53: Logging to console and directory: '/app/data/log/2022-06-28.19-51-53' filename: log.txt

Zigbee2MQTT:info  2022-06-28 19:51:53: Starting Zigbee2MQTT version 1.25.2 (commit #3c5854f)

Zigbee2MQTT:info  2022-06-28 19:51:53: Starting zigbee-herdsman (0.14.34)

Zigbee2MQTT:error 2022-06-28 19:51:53: Error while starting zigbee-herdsman

Zigbee2MQTT:error 2022-06-28 19:51:53: Failed to start zigbee

Zigbee2MQTT:error 2022-06-28 19:51:53: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions

Zigbee2MQTT:error 2022-06-28 19:51:53: Exiting...

Zigbee2MQTT:error 2022-06-28 19:51:53: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'

    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)

    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)

    at /app/node_modules/@serialport/stream/lib/index.js:242:12

Using '/app/data' as data directory
```

lukepighetti, 2022-06-28:
@Koenkk is there anything I can do to make triage easier?

lukepighetti, 2022-06-29:
I am running it on macOS Docker, and I have no USB devices connected. My coordinator is a network coordinator.

Koenkk, 2022-07-01:
Can you provide the herdsman debug log when the start fails?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

lukepighetti, 2022-07-05:
Last logs while working
```
Zigbee2MQTT:debug 2022-07-05 08:36:44: lumi.plug.maus01: Processed data into payload {"device_temperature":35,"state":"ON","energy":3.6,"consumption":3.6,"power":0}

Zigbee2MQTT:debug 2022-07-05 08:36:44: lumi.plug.maus01: Processed data into payload {"device_temperature":35,"state":"ON","energy":3.6,"consumption":3.6,"power":0}

Zigbee2MQTT:info  2022-07-05 08:36:44: MQTT publish: topic 'zigbee2mqtt/0x00158d0008069966', payload '{"consumption":3.6,"device_temperature":35,"energy":3.6,"linkquality":72,"power":0,"state":"ON","voltage":null}'

Zigbee2MQTT:debug 2022-07-05 08:38:49: Saving state to file /app/data/state.json
```

Container was restarted, fail loop begins (this loops over and over)
```
Using '/app/data' as data directory

Zigbee2MQTT:info  2022-07-05 08:46:21: Logging to console and directory: '/app/data/log/2022-07-05.08-46-21' filename: log.txt

Zigbee2MQTT:info  2022-07-05 08:46:21: Starting Zigbee2MQTT version 1.25.2 (commit #3c5854f)

Zigbee2MQTT:info  2022-07-05 08:46:21: Starting zigbee-herdsman (0.14.34)

Zigbee2MQTT:error 2022-07-05 08:46:22: Error while starting zigbee-herdsman

Zigbee2MQTT:error 2022-07-05 08:46:22: Failed to start zigbee

Zigbee2MQTT:error 2022-07-05 08:46:22: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions

Zigbee2MQTT:error 2022-07-05 08:46:22: Exiting...

Zigbee2MQTT:error 2022-07-05 08:46:22: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'

    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)

    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)

    at /app/node_modules/@serialport/stream/lib/index.js:242:12
```

Here's my data folder dump... however, the configuration is overridden with a local config file, and it is known to work

[zigbee2mqtt-data.zip](https://github.com/Koenkk/zigbee2mqtt/files/9046479/zigbee2mqtt-data.zip)

```yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mqtt
serial:
  port: tcp://192.168.4.40:6638
frontend:
  port: 8080
advanced:
  log_level: debug
devices:
  '0x00158d0008027368':
    friendly_name: '0x00158d0008027368'
    description: main
  '0x00158d000804a462':
    friendly_name: '0x00158d000804a462'
  '0x00158d00086516ea':
    friendly_name: '0x00158d00086516ea'
  '0x00158d0008068b1d':
    friendly_name: '0x00158d0008068b1d'
  '0x00158d0007e59d24':
    friendly_name: '0x00158d0007e59d24'
  '0x00158d0008069966':
    friendly_name: '0x00158d0008069966'
  '0x00158d00080689c5':
    friendly_name: '0x00158d00080689c5'
  '0x00158d000804995a':
    friendly_name: '0x00158d000804995a'
```

```yaml
  zigbee2mqtt:
    container_name: zigbee2mqtt
    restart: unless-stopped
    image: koenkk/zigbee2mqtt
    volumes:
      - ~/.dockerfs/zigbee2mqtt-data:/app/data
      - ./zigbee2mqtt-configuration.yaml:/app/data/configuration.yaml <<<<<<<<<<
    depends_on:
      - mqtt
    ports:
      - 8080:8080
    environment:
      - TZ=America/New_York
    extra_hosts:
      coordinator: 192.168.4.40
```

Koenkk, 2022-07-05:
I need the herdsman debug logging (this is not logged to a file but only to STDOUT).

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

lukepighetti, 2022-07-08:
OK, it looks like this is an issue with Docker. I'll summarize

```yaml
  zigbee2mqtt:
    container_name: zigbee2mqtt
    restart: unless-stopped
    image: koenkk/zigbee2mqtt
    volumes:
      - ~/.dockerfs/zigbee2mqtt-data:/app/data
      - ./zigbee2mqtt-configuration.yaml:/app/data/configuration.yaml
```

As you can see, we have z2m `/app/data` being stored in my home folder, but we have `/app/data/configuration.yaml` checked into my repository.

For some reason, on a fresh start, docker-compose will happily handle these nested volumes. However, on a restart it appears that these volume maps degrade, and the second entry is ignored.

As far as I know, it is a strong best practice to have `/app/data` separate from configuration files, which typically live in `/config` or `/app/config`. (You can see ESPHome for an example of this, but I have seen it elsewhere).

This allows the consumer of the Docker image to specify a place for things like database files to be saved on disk, while checking configuration files into source control, without having these odd nested volumes.

Koenkk, 2022-07-08:
What if you would create a symlink under `~/.dockerfs/zigbee2mqtt-data/configuration.yaml` which points to `/config/configuration.yaml`?

lukepighetti, 2022-07-08:
As always, there is an escape hatch. But I do think the enhancement request is sound and reasonable.

lukepighetti, 2022-07-09:
Another option would be an environment variable to specify the path of the configuration file

Koenkk, 2022-07-09:
> This enhancement request will separate z2m data from z2m configuration. This separates items that should be checked into version control from items that should be periodically backed up.

I see your point, but the `.gitignore` is also a good solution IMO, lets see how many people are interested in this

closed 2022-08-17 by github-actions[bot]

gabrielebesta, 2024-02-10:
+1
Separate z2m data from z2m configuration would be very useful in use with docker.  The configuration file could be configured with a bind from the host, while the z2m data with a docker volume.