Koenkk/zigbee2mqtt #12825: Template variable warning: 'dict object' has no attribute

uvjim, 2022-06-14:
### What happened?

When I restart Home Assistant I get warnings as per below.  I believe the `device_temperature` warning only appeared when I joined an Aqara RTCZCGQ11LM device to the mesh, so pretty sure that's the culprit for that entry.

I believe the `link_quality` warnings probably come from the other Aqara devices I have: -

- MCCGQ11LM (7 of those)
- QBKG21LM (1 of these)
- QBKG21LM (1 of these)
- RTCGQ11LM (1 of these)

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:1850
First occurred: 18:35:24 (12 occurrences)
Last logged: 18:35:50

Template variable warning: 'dict object' has no attribute 'linkquality' when rendering '{{ value_json.linkquality }}'
Template variable warning: 'dict object' has no attribute 'device_temperature' when rendering '{{ value_json.device_temperature }}'
```

### What did you expect to happen?

No warnings.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

0x26720700

### Adapter

RaspBee2

uvjim, 2022-07-15:
Sure...

```json
{
  "external_converters": [],
  "devices": [
    "devices.yaml"
  ],
  "groups": [
    "groups.yaml"
  ],
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "version": 5,
    "server": "mqtt://core-mosquitto:1883",
    "user": "mqtt",
    "password": **REDACTED**,
    "include_device_information": true
  },
  "serial": {
    "port": "/dev/ttyAMA0",
    "adapter": "deconz"
  },
  "advanced": {
    "log_level": "warn",
    "pan_id": **REDACTED**,
    "channel": 11,
    "network_key": **REDACTED**,
    "availability_blocklist": [],
    "availability_passlist": [],
    "legacy_api": false,
    "last_seen": "ISO_8601"
  },
  "device_options": {},
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "frontend": {
    "port": 8099
  },
  "experimental": {},
  "availability": false,
  "zigbee_herdsman_debug": false
}
```

Koenkk, 2022-07-15:
Do you have a [template sensor](https://www.home-assistant.io/integrations/template/) section in your HA `configuration.yaml`?

uvjim, 2022-07-15:
I do, but it only has a sensor and a binary_sensor that aren't related to anything being supplied by zigbee2mqtt.

Koenkk, 2022-07-15:
Could you provide the debug log when starting z2m ?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

uvjim, 2022-07-15:
Do you need Zigbee-herdsman logging as well?

uvjim, 2022-07-15:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9122158/log.txt)

Does this help?  The HA log didn't show the template warning after restarting the add-on so I could enable debug then restart both HA Core and the add-on if that would help at all.

Koenkk, 2022-07-17:
When exactly does the error appear? If it looks at the published payloads, all seem to contains values for `device_temperature` and `linkquality`.

uvjim, 2022-07-17:
It happens when HA core starts.

Koenkk, 2022-07-17:
@emontnemery can you confirm that this error can come from the MQTT integration?

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:1850
First occurred: 18:35:24 (12 occurrences)
Last logged: 18:35:50

Template variable warning: 'dict object' has no attribute 'linkquality' when rendering '{{ value_json.linkquality }}'
Template variable warning: 'dict object' has no attribute 'device_temperature' when rendering '{{ value_json.device_temperature }}'
```

uvjim, 2022-08-17:
The warnings still appear in HASS 2022.8.0 and Zigbee2MQTT 1.27.0

uvjim, 2022-09-18:
The warnings still appear in HASS 2022.9.4 and Zigbee2MQTT 1.27.2

Koenkk, 2022-09-19:
Can you provide the errors in HA again, if I understood correctly HA improved the error message to better determine what causes the issues.

uvjim, 2022-09-20:
Sure, does any additional logging need to be turned on?

uvjim, 2022-09-20:
This was seen after a reboot this morning with no additional logging turned on...

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:582
First occurred: 10:02:16 (24 occurrences)
Last logged: 10:02:16

Template variable warning: 'dict object' has no attribute 'linkquality' when rendering '{{ value_json.linkquality }}'
Template variable warning: 'dict object' has no attribute 'device_temperature' when rendering '{{ value_json.device_temperature }}'
Template variable warning: 'dict object' has no attribute 'power_outage_count' when rendering '{{ value_json.power_outage_count }}'
```

Koenkk, 2022-09-21:
@jbouwh if I understood correctly, your value template improvement should be in 2022.9.4, any idea why it doesn't show up? (see comment above)

jbouwh, 2022-09-21:
This happens if the (json) payload not contains the values specified in the template.

jbouwh, 2022-09-21:
> Can you provide the errors in HA again, if I understood correctly HA improved the error message to better determine what causes the issues.

The entity_id is logged now. But it is needed to see what the configuration payload is and the color mode payload causing the issue.

Koenkk, 2022-09-21:
@jbouwh this case is not about the color mode payload (which was addressed by you in https://github.com/home-assistant/core/pull/77909). Could the entity ID also be + dict object dump also logged for these messages?

```
Template variable warning: 'dict object' has no attribute 'device_temperature' when rendering '{{ value_json.device_temperature }}'
```

Something like this would be great:

```
Template variable warning: '{"state": "ON", "temperature": 9}' has no attribute 'device_temperature' when rendering '{{ value_json.device_temperature }}' for entity id sensor.kitchen
```

uvjim, 2022-09-21:
If you can get that info it would be great because I can probably narrow down more about the devices... Currently I think it's these ones: -

MCCGQ11LM (7 oglf these)
QBKG21LM (1 of these)
QBKG21LM (1 of these)
RTCGQ11LM (1 of these)

jbouwh, 2022-09-21:
> @jbouwh this case is not about the color mode payload (which was addressed by you in [home-assistant/core#77909](https://github.com/home-assistant/core/pull/77909)). Could the entity ID also be + dict object dump also logged for these messages?
> 
> ```
> Template variable warning: 'dict object' has no attribute 'device_temperature' when rendering '{{ value_json.device_temperature }}'
> ```
> 
> Something like this would be great:
> 
> ```
> Template variable warning: '{"state": "ON", "temperature": 9}' has no attribute 'device_temperature' when rendering '{{ value_json.device_temperature }}' for entity id sensor.kitchen
> ```

Ah ok, you want the entity id logged for these warnings. Sorry for the confusion.

jbouwh, 2022-09-23:
This PR https://github.com/home-assistant/core/pull/79016 should add debug logging for incoming payloads with value templates and outgoing payloads with command templates

ronniebee, 2022-09-25:
Having the same issue: 
Logger: homeassistant.helpers.template
Source: helpers/template.py:582
First occurred: 00:14:44 (13158 occurrences)
Last logged: 09:23:01

Template variable warning: 'dict object' has no attribute 'hum' when rendering '{{ value_json.hum }}'
Template variable warning: 'dict object' has no attribute 'temp' when rendering '{{ value_json.temp }}'

My logs say:

2022-09-25T07:20:58.125Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,118,194,1,1,0,39,0,150,232,163,0,0,8,24,88,10,8,5,33,0,0,118,194,29] - 95
2022-09-25T07:20:58.126Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":49782,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":39,"securityuse":0,"timestamp":10741910,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,88,10,8,5,33,0,0]}}
2022-09-25T07:20:58.133Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":88,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1288,"dataType":33,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":49782,"endpoint":1,"linkquality":39,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-09-25T07:20:58.146Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-09-25T07:21:03.122Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,33,68,129,0,0,4,11,118,194,1,1,0,36,0,103,172,168,0,0,13,24,118,10,8,5,33,30,0,5,5,33,235,0,118,194,29,32]
2022-09-25T07:21:03.122Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,33,68,129,0,0,4,11,118,194,1,1,0,36,0,103,172,168,0,0,13,24,118,10,8,5,33,30,0,5,5,33,235,0,118,194,29,32]
2022-09-25T07:21:03.122Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 33 - 2 - 4 - 129 - [0,0,4,11,118,194,1,1,0,36,0,103,172,168,0,0,13,24,118,10,8,5,33,30,0,5,5,33,235,0,118,194,29] - 32
2022-09-25T07:21:03.123Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":49782,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":11054183,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[24,118,10,8,5,33,30,0,5,5,33,235,0]}}
2022-09-25T07:21:03.128Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":118,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1288,"dataType":33,"attrData":30},{"attrId":1285,"dataType":33,"attrData":235}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":49782,"endpoint":1,"linkquality":36,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-09-25T07:21:03.154Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-09-25T07:21:05.136Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,4,11,118,194,1,1,0,33,0,78,152,170,0,0,8,24,128,10,11,5,41,2,0,118,194,29,41]
2022-09-25T07:21:05.137Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,4,11,118,194,1,1,0,33,0,78,152,170,0,0,8,24,128,10,11,5,41,2,0,118,194,29,41]
2022-09-25T07:21:05.137Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,118,194,1,1,0,33,0,78,152,170,0,0,8,24,128,10,11,5,41,2,0,118,194,29] - 41
2022-09-25T07:21:05.137Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":49782,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":33,"securityuse":0,"timestamp":11180110,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,128,10,11,5,41,2,0]}}
2022-09-25T07:21:05.143Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":128,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1291,"dataType":41,"attrData":2}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":49782,"endpoint":1,"linkquality":33,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-09-25T07:21:05.154Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-09-25T07:21:08.129Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,4,11,118,194,1,1,0,33,0,26,114,173,0,0,8,24,141,10,8,5,33,0,0,118,194,29,148]
2022-09-25T07:21:08.129Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,4,11,118,194,1,1,0,33,0,26,114,173,0,0,8,24,141,10,8,5,33,0,0,118,194,29,148]
2022-09-25T07:21:08.130Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,118,194,1,1,0,33,0,26,114,173,0,0,8,24,141,10,8,5,33,0,0,118,194,29] - 148
2022-09-25T07:21:08.130Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":49782,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":33,"securityuse":0,"time [... cut ...]
2022-09-25T07:21:55.147Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":156,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1288,"dataType":33,"attrData":30}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":49782,"endpoint":1,"linkquality":39,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-09-25T07:21:55.170Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

uvjim, 2022-10-06:
Hello... now that Home Assistant 2022.10.0 is out and the PR above was merged.  I thought I'd turn on debugging and post the results here for you.  Hope it helps track this down.

```
2022-10-06 19:01:28.558 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"contact":true,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"last_seen":"2021-03-16T20:53:42.270Z","temperature":19,"voltage":3015}' with variables {'entity_id': 'sensor.front_door_battery', 'name': 'Front Door battery', 'this': <template TemplateStateFromEntityId(sensor.front_door_battery)>} with default value 'None' and Template("{{ value_json.battery }}")
2022-10-06 19:01:28.559 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"contact":true,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"last_seen":"2021-03-16T20:53:42.270Z","temperature":19,"voltage":3015}' with variables {'entity_id': 'sensor.front_door_voltage', 'name': 'Front Door voltage', 'this': <template TemplateStateFromEntityId(sensor.front_door_voltage)>} with default value 'None' and Template("{{ value_json.voltage }}")
2022-10-06 19:01:28.561 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"contact":true,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"last_seen":"2021-03-16T20:53:42.270Z","temperature":19,"voltage":3015}' with variables {'entity_id': 'sensor.front_door_linkquality', 'name': 'Front Door linkquality', 'this': <template TemplateStateFromEntityId(sensor.front_door_linkquality)>} with default value 'None' and Template("{{ value_json.linkquality }}")
2022-10-06 19:01:28.562 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'linkquality' when rendering '{{ value_json.linkquality }}'
2022-10-06 19:01:28.578 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"contact":true,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"last_seen":"2021-03-16T20:53:42.270Z","temperature":19,"voltage":3015}' with variables {'entity_id': 'sensor.front_door_device_temperature', 'name': 'Front Door device temperature', 'this': <template TemplateStateFromEntityId(sensor.front_door_device_temperature)>} with default value 'None' and Template("{{ value_json.device_temperature }}")
2022-10-06 19:01:28.578 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'device_temperature' when rendering '{{ value_json.device_temperature }}'
2022-10-06 19:01:28.581 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"contact":true,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"last_seen":"2021-03-16T20:53:42.270Z","temperature":19,"voltage":3015}' with variables {'entity_id': 'sensor.front_door_power_outage_count', 'name': 'Front Door power outage count', 'this': <template TemplateStateFromEntityId(sensor.front_door_power_outage_count)>} with default value 'None' and Template("{{ value_json.power_outage_count }}")
2022-10-06 19:01:28.581 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'power_outage_count' when rendering '{{ value_json.power_outage_count }}'
2022-10-06 19:01:28.603 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Lounge - Single","hardwareVersion":1,"ieeeAddr":"0x60a423fffeae3467","manufacturerID":4098,"manufacturerName":"_TYZB01_mtunwanm","model":"TS011F_wall_outlet","networkAddress":26855,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2022-10-06T18:01:27.947Z","linkquality":255,"state":"OFF"}' with variables {'entity_id': 'sensor.lounge_single_linkquality', 'name': 'Lounge - Single linkquality', 'this': <template TemplateStateFromEntityId(sensor.lounge_single_linkquality)>} with default value 'None' and Template("{{ value_json.linkquality }}")
[... cut ...] 3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"device_temperature":21,"last_seen":"2022-10-06T17:18:20.255Z","linkquality":255,"power_outage_count":83,"temperature":20,"voltage":3005}' with variables {'entity_id': 'sensor.lounge_patio_door_power_outage_count', 'name': 'Lounge - Patio Door power outage count', 'this': <template TemplateStateFromEntityId(sensor.lounge_patio_door_power_outage_count)>} with default value 'None' and Template("{{ value_json.power_outage_count }}")
```

Koenkk, 2022-10-08:
@uvjim I cannot explain why this happens yet. Zigbee2MQTT should set all the properties for which it doesn't have values to `null`: https://github.com/Koenkk/zigbee2mqtt/blob/03ba647dc6b5f299f8f3ab441712999fcb3a253e/lib/extension/homeassistant.ts#L1410

Can you provide me the z2m debug log the moment these errors appear in HA?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

uvjim, 2022-10-09:
Hmmm... interestingly, I switched the logs to debug mode, restarted the HA add-on and restarted HA core. However, the log files are completely empty.  I'll try again and see what happens.

Koenkk, 2022-10-09:
Checking e.g. 

```
2022-10-09 15:11:55.347 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"contact":true,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"last_seen":"2021-03-16T20:53:42.270Z","temperature":19,"voltage":3015}' with variables {'entity_id': 'sensor.front_door_power_outage_count', 'name': 'Front Door power outage count', 'this': <template TemplateStateFromEntityId(sensor.front_door_power_outage_count)>} with default value '' and Template("{{ value_json.power_outage_count }}")
2022-10-09 15:11:55.347 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'power_outage_count' when rendering '{{ value_json.power_outage_count }}'
```

Device `0x00158d0003ceb90a` = `zigbee2mqtt/Front Door`

All published message in the log are:

```
info  2022-10-09 15:02:51: MQTT publish: topic 'zigbee2mqtt/Front Door', payload '{"battery":100,"contact":false,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"device_temperature":18,"last_seen":"2022-10-09T13:15:56.098Z","linkquality":208,"power_outage_count":184,"temperature":18,"voltage":3025}'
info  2022-10-09 15:02:52: MQTT publish: topic 'zigbee2mqtt/Front Door', payload '{"battery":100,"contact":false,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"device_temperature":18,"last_seen":"2022-10-09T13:15:56.098Z","linkquality":208,"power_outage_count":184,"temperature":18,"voltage":3025}'
info  2022-10-09 15:08:48: MQTT publish: topic 'zigbee2mqtt/Front Door', payload '{"battery":100,"contact":false,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"device_temperature":18,"last_seen":"2022-10-09T14:06:07.231Z","linkquality":255,"power_outage_count":184,"temperature":18,"voltage":3025}'
info  2022-10-09 15:08:49: MQTT publish: topic 'zigbee2mqtt/Front Door', payload '{"battery":100,"contact":false,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"device_temperature":18,"last_seen":"2022-10-09T14:06:07.231Z","linkquality":255,"power_outage_count":184,"temperature":18,"voltage":3025}'
info  2022-10-09 15:12:29: MQTT publish: topic 'zigbee2mqtt/Front Door', payload '{"battery":100,"contact":false,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"device_temperature":18,"last_seen":"2022-10-09T14:06:07.231Z","linkquality":255,"power_outage_count":184,"temperature":18,"voltage":3025}'
info  2022-10-09 15:12:30: MQTT publish: topic 'zigbee2mqtt/Front Door', payload '{"battery":100,"contact":false,"device":{"applicationVersion":3,"friendlyName":"Front Door","hardwareVersion":2,"ieeeAddr":"0x00158d0003ceb90a","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":2696,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"device_temperature":18,"last_seen":"2022-10-09T14:06:07.231Z","linkquality":255,"power_outage_count":184,"temperature":18,"voltage":3025}'
```

But all contain `power_outage_count`.

Therefore I think these are retained MQTT messages (probably you set `retain: true` for this device before and now switched back to `retain: false`, you can clear them by:
- Clearing your mqtt broker database
- OR publishing to `zigbee2mqtt/Front Door` an empty message with retain enabled (you can use e.g. MQTT.fx)

uvjim, 2022-10-09:
Hmmm... I don't remember setting anything to retain. I'll clear the broker database and see what happens. I know, certainly for the newer devices this definitely wasn't done. I wonder if something is stuck somewhere though.

uvjim, 2022-10-09:
OK.  So it appears that there was something stuck somewhere and I don't ever remember turning `retain` on at any point.  I unsinstalled the MQTT addon in Home Assistant, reinstalled it, reconfigured the integraion and everything kicked back to life.  I then restarted Home Assistant core a few times to see if the messages came back.  They haven't so all seems good now.

Thank you for you time looking into this for me and pointing me in the right direction.

Koenkk, 2022-10-10:
Great that you've solved it! Assuming this can be closed now.

closed 2022-10-10 by Koenkk