Koenkk/zigbee2mqtt #12828: [New device support]: RTX Tuya ZigBee ZCB16-2P

bdkacz, 2022-06-15:
### Link

https://rtx24.eu/products/bezpiecznik-sterowany-2x-16a-zigbee-tuya-pomiar-energii

### Database entry

{"id":22,"type":"Router","ieeeAddr":"0xa4c138bf020d6e90","nwkAddr":37821,"manufId":4417,"manufName":"_TZE200_abatw3kj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"ďż˝K<*iďż˝K<*iďż˝K<*iďż˝K<*iďż˝K<*iďż˝K<*i","65506":34,"65508":0,"appVersion":64,"modelId":"TS0601","manufacturerName":"_TZE200_abatw3kj","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a8d6","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1655272767284,"defaultSendRequestWhen":"immediate"}

### Comments

Hello. 

I bought product from link. Based on https://zigbee.blakadder.com/RTX_ZCB16-2P.html it should be compatibile with zigbee2mqtt, but my vendorid is different - _TZE200_abatw3kj . So i Try to write attached converter, but it doesn't work and I get in log errors :

> debug 2022-06-15 07:51:09: Received Zigbee message from 'RTX', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":16}],"seq":53541}' from endpoint 1 with groupID null
> info  2022-06-15 07:51:09: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
> debug 2022-06-15 07:51:09: Received Zigbee message from 'RTX', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":16}],"seq":53541}' from endpoint 1 with groupID null
> debug 2022-06-15 07:51:09: Received Zigbee message from 'RTX', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":16}],"seq":53541}' from endpoint 1 with groupID null
> 
> info  2022-06-15 07:51:11: Configuring 'RTX'
> error 2022-06-15 07:51:11: Failed to configure 'RTX', attempt 1 (Error: Bind 0xa4c138bf020d6e90/1 genOnOff from '0x00212effff06a8d6/1' failed (Error: status: 130)
>     at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:765:19)
>     at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
>     at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9)
>     at Object.configure (/app/dist/util/externally-loaded.js:18:13)
>     at Configure.configure (/app/lib/extension/configure.ts:115:13)
>     at Configure.start (/app/lib/extension/configure.ts:68:13)
>     at Controller.callExtensions (/app/lib/controller.ts:317:17)
>     at Controller.start (/app/lib/controller.ts:154:9)
>     at start (/app/index.js:109:5))
> 
> 
> debug 2022-06-15 07:52:16: Received Zigbee message from 'RTX', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":16}],"seq":59685}' from endpoint 1 with groupID null
> warn  2022-06-15 07:52:16: zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #16 with data {"dp":16,"datatype":1,"data":{"type":"Buffer","data":[0]}}
> debug 2022-06-15 07:52:16: Received Zigbee message from 'RTX', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,3],"type":"Buffer"},"datatype":2,"dp":1}],"seq":59941}' from endpoint 1 with groupID null
> info  2022-06-15 07:52:16: MQTT publish: topic 'zigbee2mqtt/RTX', payload '{"current":null,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"RTX","hardwareVersion":1,"ieeeAddr":"0xa4c138bf020d6e90","manufacturerID":4417,"manufacturerName":"_TZE200_abatw3kj","model":"ZCB16-2P","networkAddress":37821,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":null,"linkquality":232,"power":null,"state":"ON","voltage":null}'
> debug 2022-06-15 07:52:16: Received Zigbee message from 'RTX', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":5,"dp":9}],"seq":60197}' from endpoint 1 with groupID null
> warn  2022-06-15 07:52:16: zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #9 with data {"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}
> debug 2022-06-15 07:52:16: Received Zigbee message from 'RTX', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,126,0,0,0,0,0,0],"type":"Buffer"},"datatype":0,"dp":6}],"seq":60453}' from endpoint 1 with groupID null
> warn  2022-06-15 07:52:16: zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #6 with data {"dp":6,"datatype":0,"data":{"type":"Buffer","data":[9,126,0,0,0,0,0,0]}}
> debug 2022-06-15 07:52:16: Received Zigbee message from 'RTX', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1,1,0,63,3,1,1,19,4,1,0,175],"type":"Buffer"},"datatype":0,"dp":18}],"seq":60709}' from endpoint 1 with groupID null
> info  2022-06-15 07:52:16: MQTT publish: topic 'zigbee2mqtt/RTX', payload '{"current":null,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"RTX","hardwareVersion":1,"ieeeAddr":"0xa4c138bf020d6e90","manufacturerID":4417,"manufacturerName":"_TZE200_abatw3kj","model":"ZCB16-2P","networkAddress":37821,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":null,"linkquality":220,"power":null,"state":"ON","voltage":null}'
> debug 2022-06-15 07:52:16: Received Zigbee message from 'RTX', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[69,84,85,49,45,73,79,84,45,90,105,103,98,101,101,0],"type":"Buffer"},"datatype":3,"dp":19}],"seq":60965}' from endpoint 1 with groupID null
> 
> 
> 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

 const definition = {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_abatw3kj'} ],
        model: 'ZCB16-2P',
        vendor: 'RTX',
        description: 'Circuit breaker with meter',
        fromZigbee: [fz.tuya_dinrail_switch],
        toZigbee: [tz.tuya_switch_state],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        },
        exposes: [e.switch().setAccess('state', ea.STATE_SET), e.voltage(), e.power(), e.current(), e.energy()],
    };

module.exports = definition;
```

bdkacz, 2022-06-15:
Under data point 6 probably we have  info about current,power,voltage  but how convert this to usable data ? 

> {"dp":6,"datatype":0,"data":{"type":"Buffer","data":[9,116,0,0,0,0,0,0]}}

bdkacz, 2022-06-18:
Here is partially working converter with some notes :

1) Readings of voltage is completly nonsense - somtimes 97V,where real voltage is 230V, sometimes 112V where Voltage is 232V etc.  I suspect more and more that the smartlife application counts the voltage using the P / I method, and not the voltage reading from the device

2) I don't know how to send to MQTT/HA data from PIDS:

9 - fault 
17 - alarm set 1
18 - alarm set 2

18 are settings from  smartlife apps where you can set low voltage,hight voltage and current alarm

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const fzLocal = {
    tuya_dinrail_switch: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            for (const dpValue of msg.data.dpValues) {
                const value = tuya.getDataValue(dpValue);
                const dp = dpValue.dp
                meta.logger.info(`RECEIVED DP #${dp} -- VALUE = ${value}`);

                switch (dp) {
                    case 16: // DPID that we added to common
                        return {
                            state: value ? 'ON' : 'OFF'
                        };
                    case 1:
                        return {
                            energy: value / 100
                        };
                    case 6:   
                     return {
                            current: (value[4] / 1000), voltage: (value[1]), power: value[7]
                        };
                    case 9:{
                        meta.logger.warn(`zigbee-herdsman-converters:: NOT RECOGNIZED DP ` +
                            `#${dp} with data ${JSON.stringify(msg.data)} VALUE = ${value}`);
                    };
                    case 17:{
                        meta.logger.warn(`zigbee-herdsman-converters:: NOT RECOGNIZED DP ` +
                            `#${dp} with data ${JSON.stringify(msg.data)} VALUE = ${value}`);
                    };
                    case 18:{
                        meta.logger.warn(`zigbee-herdsman-converters:: NOT RECOGNIZED DP ` +
                            `#${dp} with data ${JSON.stringify(msg.data)} VALUE = ${value}`);
                    };
                    default: {
                        meta.logger.warn(`zigbee-herdsman-converters:: NOT RECOGNIZED DP ` +
                            `#${dp} with data ${JSON.stringify(msg.data)} VALUE = ${value}`);
                    };
                }
            }
        },
    },
};

const tzLocal = {
    state: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, 16, value === 'ON');
        },
    },

};

const definition = {
    fingerprint: [{
        modelID: 'TS0601',
        manufacturerName: '_TZE200_abatw3kj'
    }],
    model: 'ZCB16-2P',
    vendor: 'RTX',
    extend: extend.switch(),
    description: 'Circuit breaker with meter',
    fromZigbee: [fzLocal.tuya_dinrail_switch, ],
    toZigbee: [tzLocal.state],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.switch().setAccess('state', ea.STATE_SET), e.voltage(), e.power(), e.current(), e.energy()],
};

module.exports = definition;
```

closed 2022-07-26 by github-actions[bot]

Autogarsas, 2022-10-07:
Hello, do you manage to make it work? Im interested in ZCB16-2P and ZCB25-4P, price is very reasonable, but not sure if it will work with z2m?

bdkacz, 2022-10-07:
> Hello, do you manage to make it work? Im interested in ZCB16-2P and ZCB25-4P, price is very reasonable, but not sure if it will work with z2m?

I can only turn it on and off and read power and total energy. Voltage is  incorrect, current also probably is wrong.

Autogarsas, 2022-10-07:
Maybe it doesnt support voltage and current (cant see that in device description) Actually im searching for energy consumption device. So this one should work for me?

bdkacz, 2022-10-07:
But when I connect it via tuya gateway and smarlife app id display voltage and current  values correctly, so there is problem with converting data from device to correct values

If You only want to know how many energy ( current in Watts and total in kWh)  use your device my converter show it correctly. But if You want to read voltage or current- sorry I can't get correct values

Autogarsas, 2022-10-07:
Do you think it will work for 4P version too? 
https://www.rtx24.pl/bezpiecznik-sterowany-4x-25a-zigbee-tuya-p-476.html

Also, ON/OFF works for all breakers together? You cant control them separate?

P.S. In my opinion very underrated device, on ali similar devices are 2x in price.

bdkacz, 2022-10-07:
About 4P-I don't know

About turn on-off - both z2m and smartlife turn on/off it together (L and N)

AdamWeglarz, 2023-02-21:
@Autogarsas did you purchase 4 channel device?

PetrozPL, 2023-03-05:
I've done  some additional research on this device and got some new functionality working (thanks to materials from https://blog.ravindra-job.com/iot/tuya_meter/) : 
> fixed current, power and voltage measurement
> read overcurrent, overvoltage and undervoltage tresholds
> set overcurrent, overvoltage and undervoltage tresholds (overcurrent value change is working, checked it)
> read alarm occurence but not parsed correctly
> read meter_number (string)

I would like to prepare the complete configuration for this device. Can someone help me? Below is my current development configuration for this device. 

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const fzLocal = {
    tuya_dinrail_switch: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            for (const dpValue of msg.data.dpValues) {
                const value = tuya.getDataValue(dpValue);
                const dp = dpValue.dp
                //meta.logger.info(`RECEIVED DP #${dp} -- VALUE = ${value}`);

                switch (dp) {
                    case 16: // DPID that we added to common
                        return {
                            state: value ? 'ON' : 'OFF'
                        };
                    case 1:
                        return {
                            energy: value / 100
                        };
                    case 6:
                     return {
                            current: ((value[4] | value[3] << 8) / 1000), 
                            voltage: ((value[1] | value[0] << 8) / 10), 
                            power: ((value[7] | value[6] << 8)),
                            dp6_i2: value[2],
                            dp6_i5: value[5]
                        };
                    case 9:{
                      const alarm_lookup = {
                          0: 'clear',
                          1: 'over current threshold',
                          2: 'over power threshold',
                          4: 'over voltage threshold',
                          8: 'Under voltage threshold',
                       };
                       return { 
                           alarm : alarm_lookup[value[0]],
                           alarm_int : value
                       } ;
                    }
                    case 18:
                        const name_lookup = {
                           0: 'not set',
                           1: 'Over current threshold',
                           2: 'Over power threshold',
                           3: 'Over voltage threshold',
                           4: 'Under voltage threshold',
                        };
                        const unit_lookup = {
                           0: '-',
                           1: 'A',
                           3: 'V',
[... 123 lines cut ...]
              exposes.numeric('threshold_1_value', ea.STATE),
              exposes.text('threshold_2', ea.STATE),
              exposes.binary('threshold_2_state', ea.STATE, 'ON', 'OFF'),
              exposes.numeric('threshold_2_value', ea.STATE),
              exposes.text('threshold_3', ea.STATE),
              exposes.binary('threshold_3_state', ea.STATE, 'ON', 'OFF'),
              exposes.numeric('threshold_3_value', ea.STATE),
              exposes.text('alarm', ea.STATE),
			  exposes.text('meter_number',ea.STATE),
			  exposes.numeric('dp6_i2',ea.STATE),
			  exposes.numeric('dp6_i5',ea.STATE),
			  exposes.numeric('alarm_int',ea.STATE),
              ]
};

module.exports = definition;`

The things that I would like to add or check : 
 > the functionality of the second byte of treshold_n register (thought that it's enable/disable alarm, but alarm is reported despite the value set on this byte - 0 or 1).
 > the automatic update of 'alarm_setting values while obtaining report from device (now values to set are unknown until set by user)
 > research alarm reporting - when alarm is not set tp 9 is reporting only single byte, but when alarm is set, two bytes are read instead of 1. Below is this situation:

ALARM_NSET : Debug 2023-03-05 12:45:23Received Zigbee message from 'Test_RCBO_P2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"**data":{"data":[0]**,"type":"Buffer"},"datatype":5,"dp":9}],"seq":23555}' from endpoint 1 with groupID 0 --> **alarm_int = 0**

ALARM_SET : Debug 2023-03-05 12:38:26Received Zigbee message from 'Test_RCBO_P2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"**data":{"data":[1,0]**,"type":"Buffer"},"datatype":5,"dp":9}],"seq":53250}' from endpoint 1 with groupID 0 --> **alarm_int = 256**

bdkacz, 2023-03-05:
I tested your converter with my RTX ZCB16-2P and looks working  fine

PetrozPL, 2023-03-05:
Yes, alarms tresholds are read and set correctly but I still don't know what is the function of the second byte - as wrote before, I initially assumed that it's ON/OFF for particular treshold alarm, but it's not - when I was checking it with overcurrent treshold I've set it to 1 and next time to 0, and alarm was triggered in both situations. So this second byte is not ON/OFF for treshold alarm. The second thing that is still not working correctly is Alarm - as I wrote, the dp 6 changes it's size depending on state of alarm. When alarm is not triggered, the size is 1 byte. When alarm was triggered it returns 2 bytes. The third problem is linked to configuration of tresholds - as on your print screen - the Set Alarm values are unknown, and I would like to update them at least once with values read from device (to have the starting point for modification). Do you have a compatible gate to use this device with tuya application? Could you send me the screenshots of possible configuration actions for this device (setting treshold/alarms etc)?

PetrozPL, 2023-03-07:
Ok, just checked the 4P device and this one seems to be detected the same as 2P. Unfortunately it seems to be not fully operational. I can power on/off the device, read all alarm thresholds and set them correctly. But when it comes to measurements, only voltage is measured. Current, power and energy seems to be constantly set to 0, no matter which phase is used. And there's only a single phase reported - no additional data for other phases. So it seems that either my 4P device is broken or whole implementation for 4P device is broken. Does anyone else has this 4P device and can provide any feedback?

kacprzak-michael, 2023-06-15:
@PetrozPL I just installed it. Maybe I could help 😄

nexustp, 2023-07-21:
Hi I have a similar device: 

https://www.rtx24.pl/bezpiecznik-sterowany-1x-16a-zigbee-tuya-pomiar-en-p-850.html

I tried using the converter from earlier posts  (@PetrozPL)  but not works I have similar debugs logs:

`2023-07-21 09:17:51Received Zigbee message from '0xa4c1384c97edfb66', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":16}],"seq":6487}' from endpoint 1 with groupID null
2023-07-21 09:17:51Received Zigbee message from '0xa4c1384c97edfb66', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,4],"type":"Buffer"},"datatype":2,"dp":1}],"seq":6743}' from endpoint 1 with groupID null
2023-07-21 09:17:51Received Zigbee message from '0xa4c1384c97edfb66', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":5,"dp":9}],"seq":6999}' from endpoint 1 with groupID null
2023-07-21 09:17:51Received Zigbee message from '0xa4c1384c97edfb66', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,26,0,0,170,0,0,2],"type":"Buffer"},"datatype":0,"dp":6}],"seq":7255}' from endpoint 1 with groupID null
2023-07-21 09:17:51Received Zigbee message from '0xa4c1384c97edfb66', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1,1,0,16,3,1,1,19,4,1,0,175],"type":"Buffer"},"datatype":0,"dp":18}],"seq":7511}' from endpoint 1 with groupID null
2023-07-21 09:17:51Received Zigbee message from '0xa4c1384c97edfb66', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[69,84,85,49,45,73,79,84,45,90,105,103,98,101,101,0],"type":"Buffer"},"datatype":3,"dp":19}],"seq":7767}' from endpoint 1 with groupID null`

but I get a message in the logs:

Exception while calling fromZigbee converter: tuya.getDataValue is not a function}

bdkacz, 2023-08-11:
> Hi I have a similar device:
> 
> https://www.rtx24.pl/bezpiecznik-sterowany-1x-16a-zigbee-tuya-pomiar-en-p-850.html
> 
> I tried using the converter from earlier posts (@PetrozPL) but not works I have similar debugs logs:

add in converter on top  const legacy = require('zigbee-herdsman-converters/lib/legacy');

and replace line

 const value = tuya.getDataValue(dpValue);

with  

const value = legacy.getDataValue(dpValue);

SCiunczyk, 2024-03-26:
Hi guys, looks like the ZMB9-Z-?P Smart Circuit Breaker is working fine. Thanks for your efforts. Maybe it's time to add it to the official repo? What do you think?

knakonieczny, 2024-05-20:
Hi , after last updates converter stop to work for me.

[image]

knakonieczny, 2024-05-23:
I rewrote basic code , working standard functions below.
```

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const {Buffer} = require('buffer');

const phaseVariantNew = {
    from: (v) => {
        const buf = Buffer.from(v, 'base64');
        return {
        voltage: (   (buf[0] << 8)  | buf[1]    ) / 10,
        current: (   (buf[2] << 16) |  (buf[3] << 8)  |  buf[4]   ) / 1000,
        power : (    (buf[5] << 16) |  (buf[6] << 8)  |  buf[7]   )
        }; 
    },
  };

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_abatw3kj',
        },
    ],
    model: 'TS0601_badz3redition',
    vendor: 'RTX',
    description: 'Circuit breaker with meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [e.switch(),e.energy(),e.current(),e.voltage(),e.power(),

    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1,'energy',tuya.valueConverter.divideBy100],
            [6, null, phaseVariantNew],
            [16, 'state', tuya.valueConverter.onOff],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

[image]