Koenkk/zigbee2mqtt #12829: Unable to add Aqara to group WSDCGQ11LM

balucio, 2022-06-15:
### What happened?

I tried to add an WSDCGQ11LM sensor to group. 

### What did you expect to happen?

The sensor will added to the group

### How to reproduce it (minimal and precise)

Adding the sensor to specific group

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20211115

### Adapter

Xiaomi Aqara T/H/P sensor WSDCGQ11LM

### Debug log

```
domus-zigbee-router | Zigbee2MQTT:info  2022-06-15 12:01:46: MQTT publish: topic 'zigbee2mqtt/camera da letto', payload ''
domus-zigbee-router | Zigbee2MQTT:info  2022-06-15 12:01:46: MQTT publish: topic 'zigbee2mqtt/camera da letto/availability', payload 'null'
domus-zigbee-router | Zigbee2MQTT:info  2022-06-15 12:01:46: MQTT publish: topic 'zigbee2mqtt/camera_letto/availability', payload 'online'
domus-zigbee-router | Zigbee2MQTT:info  2022-06-15 12:01:47: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/rename', payload '{"data":{"from":"camera da letto","homeassistant_rename":false,"to":"camera_letto"},"status":"ok","transaction":"iua4z-2"}'
domus-zigbee-router | Zigbee2MQTT:info  2022-06-15 12:02:11: Adding 'clima/stanza_letto/aqara_weather' to 'camera_letto'
domus-zigbee-router | Zigbee2MQTT:info  2022-06-15 12:03:02: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x00158d0007f29f3b/1","group":"camera_letto"},"error":"Failed to add from group (Command 0x00158d0007f29f3b/1 genGroups.add({\"groupid\":1,\"groupname\":\"\"}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC transaction expired' (240)))","status":"error","transaction":"iua4z-3"}'
domus-zigbee-router | Zigbee2MQTT:error 2022-06-15 12:03:02: Failed to add from group (Command 0x00158d0007f29f3b/1 genGroups.add({"groupid":1,"groupname":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)))
```

balucio, 2022-06-15:
Oh sorry I didn't know.
I'd like to have the group name somewhere in MQTT payload, to avoid to put in in topic. 
Indeed I don't know if it possible to add custom attribute based on sensor, for example position and some other custom attributes.
Sorry for the question i'm new in zigbee/mqtt world.

tiagomaricate, 2022-06-19:
I'm having this message when I try to put in a group some sensors from Aqara
[image]
 .

tiagomaricate, 2022-06-20:
> You cannot put Aqara sensors in a group, what do you want to achieve?

Hummmm...
I'd like to control/monitor a group of sensors.
Like all from the kitchen. Or all door sensors.

Koenkk, 2022-06-21:
That is not where Zigbee groups are meant for, they are meant for controlling devices (e.g. lights/switches)

closed 2022-07-29 by github-actions[bot]