Koenkk/zigbee2mqtt #12837: Skydance Tuya WZ5 LED 5 in 1 - No converter available for 'get' 'color' ()

jmbinette, 2022-06-16:
### What happened?

When trying to get the color status, we get the error :
No converter available for 'get' 'color' ()

I have no experience in writing converters. 
@kirovilya could you help ?

### What did you expect to happen?

Get the actual color status of the controller (Could be changed via a RF Remote)

### How to reproduce it (minimal and precise)

In the EXPOSE section of the device, click the refresh button on the left of colors

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0

jmbinette, 2022-06-22:
bump anyone ?

closed 2022-07-31 by github-actions[bot]

Sangoku, 2023-06-09:
bumpt again?

pLavrenov, 2024-04-24:
up

Koenkk, 2024-04-25:
TuYa device don't support reading attributes, that's why you get the error.

pLavrenov, 2024-04-26:
> Устройство TuYa не поддерживает чтение атрибутов, поэтому вы получаете сообщение об ошибке.

What needs to be done to send parameters “h” and “s” to the mqtt /set channel. Now "hex" is sent and it doesn't work on the device. I rewrote the converter but it seems to me that this is not a very universal approach

[https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/skydance.ts](url)